paxapos/fiscalberry #1: python-serial windows

adrianrey05, 2016-07-14:
Hola, 
De qué manera puedo instalar python-serial bajo el sistema operativo windows.

Desde ya muchas gracias

alevilar, 2016-07-14:
Hola! podes probar con
pip install pyserial

y comentame si te funcionó!!

saludos

adrianrey05, 2016-07-14:
Hola @alevilar 
si, googleando un poco me encontré con ese simple comando con pip, muchas gracias.
Ahora el tema es el siguiente.
estoy haciendo un test usando la pagina js_browser_client.
Cuando hago la primer venta me da el siguiente error

`
  File "e:\fiscalberry-master\Comandos\EpsonComandos.py", line 205, in closeDocument
    reply = self._sendCommand(self.CMD_CLOSE_FISCAL_RECEIPT[self._getCommandIndex()], ["T"])
  File "e:\fiscalberry-master\Comandos\EpsonComandos.py", line 77, in _sendCommand
    raise ComandoException("Error de la impresora fiscal: " + str(e))
ComandoException: Error de la impresora fiscal: Es necesario hacer un cierre de la jornada fiscal o se superó la cantidad máxima de tickets en una factura.
pyProc.stderr: ERROR:tornado.application:Uncaught exception in /ws
Traceback (most recent call last):
  File "C:\Python27\lib\site-packages\tornado\websocket.py", line 417, in _run_callback
    callback(*args, **kwargs)
  File "e:\fiscalberry-master/server.py", line 34, in on_message
    response = self.traductor.json_to_comando( jsonMes )
  File "e:\fiscalberry-master\Traductor.py", line 236, in json_to_comando
    return self._cerrarComprobante()
  File "e:\fiscalberry-master\Traductor.py", line 156, in _cerrarComprobante
    return self.printer.closeDocument()
  File "e:\fiscalberry-master\Comandos\EpsonComandos.py", line 205, in closeDocument
    reply = self._sendCommand(self.CMD_CLOSE_FISCAL_RECEIPT[self._getCommandIndex()], ["T"])
  File "e:\fiscalberry-master\Comandos\EpsonComandos.py", line 77, in _sendCommand
    raise ComandoException("Error de la impresora fiscal: " + str(e))
ComandoException: Error de la impresora fiscal: Es necesario hacer un cierre de la jornada fiscal o se superó la cantidad máxima de tickets en una factura.
`

alevilar, 2016-07-14:
Creo que estas trabajando con una impresora fiscal inicializada?, puede ser? en ese caso deberás hacer un cierre Z primero.
(OJO al hacer pruebas, para ello lo ideal es trabajar con una impresora fiscal en modo "test" o sin inicializar.)

para hacer un cierre Z probá desde la página de tests (js_browser_client) este JSON:

{"dailyClose": "Z"}

Deberia imprimirte un cierre fiscal Zeta

Haciendo estas pruebas descubrí un bug en la función "dailyClose", por lo tanto, por favor, descarga la ultima version nuevamente para que te funcione correctamente.
Tené en cuenta que la aplicacion aun puede tener bugs (la subi ayer), por lo tanto, agradeceria que me vayas enviando todo lo que encuentres para arreglarlo lo mas rapido posible.

adrianrey05, 2016-07-14:
@alevilar 
Perfecto, ahora me bajo la nueva version y hago las prubas.
La fiscal que tengo es una Epson TM-U220AF2 en modo desarrollo. Tengo entendido que lleva los mismos comandos por eso es que funciona :+1: 

Hago las pruebas y te voy avisando. 

Muchas gracias!

alevilar, 2016-07-14:
Dale! avisame asi voy agregando nuevos modelos como compatibles. 
en el archivo de configuracion, **_config.ini_**, ponele  **_model=tm-220-af_** 

Yo vengo trabajando con impresoras Hasar 441 y 715 y me anda perfecto. Con las otras deberia andar, pero no las pude testear. Asique espero tu respuesta a ver si todo salio bien.

Prometo ir mejorando los mensajes de error. Por ahora me quedo conforme con que logres imprimir!! saludos

adrianrey05, 2016-07-14:
@alevilar 
al querer hacer el cierre **Z** o **X** me da este error.

```cmd
Iniciando procesamiento de json...
esto es lo que estoy viendo
{u'dailyClose': u'X'}
X
_sendCommand 57 [u'X', 'P']
ERROR:tornado.application:Uncaught exception in /ws
Traceback (most recent call last):
  File "C:\Python27\lib\site-packages\tornado\websocket.py", line 417, in _run_callback
    callback(*args, **kwargs)
  File "e:\fiscalberry-master/server.py", line 37, in on_message
    response = self.traductor.json_to_comando( jsonMes )
  File "e:\fiscalberry-master\Traductor.py", line 247, in json_to_comando
    return self._dailyClose(jsonTicket["dailyClose"])
  File "e:\fiscalberry-master\Traductor.py", line 163, in _dailyClose
    ret = self.printer.dailyClose(type)
  File "e:\fiscalberry-master\Comandos\EpsonComandos.py", line 304, in dailyClose
    reply = self._sendCommand(self.CMD_DAILY_CLOSE, [type, "P"])
  File "e:\fiscalberry-master\Comandos\EpsonComandos.py", line 74, in _sendCommand
    return self.conector.sendCommand(commandNumber, parameters, skipStatusErrors)
  File "e:\fiscalberry-master\ConectorDriverComando.py", line 41, in sendCommand
    return self.driver.sendCommand(*args)
  File "e:\fiscalberry-master\Drivers\FiscalPrinterDriver.py", line 67, in sendCommand
    reply = self._sendMessage( message )
  File "e:\fiscalberry-master\Drivers\EpsonDriver.py", line 49, in _sendMessage
	self._write( message )
  File "e:\fiscalberry-master\Drivers\FiscalPrinterDriver.py", line 76, in _write
    self._serialPort.write( s )
  File "C:\Python27\lib\site-packages\serial\serialwin32.py", line 308, in write
    data = to_bytes(data)
  File "C:\Python27\lib\site-packages\serial\serialutil.py", line 63, in to_bytes
    raise TypeError('unicode strings are not supported, please encode to bytes: {!r}'.format(seq))
TypeError: unicode strings are not supported, please encode to bytes: u'\x02<9\x1cX\x1cP\x03015A'
pyProc.stderr: ERROR:tornado.application:Uncaught exception in /ws
Traceback (most recent call last):
  File "C:\Python27\lib\site-packages\tornado\websocket.py", line 417, in _run_callback
    callback(*args, **kwargs)
  File "e:\fiscalberry-master/server.py", line 37, in on_message
    response = self.traductor.json_to_comando( jsonMes )
  File "e:\fiscalberry-master\Traductor.py", line 247, in json_to_comando
    return self._dailyClose(jsonTicket["dailyClose"])
  File "e:\fiscalberry-master\Traductor.py", line 163, in _dailyClose
    ret = self.printer.dailyClose(type)
  File "e:\fiscalberry-master\Comandos\EpsonComandos.py", line 304, in dailyClose
    reply = self._sendCommand(self.CMD_DAILY_CLOSE, [type, "P"])
  File "e:\fiscalberry-master\Comandos\EpsonComandos.py", line 74, in _sendCommand
    return self.conector.sendCommand(commandNumber, parameters, skipStatusErrors)
  File "e:\fiscalberry-master\ConectorDriverComando.py", line 41, in sendCommand
    return self.driver.sendCommand(*args)
  File "e:\fiscalberry-master\Drivers\FiscalPrinterDriver.py", line 67, in sendCommand
    reply = self._sendMessage( message )
  File "e:\fiscalberry-master\Drivers\EpsonDriver.py", line 49, in _sendMessage
    self._write( message )
  File "e:\fiscalberry-master\Drivers\FiscalPrinterDriver.py", line 76, in _write
    self._serialPort.write( s )
  File "C:\Python27\lib\site-packages\serial\serialwin32.py", line 308, in write
    data = to_bytes(data)
  File "C:\Python27\lib\site-packages\serial\serialutil.py", line 63, in to_bytes
    raise TypeError('unicode strings are not supported, please encode to bytes: {!r}'.format(seq))
TypeError: unicode strings are not supported, please encode to bytes: u'\x02<9\x1cX\x1cP\x03015A'
connection closed
```

adrianrey05, 2016-07-14:
@alevilar 
La version de python que uso es la 2.7
Perfecto! funcionó de 10! Muchas gracias por tu ayuda! 
Cualquier cosa que necesites para probar a tu disposición. :+1:

closed 2016-07-14 by adrianrey05

alevilar, 2016-07-14:
me alegro!, ahi subo la version al repo con ese cambio!

si ves cosas raras, tenés sugerencias o lo que sea.... me avisas!, si
llegas a probarla en otras marcas o modelos, también asi voy armando una
lista de compatibilidad

saludos

adrianrey05, 2016-07-14:
@alevilar 
Lo que note es que al momento de dar un error cierra la conexión web socket.
En la misma instancia del servidor corriendo no puedo volver a conectarme ya que queda la conexión com1 abierta.
Tengo que parar y volver a correr el servidor para que "suelte" el COM1.
Estaria bueno que al cerrar la conexión o error libere el puerto para poder realizar una nueva conexión.

Sldos!

alevilar, 2016-07-14:
ok, tomado como pendiente! gracias por tu aporte. todo lo que creas puede
servir como mejora me avisas! no me molesta para nada, dado que soy usuario
del fiscalberry, tengo clientes que lo usan y debo asegurarme que no tenga
bugs.
:)
saludos