paxapos/fiscalberry #2: error puerto abierto

adrianrey05, 2016-07-15:
Buenas @alevilar 
Con js_browser_client, al actualizar la página intentan conectarse de nuevo al ws y da este error.

```sh
Reading config file: config.ini
Impresora disponible:
  - IMPRESORA_FISCAL
      marca: Epson, modelo: tm-220-af (COM1)
*** Websocket Server Started at 169.254.18.61 port 8080***
new connection
Iniciando procesamiento de json...
esto es lo que estoy viendo
{u'getStatus': {}}
connection closed
ERROR:tornado.application:Uncaught exception
Traceback (most recent call last):
  File "C:\Python27\lib\site-packages\tornado\http1connection.py", line 238, in _read_message
    delegate.finish()
  File "C:\Python27\lib\site-packages\tornado\httpserver.py", line 289, in finish
    self.delegate.finish()
  File "C:\Python27\lib\site-packages\tornado\web.py", line 2022, in finish
    self.execute()
  File "C:\Python27\lib\site-packages\tornado\web.py", line 2042, in execute
    **self.handler_kwargs)
  File "e:\fiscalberry-master/server.py", line 35, in __init__
    self.traductor = Traductor('IMPRESORA_FISCAL')
  File "e:\fiscalberry-master\Traductor.py", line 74, in __init__
    self.comando = EpsonComandos(deviceFile, driverName=driverName)
  File "e:\fiscalberry-master\Comandos\EpsonComandos.py", line 59, in __init__
    self.conector = ConectorDriverComando(self, driverName, deviceFile, speed)
  File "e:\fiscalberry-master\ConectorDriverComando.py", line 36, in __init__
    raise ConectorError("Imposible establecer comunicación.", e)
ConectorError: ('Imposible establecer comunicaci\xc3\xb3n.', SerialException("could not open port 'COM1': WindowsError(5, 'Acceso denegado.')",))
pyProc.stderr: ERROR:tornado.application:Uncaught exception
Traceback (most recent call last):
  File "C:\Python27\lib\site-packages\tornado\http1connection.py", line 238, in _read_message
    delegate.finish()
  File "C:\Python27\lib\site-packages\tornado\httpserver.py", line 289, in finish
    self.delegate.finish()
  File "C:\Python27\lib\site-packages\tornado\web.py", line 2022, in finish
    self.execute()
  File "C:\Python27\lib\site-packages\tornado\web.py", line 2042, in execute
    **self.handler_kwargs)
  File "e:\fiscalberry-master/server.py", line 35, in __init__
    self.traductor = Traductor('IMPRESORA_FISCAL')
  File "e:\fiscalberry-master\Traductor.py", line 74, in __init__
    self.comando = EpsonComandos(deviceFile, driverName=driverName)
  File "e:\fiscalberry-master\Comandos\EpsonComandos.py", line 59, in __init__
    self.conector = ConectorDriverComando(self, driverName, deviceFile, speed)
  File "e:\fiscalberry-master\ConectorDriverComando.py", line 36, in __init__
    raise ConectorError("Imposible establecer comunicación.", e)
ConectorError: ('Imposible establecer comunicaci\xc3\xb3n.', SerialException("could not open port 'COM1': WindowsError(5, 'Acceso denegado.')",))
```

alevilar, 2016-07-15:
eso falla cuando quieren conectarse el 2do clieente, no? si queres agregame al hangout de google para resolverlo mas rapido alevilar@gmail.com

adrianrey05, 2016-07-15:
no probé dos clientes, ahora hago la prueba, es simplemente cuando actualizo la pagina.
Ya te agregue al gtalk

adrianrey05, 2016-07-15:
fix en commit f99b01683261406fe5e8e5010ae6630de945acae por @alevilar 
Gracias!

closed 2016-07-15 by adrianrey05