paxapos/fiscalberry #3: AttributeError: EpsonComandos instance has no attribute 'traductor'

catrielr, 2016-10-05:
Estoy probando desde una pc con windows, este es mi CONFIG.INI, solo falla al cofiguarar marca y driver con Epson. Con Hasar funciona OK

CONFIG.INI
=====================
[SERVIDOR]
puerto = 12000

[IMPRESORA_FISCAL]
marca = Epson		
modelo = tm-220-af
path = COM2
driver = Epson

[IMPRESORA_RED]
marca = EscP
host = 127.0.0.1
driver = ReceiptDirectJet

=====================
Salida de consola:

C:\Python27\python.exe C:/xampp/htdocs/dolibarr35/htdocs/fiscalprinter/fiscalberry-master/server.py
inicializando ConectorDriverComando driver de Epson
Traceback (most recent call last):
  File "C:/xampp/htdocs/dolibarr35/htdocs/fiscalprinter/fiscalberry-master/server.py", line 37, in <module>
    traductor = TraductoresHandler()
  File "C:\xampp\htdocs\dolibarr35\htdocs\fiscalprinter\fiscalberry-master\Traductores\TraductoresHandler.py", line 83, in __init__
    self.__init_cola_traductores_printer()
  File "C:\xampp\htdocs\dolibarr35\htdocs\fiscalprinter\fiscalberry-master\Traductores\TraductoresHandler.py", line 160, in __init_cola_traductores_printer
    self.init_printer_traductor(s)
  File "C:\xampp\htdocs\dolibarr35\htdocs\fiscalprinter\fiscalberry-master\Traductores\TraductoresHandler.py", line 145, in init_printer_traductor
    traductorComando = comando.traductor
AttributeError: EpsonComandos instance has no attribute 'traductor'

alevilar, 2016-10-05:
Hola, justo no tengo una EPSON para probar. Podrias realizar la siguiente prueba:
eliminá todas las lineas correspondiente al __init__ de Comandos/EpsonComandos.py para que herede directamente la funcion de inicializacion de la clase ComandoInterface a ver si de esa forma te funciona.

catrielr, 2016-10-05:
Ahi comente las lineas del init, te paso que que hace. Estoy probrando con el emulador fiscar, no con una impresora real.
Pero ya te digo, con Hazar funciona OK.
Ahora el server.py sigue corriendo, pero cualquier comando que envio tira:
[ERR] AttributeError("ConectorDriverComando instance has no attribute 'driver'",)- ConectorDriverComando instance has no attribute 'driver'

Cualquier prueba que necesites decime.
Abrazo!

Iniciando Fiscalberry Server
Hay 2 impresoras disponibles
  - IMPRESORA_FISCAL
      marca: Epson, driver: Epson
  - IMPRESORA_RED
      marca: EscP, driver: ReceiptDirectJet

*** Websocket Server Started at 192.168.56.1 port 12000***
new connection
{"getAvaliablePrinters":""}
{u'getAvaliablePrinters': u''}
{'rta': {'action': 'getAvaliablePrinters', 'rta': ['IMPRESORA_FISCAL', 'IMPRESORA_RED']}}
_sendCommand 42 ['N']
Exception in thread Thread-8:
Traceback (most recent call last):
  File "C:\Python27\lib\threading.py", line 801, in __bootstrap_inner
    self.run()
  File "C:\Python27\lib\threading.py", line 1073, in run
    self.function(*self.args, **self.kwargs)
  File "C:/xampp/htdocs/dolibarr35/htdocs/fiscalprinter/fiscalberry-master/server.py", line 166, in send_printer_warnings
    warns = traductor.getWarnings()
  File "C:/xampp/htdocs/dolibarr35/htdocs/fiscalprinter/fiscalberry-master\Traductores\TraductoresHandler.py", line 116, in getWarnings
    warn = traductores[trad].comando.getWarnings()
  File "C:/xampp/htdocs/dolibarr35/htdocs/fiscalprinter/fiscalberry-master\Comandos\EpsonComandos.py", line 356, in getWarnings
    reply = self._sendCommand(self.CMD_STATUS_REQUEST, ["N"], True)
  File "C:/xampp/htdocs/dolibarr35/htdocs/fiscalprinter/fiscalberry-master\Comandos\EpsonComandos.py", line 83, in _sendCommand
    return self.conector.sendCommand(commandNumber, parameters, skipStatusErrors)
  File "C:/xampp/htdocs/dolibarr35/htdocs/fiscalprinter/fiscalberry-master\ConectorDriverComando.py", line 30, in sendCommand
    return self.driver.sendCommand(*args)
AttributeError: ConectorDriverComando instance has no attribute 'driver'

**> ACA liste las impresoras**

{"getLastNumber":"T","printerName":"IMPRESORA_FISCAL"}
{u'getLastNumber': u'T', u'printerName': u'IMPRESORA_FISCAL'}
_sendCommand 42 ['A']
Traceback (most recent call last):
  File "C:/xampp/htdocs/dolibarr35/htdocs/fiscalprinter/fiscalberry-master/server.py", line 55, in on_message
    response = traductor.json_to_comando( jsonMes )
  File "C:/xampp/htdocs/dolibarr35/htdocs/fiscalprinter/fiscalberry-master\Traductores\TraductoresHandler.py", line 267, in json_to_comando
    rta["rta"] = traductor.run( jsonTicket )
  File "C:/xampp/htdocs/dolibarr35/htdocs/fiscalprinter/fiscalberry-master\Traductores\TraductorInterface.py", line 28, in run
    res = fnAction( jsonTicket[action] )
  File "C:/xampp/htdocs/dolibarr35/htdocs/fiscalprinter/fiscalberry-master\Traductores\TraductorFiscal.py", line 38, in getLastNumber
    return self.comando.getLastNumber(letra_cbte)
  File "C:/xampp/htdocs/dolibarr35/htdocs/fiscalprinter/fiscalberry-master\Comandos\EpsonComandos.py", line 317, in getLastNumber
    reply = self._sendCommand(self.CMD_STATUS_REQUEST, ["A"], True)
  File "C:/xampp/htdocs/dolibarr35/htdocs/fiscalprinter/fiscalberry-master\Comandos\EpsonComandos.py", line 83, in _sendCommand
    return self.conector.sendCommand(commandNumber, parameters, skipStatusErrors)
  File "C:/xampp/htdocs/dolibarr35/htdocs/fiscalprinter/fiscalberry-master\ConectorDriverComando.py", line 30, in sendCommand
    return self.driver.sendCommand(*args)
AttributeError: ConectorDriverComando instance has no attribute 'driver'
{'err': 'AttributeError("ConectorDriverComando instance has no attribute \'driver\'",)- ConectorDriverComando instance has no attribute \'driver\''}

alevilar, 2016-10-07:
Yo uso Linux (ubuntu) y no conozco algun emulador que funcione con este SO, por lo tanto para hacer pruebas uso la salida a un archivo.

Bajate la ultima version y proba de configurar tu impresora fiscal en tu config.ini de esta manera, cofirmame si te funciono! 

[IMPRESORA_FISCAL]
marca = Epson
modelo = tm-220-af
path = /tmp/respuestas.txt (ACA PONE TU CARPETA)
driver = File

catrielr, 2016-10-08:
Ale, estas pruebas las hice con las ultimas actualizaciones, cuando le pongo en el config.ini, driver = File, funciona OK, pero cuando lo cambio a Epson y el path = COM2 tira los errores que te pego a continuacion.
Con Hazar en COM2 funciona OK
Abrazo
Catriel

CONFIG.INI
==========
[SERVIDOR]
puerto = 12000

[IMPRESORA_FISCAL]
marca = Epson
modelo = tm-220-af
path = COM2
driver = Epson

SALIDA DE CONSOLA:
===============
Connected to pydev debugger (build 139.1547)
inicializando ConectorDriverComando driver de Epson
Exception in thread Thread-6:
Traceback (most recent call last):
  File "C:\Python27\lib\threading.py", line 801, in __bootstrap_inner
    self.run()
  File "C:\Python27\lib\threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "C:/xampp/htdocs/dolibarr35/htdocs/fiscalprinter/fiscalberry-master\ConectorDriverComando.py", line 21, in init_driver
    self.driver = driverClass(**kwargs)
**TypeError: __init__() got an unexpected keyword argument 'path'**

Iniciando Fiscalberry Server
Impresora disponible:
  - IMPRESORA_FISCAL
      marca: Epson, driver: Epson

*** Websocket Server Started at 192.168.56.1 port 12000***
new connection
new connection
new connection
_sendCommand 42 ['N']
Exception in thread Thread-7:
Traceback (most recent call last):
  File "C:\Python27\lib\threading.py", line 801, in __bootstrap_inner
    self.run()
  File "C:\Python27\lib\threading.py", line 1073, in run
    self.function(*self.args, **self.kwargs)
  File "C:/xampp/htdocs/dolibarr35/htdocs/fiscalprinter/fiscalberry-master/server.py", line 171, in send_printer_warnings
    warns = traductor.getWarnings()
  File "C:/xampp/htdocs/dolibarr35/htdocs/fiscalprinter/fiscalberry-master\Traductores\TraductoresHandler.py", line 116, in getWarnings
    warn = traductores[trad].comando.getWarnings()
  File "C:/xampp/htdocs/dolibarr35/htdocs/fiscalprinter/fiscalberry-master\Comandos\EpsonComandos.py", line 354, in getWarnings
    reply = self._sendCommand(self.CMD_STATUS_REQUEST, ["N"], True)
  File "C:/xampp/htdocs/dolibarr35/htdocs/fiscalprinter/fiscalberry-master\Comandos\EpsonComandos.py", line 76, in _sendCommand
    return self.conector.sendCommand(commandNumber, parameters, skipStatusErrors)
  File "C:/xampp/htdocs/dolibarr35/htdocs/fiscalprinter/fiscalberry-master\ConectorDriverComando.py", line 30, in sendCommand
    return self.driver.sendCommand(*args)
AttributeError: ConectorDriverComando instance has no attribute 'driver'

CONFIG.INI
=========
[SERVIDOR]
puerto = 12000

[IMPRESORA_FISCAL]
marca = Epson
modelo = tm-220-af
path = C:\xampp\htdocs\dolibarr35\htdocs\fiscalprinter\fiscalberry-master\respuestas.txt
driver = File

SALIDA DE CONSOLA
==============

Connected to pydev debugger (build 139.1547)
inicializando ConectorDriverComando driver de File
Iniciando Fiscalberry Server
Impresora disponible:
  - IMPRESORA_FISCAL
      marca: Epson, driver: File

*** Websocket Server Started at 192.168.56.1 port 12000***
new connection
_sendCommand 42 ['N']
{'IMPRESORA_FISCAL': ['Poco papel para comprobantes o tickets']}

tioeze, 2016-10-11:
A mi también me pasa lo mismo que a catrielr, de hecho lo charlamos por privado.
Las condiciones son iguales: SO Windows con emulador fiscal.
El error es el siguiente:

`inicializando ConectorDriverComando driver de Epson
no se pudo conctar con el driver: Epson
Reading config file: config.ini
Impresora disponible:
  - IMPRESORA_FISCAL
      marca: Epson, driver: Epson

*** Websocket Server Started at 10.0.2.15 port 12001***
new connection
{u'getAvaliablePrinters': u''}
{'rta': {'action': 'getAvaliablePrinters', 'rta': ['IMPRESORA_FISCAL']}}
new connection
{u'getLastNumber': u'T', u'printerName': u'IMPRESORA_FISCAL'}
{'rta': ''}
_sendCommand 42 ['N']
Exception in thread Thread-1:
Traceback (most recent call last):
  File "C:\Python27\lib\threading.py", line 801, in __bootstrap_inner
    self.run()
  File "C:\Python27\lib\threading.py", line 1073, in run
    self.function(*self.args, **self.kwargs)
  File "C:\Documents and Settings\Administrador\Mis documentos\Descargas\fiscalberry\server.py", line 94, in send_printer_warnings
    warns = traductor.getWarnings()
  File "C:\Documents and Settings\Administrador\Mis documentos\Descargas\fiscalberry\Traductores\TraductoresHandler.py", line 76, in getWarnings
    warn = self.traductores[trad].comando.getWarnings()
  File "C:\Documents and Settings\Administrador\Mis documentos\Descargas\fiscalberry\Comandos\EpsonComandos.py", line 354, in getWarnings
    reply = self._sendCommand(self.CMD_STATUS_REQUEST, ["N"], True)
  File "C:\Documents and Settings\Administrador\Mis documentos\Descargas\fiscalberry\Comandos\EpsonComandos.py", line 76, in _sendCommand
    return self.conector.sendCommand(commandNumber, parameters, skipStatusErrors)
AttributeError: 'NoneType' object has no attribute 'sendCommand'`

alevilar, 2016-10-11:
Pareciera ser un problema con la conexión a la impresora (o emulador).
Quizás un problema con el puerto COM?

El objeto "conector" no está siendo instanciado. El conector es el
"Driver". EpsonDriver en este caso.

Habría que buscarlo por ese lado. Yo estos días no lo podré ver a fondo.
Pero cerca del finde tratare de resolverlo.

tioeze, 2016-10-11:
>Pareciera ser un problema con la conexión a la impresora (o emulador). Quizás un problema con el puerto COM?

Algo que me pasó una vez con el pyfiscalprinter de Mariano Reingart (desde donde entiendo que se basa este) para Epson hay que levantar el servicio en windows para que funcione y ahi recién podés "conectarte" a la impresora.

No recuerdo como lo logré, pero luego de registrar el servicio y ejecutarlo, pude imprimir con epson (en el emulador al menos)

Tiro esta info porque por ahí Uds que saben mas, pueden encontrarle la vuelta.

catrielr, 2016-10-12:
Lo mas raro que cuando pones driver = File, funca. mañana hago mas pruebas

alevilar, 2016-10-12:
Así es, los archivos EpsonDriver y HasarDriver están basados en el proyecto
de Mariano Reingart.
Por lo que debe tener algo que ver con lo que dice ezeq. Alguna relación
entre el driver y el emulador.
Pareciera como que el puerto COM no está conectando con el driver.

tioeze, 2016-10-15:
Bueno, tuve un ratito para mirarlo y encontré lo que les decía mas arriba.

El servicio es este [https://github.com/reingart/pyfiscalprinter/blob/trunk/epsonFiscalDriver_service.py](https://github.com/reingart/pyfiscalprinter/blob/trunk/epsonFiscalDriver_service.py) y cuando se "enciende" deja imprimir en epson con el código "a pelo" de pyfiscalprinter. 

Intenté "encender" el servicio y usar fiscalberry, pero sigue con el mismo error. (Tiro esta data porque mas no sé, pero por ahi puede ayudar)

Espero sus comentarios

eoyarzabal, 2016-11-26:
Estimados, 
Estuve probando mi ["código" implementado para Dolibarr](https://github.com/eoyarzabal/fiscaldoli) ultra básico y de poca base de programación, que lo único que tiene que ver con este es que usa pyfiscalprinter para imprimir en los fiscales, pero no tuve que levantar el "server" del cual hablé mas arriba.

Es decir que creo que el problema es del emulador por lo que habría que probarlo en un fiscal de carne y hueso para descartarlo.

Saludos

closed 2017-06-28 by alevilar

juanmitriatti, 2017-11-25:
Hola Gente, tengo el mismo error(tengo una epson-220-afii). Lo pudieron solucionar?

alevilar, 2017-11-25:
buenas!, copianos tu config.ini, seguramente sea un problema de configuracion

juanmitriatti, 2017-11-25:
[SERVIDOR]
puerto = 12001
ip_privada = 192.168.1.101

[IMPRESORA_FISCAL]
marca = Epson
modelo = tm-220-af
path = COM2
driver = Epson

[IMPRESORA_RED]
marca = EscP
host = 127.0.0.1
driver = ReceiptDirectJet

juanmitriatti, 2017-11-25:
La tengo conectada al com2, probando el programa (tipo IFXBATCH) que trae EPSON me funciona correctamente  En esta aplicación uso Host Port (USB) y Fiscal Port (COM2).

alevilar, 2017-11-25:
copiá, por favor, el detalle del error. en que linea del codigo te lo está
tirando?

juanmitriatti, 2017-11-25:
[ERR] 'NoneType' object has no attribute 'sendCommand'

Ya no tengo errores con el puerto. El error actual es el error más reciente desde el example_ws_client.html.

alevilar, 2017-11-25:
acá, en el titulo de esta issue dice:
"AttributeError: EpsonComandos instance has no attribute 'traductor'"

si tu error es "[ERR] 'NoneType' object has no attribute 'sendCommand'",
entoces es otro el problema.

pasame una copia completa del error que te sale en la consola.

juanmitriatti, 2017-11-25:
PERDON! MIL DISCULPAS.  (TENIA ABIERTO DOS SERVIDORES) 
EL error siempre fue este :
[ERR] SerialException("could not open port 'COM2': WindowsError(2, 'El sistema no puede encontrar el archivo especificado.')",)- could not open port 'COM2': WindowsError(2, 'El sistema no puede encontrar el archivo especificado.')
[image]

Adjunto copia del error.

alevilar, 2017-11-25:
el error se puede dar debido a que:
1- necesites correr la consola como administrador.
2- actualmente el puerto COM2 este siendo utilizado por otro programa u otra instancia de fiscalberry.

no soy experto en windows, asique muchisimo no te puedo ayudar con ese SO, pero segun googleé esos pueden ser motivos de ese error.