paxapos/fiscalberry #4: UnicodeDecodeError: 'ascii' codec can't decode byte 0x93 in position 2: ordinal not in range(128)

juank3bytes, 2017-01-29:
Buenas tardes. Estamos usando la version para PHP con una impresora Hasar 441F.
Para Imprimir facturas y ticket no tenemos problemas pero para imprimir nota de creditos no da el siguiente error:
UnicodeDecodeError: 'ascii' codec can't decode byte 0x93 in position 2: ordinal not in range(128)

El formato del json es el siguiente:
{
	"printTicket": {
		"encabezado": {
			"tipo_cbte": "NCB",
			"referencia": "00105838"
		},
		"items": [{
			"alic_iva": "21.00",
			"importe": 1,
			"ds": "DEPILACION",
			"qty": "1"
		}],
		"pagos": []
	},
	"printerName": "IMPRESORA_FISCAL"
}

alevilar, 2017-03-31:
ya pudimos resolver este error. probalo con la ultima version

juank3bytes, 2017-04-07:
Buen dia. Muchas gracias por la respuesta. Tenemos  funcionando actualmente
el modulo pero sin hacer nota de credito.
Ya probaremos con la ultima version. Desde ya muchas gracias.

Juan Carlos Silva Cornejo
Software Developer - Yii Framework - Oracle Apex
3bytes - Soluciones de Software

alevilar, 2017-04-07:
no. la NC te la debo :( no la pudimos hacer funcionar aun. No entedemos por
que.. pero tampoco pudimos dedicarle tiempo.
De todos modos este issue es debido al uso de algun caracter no unicode,
como por ejemplo una EÑE o un acento.

Ahora deberia comvertir las eñes en "enes" y acentos en letras sin acento,
para que no tire ese error.

en las impresoras que soportas caracteres unicode (como por ejemplo una
comandera) imprime correctamente con eñe o acento

juank3bytes, 2017-04-07:
Alejandro la verdad que no recordaba el error que te mande porque fue hace
mucho. De ahi hasta ahora hicimos mejoras en nuestro backend para que todo
funciones. Armamos una funcion en PHP para arreglar cualquier caracter
extraño.
Pero bueno lo que tenemos pendiente al igual que vos es la Nota de Credito.
Desde ya te estamos muy agradecidos por tu aporte porque la verdad que el
codigo funciona a la perfeccion.
Si nosotros llegamos a trabajar algo en python trataremos de aportarte.
Gracias

juank3bytes, 2017-04-20:
Buenas Tardes Alejandro:
Estimado tengo una consulta.Estamos trabajando con un modelo de Hasar que
es compatible con estos tamaños:

textSizeDict = {
        "615": {'nonFiscalText': 40,
                 'customerName': 30,
                 'custAddressSize': 40,
                 'paymentDescription': 30,
                 'fiscalText': 20,
                 'lineItem': 20,
                 'lastItemDiscount': 20,
                 'generalDiscount': 20,
                 'embarkItem': 108,
                 'receiptText': 106,
                },

El problema es que nos corta la descripción de un producto siempre en 20.
Mi pregunta es la siguiente si modifico el "lineItem" a mas caracteres la
impresion del ticket sera correcta o se rompera todo jajajaja.
Si cambiamos el tamaño de caracteres por linea tendremos que hacer algo
para saltos de linea o algo asi?. Te consulto porque no conozco mucho del
lenguaje y estamos revisando todos los archivos para entender como funciona.
Despues de investigar un poco el archivo HasarComandos.py creo que en la
linea:
self._formatText(description[-1], 'lineItem') esta haciendo el corte en 20
caracteres de la descripcion verdad?

Desde ya muchas gracias. Espero una pronta respuesta.

El 7 de abril de 2017, 11:51, Juan Carlos Silva Cornejo <jucadasrl@gmail.com

> Alejandro la verdad que no recordaba el error que te mande porque fue hace
> mucho. De ahi hasta ahora hicimos mejoras en nuestro backend para que todo
> funciones. Armamos una funcion en PHP para arreglar cualquier caracter
> extraño.
> Pero bueno lo que tenemos pendiente al igual que vos es la Nota de
> Credito.
> Desde ya te estamos muy agradecidos por tu aporte porque la verdad que el
> codigo funciona a la perfeccion.
> Si nosotros llegamos a trabajar algo en python trataremos de aportarte.
> Gracias
>
>
>> no. la NC te la debo :( no la pudimos hacer funcionar aun. No entedemos
>> por
>> que.. pero tampoco pudimos dedicarle tiempo.
>> De todos modos este issue es debido al uso de algun caracter no unicode,
>> como por ejemplo una EÑE o un acento.
>>
>> Ahora deberia comvertir las eñes en "enes" y acentos en letras sin acento,
>> para que no tire ese error.
>>
>> en las impresoras que soportas caracteres unicode (como por ejemplo una
>> comandera) imprime correctamente con eñe o acento
>>
>> 2017-04-07 11:38 GMT-03:00 Juan Carlos Silva Cornejo <
>>
>>
>> > Buen dia. Muchas gracias por la respuesta. Tenemos funcionando
>> actualmente
>> > el modulo pero sin hacer nota de credito.
>> > Ya probaremos con la ultima version. Desde ya muchas gracias.
>> >
>> > Juan Carlos Silva Cornejo
>> > Software Developer - Yii Framework - Oracle Apex
>> > 3bytes - Soluciones de Software
>> >
>> > <https://github.com/paxapos/fiscalberry/issues/4#issuecomment-292554589
>> >,
>> 6KUut3fZbxOiNMvMqgK93-6ks5rtkpigaJpZM4Lw6Aj>
>>
>> >
>>
>>
>
>
>
> Juan Carlos Silva Cornejo
> Software Developer - PHP (Yii,Laravel) - Oracle Apex
> 3bytes - Soluciones de Software

alevilar, 2017-04-24:
El 20 de abril de 2017, 18:34, Juan Carlos Silva Cornejo <
notifications@github.com> escribió:

> textSizeDict = {
> "615": {'nonFiscalText': 40,
> 'customerName': 30,
> 'custAddressSize': 40,
> 'paymentDescription': 30,
> 'fiscalText': 20,
> 'lineItem': 20,
> 'lastItemDiscount': 20,
> 'generalDiscount': 20,
> 'embarkItem': 108,
> 'receiptText': 106,
> },
>

Hola Juan!,
acabo de subir una nueva version incorporando el modelo de impresora que me
pasaste (con los parametros que me enviaste).

Fijate que uno de los parametros es  'lineItem': 20, y 'fiscalText': 20.
Eso quiere decir que la impresora no permite mas de 20 caracteres, por lo
que te lo recorta a ese ancho. Te soy sincero: no se que pasaria si le
pasas mas caracteres.. para colmo esa es una impresora bastante vieja, no?

probá a ver que tal anda la ultima version, y cualquier cosa avisame los
resultados que obtuviste !

saludos!!

Alejandro Vilar

closed 2017-05-26 by alevilar

juank3bytes, 2017-06-15:
Buenos dias Alejandro:
Estoy teniendo el siguiente problema con un ticket de un monto mayor a
$10000.
Me sale "Desbordamiento de totales". La impresora es hasar 441f.
Segun lei el manual dice que viene seteada de fabrica con un tope de 10mil.
Mi consulta era si en alguna parte del codigo de python esta seteada alguna
variable para ese limite.
Desde ya muchas gracias.

-- 
Juan Carlos Silva Cornejo
Software Developer - PHP (Yii,Laravel) - Oracle Apex
3bytes - Soluciones de Software

alevilar, 2017-06-15:
Hola!, lamentablemente hay una ley (totalmente desactualizada) que indica
que, cuando el monto es mayor a los $1000 pesos, y es Consumidor final,
será necesario poner el DNI del cliente... por lo tanto vas a tener que
probar de armar el ticket pero con DOC, tipo doc y nombre.

"encabezado": {
   "tipo_cbte": "T",
   "nro_doc": "30256256",
   "domicilio_cliente": "Apache, Fuerte 21212",
   "tipo_doc": "DNI",
   "nombre_cliente": "Carlitos Tevez",
},

Existe un ejecutable llamado "pruf", que te lo podes descargar desde la
pagina de Hasar, con el cual podrias modificar la configuracion de la
impresora para incrementar ese parametro y aceptar facturas "B" con un
limite superior a los $1000. Es ilegal modificar ese parametro. Pero
tecnicamente es posible y facil de hacerlo.

gonzaabel, 2017-06-20:
Hola Juan!
Proba la nota de crédito, ya deberías poder imprimirla.
Ejemplo Nota de Crédito "A":
```
{
"printTicket":{ 
"encabezado":{
"tipo_cbte":"NCA",
"referencia":"00066778",
"nombre_cliente":"Carlos Tevez",
"domicilio_cliente":"zaraza",
"tipo_responsable":"RESPONSABLE_INSCRIPTO",
"tipo_doc":"CUIT",
"nro_doc":20407778884
},
"items":
[ 
{
 "alic_iva":21,
"importe":1,
"ds":"PEPSI",
"qty":1
}
]
},"printerName":"IMPRESORA_FISCAL"}
```
Ejemplo Nota de Crédito "B":
```
{
"printTicket":
{
"encabezado":{
"tipo_cbte":"NCB",
"referencia":"00066811"
},
"items":
[
{
"alic_iva":21,
"importe":1,
"ds":"PEPSI",
"qty":1
}
]
},"printerName":"IMPRESORA_FISCAL"}
```
Esos son los JSON según el tipo de factura (Están agregados a la página de WS 2 botones con estos JSON). Cualquier cosa nos preguntas o creas una nueva issue :+1: 
Saludos! :smile: