paxapos/fiscalberry #14: Problemas con metodo getAvaliablePrinters

catrielr, 2017-07-04:
No lista las impresoras.
OS: Windows 10
Python: 2.7.13

A la impresora llega bien el comando 0x2A 

Copio el **config.ini**
`[SERVIDOR]
puerto = 12000

[IMPRESORA_FISCAL]
marca = Hasar
modelo = 615
path = COM2
driver = Hasar

[IMPRESORA_RED]
marca = EscP
host = 127.0.0.1
driver = ReceiptDirectJet`

Respuesta desde la consola:
`C:\fiscalberry>python server.py
inicializando ConectorDriverComando driver de Hasar
inicializando ConectorDriverComando driver de ReceiptDirectJet
Iniciando Fiscalberry Server
iniciando en modo HTTP
Hay 2 impresoras disponibles
  - IMPRESORA_FISCAL
      marca: Hasar, driver: Hasar
  - IMPRESORA_RED
      marca: EscP, driver: ReceiptDirectJet

*** Websocket Server Started at 192.168.56.1 port 12000***
new connection
connection closed
new connection
----- - -- - - - ---
{"getAvaliablePrinters":""}
{u'getAvaliablePrinters': u''}
Traceback (most recent call last):
  File "C:\fiscalberry\FiscalBerryStarter.py", line 41, in on_message
    response = traductor.json_to_comando( jsonMes )
  File "C:\fiscalberry\Traductores\TraductoresHandler.py", line 116, in json_to_comando
    rta["rta"] =  self._getAvaliablePrinters()
  File "C:\fiscalberry\Traductores\TraductoresHandler.py", line 276, in _getAvaliablePrinters
    self.__getPrintersAndWriteConfig()
  File "C:\fiscalberry\Traductores\TraductoresHandler.py", line 189, in __getPrintersAndWriteConfig
    printer = self.__getDeviceData(macs)
  File "C:\fiscalberry\Traductores\TraductoresHandler.py", line 151, in __getDeviceData
    nm = nmap.PortScanner()
AttributeError: 'module' object has no attribute 'PortScanner'
`

Saludos
Catriel

gonzaabel, 2017-07-04:
Hola Catriel, acabo de instalar Fiscalberry en Windows 10 y no puedo replicar el error.
Al parecer nmap no te esta funcionando correctamente, proba re-instalándolo con este installer: https://nmap.org/dist/nmap-7.50-setup.exe
Saludos.

alevilar, 2017-07-05:
@catrielr en la Wikipedia cargamos una serie de pasos con todas las dependencias que hay que tener instaladas. Pudiste instalarlas a todas correctamente? Pareciera que hay algún problema con el módulo nmap que es quien tiene la clase PorScanner.

Otra cuestión es que, me parece que no funciona con python 3. Que versión estas usando?

alevilar, 2017-07-05:
Perdón, ahí leí que pusiste la versión de python que estás usando y es correcta. Evidentemente hay algo mal instalado del nmap. 
Esto puede ser porque no instalaste el módulo desde pip install o, como dijo @gonzaabel, te faltó instalar en nmap para Windows.

En la wiki de este proyecto pusimos los pasos para instalar en Windows con todas las dependencias requeridas.

El nmap lo que hace es auto detectar las impresoras de comandas ethernet (por ahora solo marca bematech) que están en la red.

catrielr, 2017-07-05:
Estoy borrando y desinstalando todo y hacer una instalacion limpia siguendo el wiki.
Les comento mas tarde como quedo.
Gracias por la ayuda
Abrazos!

catrielr, 2017-07-05:
Borre toda instalacion de python, toda instalacion de nmap, instale todo como dice el wiki, el nmap lo instale en el c:\nmap, creo que con los nombres largos de carpetas era el tema.
Pero sigue sin listarme las impresoras.
En versiones anteriores de fiscalberry me las listaba bien.
Pego salidas de consola y archivo de configurarion:

`C:\fiscalberry>python server.py
inicializando ConectorDriverComando driver de Hasar
inicializando ConectorDriverComando driver de ReceiptDirectJet
Iniciando Fiscalberry Server
iniciando en modo HTTP
Hay 2 impresoras disponibles
  - IMPRESORA_FISCAL
      marca: Hasar, driver: Hasar
  - IMPRESORA_RED
      marca: EscP, driver: ReceiptDirectJet

*** Websocket Server Started at 169.254.249.250 port 12000***
new connection
----- - -- - - - ---
{
        "getStatus": {}
}

{u'getStatus': {}}
----- - -- - - - ---
{"getAvaliablePrinters":""}
{u'getAvaliablePrinters': u''}`

# Salida de la web de gestion:

`Mensajes
[MSG] Sending Message: {"getAvaliablePrinters":""}
[RTA] getStatus : [object Object]
[MSG] Sending Message: { "getStatus": {} }
[MSG] ***Connection Opened***`

Voy a seguir probando, si encuentro algo lo publico
Saludos 
Catriel

catrielr, 2017-07-05:
Despues de aprx 10 min recibi respuesta y listo las impresoras.
Envio una facturas y en consola tira lo siguiente:
`----- - -- - - - ---
{"printTicket":{"encabezado":{"tipo_cbte":"NCA","referencia":"00066778","nombre_cliente":"Carlos Tevez","domicilio_cliente":"zaraza","tipo_responsable":"RESPONSABLE_INSCRIPTO","tipo_doc":"CUIT","nro_doc":20407778884},"items":[{"alic_iva":21,"importe":1,"ds":"PEPSI","qty":1}]},"printerName":"IMPRESORA_RED"}
{u'printerName': u'IMPRESORA_RED', u'printTicket': {u'encabezado': {u'tipo_cbte': u'NCA', u'referencia': u'00066778', u'nro_doc': 20407778884L, u'domicilio_cliente': u'zaraza', u'tipo_doc': u'CUIT', u'nombre_cliente': u'Carlos Tevez', u'tipo_responsable': u'RESPONSABLE_INSCRIPTO'}, u'items': [{u'alic_iva': 21, u'qty': 1, u'ds': u'PEPSI', u'importe': 1}]}}
Traceback (most recent call last):
  File "C:\fiscalberry\FiscalBerryStarter.py", line 41, in on_message
    response = traductor.json_to_comando( jsonMes )
  File "C:\fiscalberry\Traductores\TraductoresHandler.py", line 99, in json_to_comando
    rta["rta"] = traductor.run( jsonTicket )
  File "C:\fiscalberry\Traductores\TraductorInterface.py", line 17, in run
    fnAction = getattr(self, action)
AttributeError: TraductorReceipt instance has no attribute 'printTicket'
`

catrielr, 2017-07-05:
Despues de aprox 5 min recibi el RTA con la lista de las impresoras. No se porque tarda tanto.
Lo mismo pasa cuando envio a imprimir un ticket, luego de ese ticket los demas salen al toke.

Saludos 
Catriel

gonzaabel, 2017-07-05:
Es normal que demore unos segundos (no debería tardar más de 15 segundos) en recibir una respuesta del getAvailablePrinters, porque con nmap primero hace un escaneo para ver en la red cuantas impresoras bematech hay y en caso de encontrar alguna te la agrega en el config.ini.
Es rarisimo que te demore demasiado, de hecho si te pasa lo mismo con el ticket y a partir del 2do te salen al instante es más extraño aun.
Mañana tratare de replicar el error haber si lo consigo.

alevilar, 2017-07-05:
Quizás se podría poner un timeout corto, seguramente el nmap tenga esa
posibilidad.

La otra alternativa es separar esa lógica y crear un nuevo comando.
Por ejemplo podríamos crear un comando que sea: *discoverDevices* Que haga
el trabajo del nmap.

Y el *getAvaliablePrinters* simplemente lista las impresoras configuradas
tal como lo hacía antes.

gonzaabel, 2017-07-10:
Hola @catrielr , hoy encontré lo que te esta ocasionando el problema.
El tema esta en la función que detecta impresoras en red, usando nmap. El error que tira la consola es el siguiente:

`----- - -- - - - ---

{"getAvaliablePrinters":""}

{u'getAvaliablePrinters': u''}

Traceback (most recent call last):

  File "/home/alevilar/fiscalberry-gonza/fiscalberry/FiscalBerryStarter.py", line 41, in on_message

    response = traductor.json_to_comando( jsonMes )

  File "/home/alevilar/fiscalberry-gonza/fiscalberry/Traductores/TraductoresHandler.py", line 116, in json_to_comando

    rta["rta"] =  self._getAvaliablePrinters()

  File "/home/alevilar/fiscalberry-gonza/fiscalberry/Traductores/TraductoresHandler.py", line 276, in _getAvaliablePrinters

    self.__getPrintersAndWriteConfig()

  File "/home/alevilar/fiscalberry-gonza/fiscalberry/Traductores/TraductoresHandler.py", line 189, in __getPrintersAndWriteConfig

    printer = self.__getDeviceData(macs)

  File "/home/alevilar/fiscalberry-gonza/fiscalberry/Traductores/TraductoresHandler.py", line 162, in __getDeviceData

    return_dict[nm[h]['vendor'][nm[h]['addresses']['mac']]+'_'+separator[index]] = {'host' : nm[h]['addresses']['ipv4'], 'marca' : 'EscP', 'driver' : 'ReceiptDirectJet'}

KeyError: 'vendor'
`
### ¿Porque ocurre esto?
Según me acuerdo al usar nmap por consola en Linux: el resultado suele ser variable.
Hay veces que no te muestra los vendor, y otras que sí. Esto causa errores en la función realizada por @wDirac.
Vamos a analizar si hay una manera (es probable) de hacer de manera forzada que muestre los vendors (hay un montón de parámetros para usar). De momento tome la decisión de comentar la llamada a mi función que a su vez llama a la de @wDirac para que Fiscalberry te funcione correctamente.
Así que hace un pull para que te bajen los cambios y avísanos si no te surge algún otro error para poder cerrar la issue.
Cualquier otro problema no dudes en abrirnos issues, saludos!

wDirac, 2017-07-10:
Se debe, a que lo devuelto por nmap no es un resultado totalmente simetrico, en ocasiones los devices, no vienen con el vendor y cositas asi, muy raro.. se incluiran unos condicionales que arreglaran el problema, gracias por reportar esto catriel

closed 2017-07-20 by gonzaabel