paxapos/fiscalberry #16: Descuentos en items de facturas

catrielr, 2017-07-12:
Hola amigos, lo que no veo en la documentacion es como informar un descuento en los Items de las facturas.

Saludos
Catriel

gonzaabel, 2017-07-13:
Hola Catriel, actualmente no se puede indicar el descuento de un item en particular.
Me asigno la issue para hacerla pronto, saludos!

alevilar, 2017-07-15:
los descuentos se podrian agregar facilmente a los comandos de fiscalberry llamando a la funcion **_addAdditional_** que ya esta dentro de los comandos actuales Hasar como Epson. Solo restaria agregar la logica desde el punto del protocolo JSON generado en fiscalberry para que el producto pueda tener un descuento o recargo.

El boceto del protocolo seria algo similar a esto:

```js
{
 "items": [
    {
        "alic_iva": 21.0,
        "importe": 22,
        "ds": "Coca Cola"
        "qty": 1.0,
       "descuento": {
              "description": "Promo Martes", // Obligatorio
              "amount": 10, // Obligatorio. En Factura "A" va sin IVA, en Factura "B" va con Iva
              "iva": 0.21, // Opcional, porcentaje del IVA. Sino toma el valor de "alic_iva" del producto
              "negative": true // Opcional, true->Descuento (valor por defecto), false->Recargo
             }
    },
    {
        "alic_iva": 21.0,
        "importe": 100,
        "ds": "Producto con recargo",
        "qty": 2.0,
        "descuento": {
              "description": "50 pe de Recargo por x motivo",
              "amount": 50,
              "negative": false // false->Recargo
             }
    }
]
}     

```

alevilar, 2017-07-15:
@gonzaabel gonza, creo que confundiste. En realidad no esta implementado aun. Lo que puse era una idea de desarrollo. Perdon si expresé mal...

Por un lado en HasarComandos y EpsonComandos ya esta implementada la funcion

Pero en el Traductor habria que agregarle la logica para que soporte el "descuento" y pueda convertir el JSON en comando de impresora fiscal basado en los parametros mencionados.

O sea, faltaria desarrollar la parte correspondiente dentro de "TraductorFiscal.py". Sio estas de acuerdo reabri la issue o como quieras!
slds

gonzaabel, 2017-07-15:
Ahh, esta bien. Pense que se podía hacer ya eso. Interprete mal tu comentario 😅 
Bien, issue re-abierta.

gonzaabel, 2017-07-21:
Esto ya esta listo y testeado. Cualquier problema avísanos! :+1:

closed 2017-07-21 by gonzaabel