paxapos/fiscalberry #17: Obtener el numero de punto de venta

catrielr, 2017-07-14:
Hola amigos, es para agregar una mejora, un comando que devuelva el punto de venta que esta configurado en la impresora fiscal, por lo que leí en hasar esta el comando GetInitData  ese da el punto de venta.
En Epson enviando el comando 2a Tipo C también se podría obtener el punto de venta.
Yo todavía no tuve tiempo de analizar el código del fiscalberry, pero si puedo agregar esto se los paso.

Es importante lo del punto de venta para guardar el numero completo del ticket Ej.: FA-0001-000001234

Saludos
Catriel

gonzaabel, 2017-07-15:
Hola Catriel, sí. Para las Hasar eso esta disponible y gracias a eso en el json solamente enviamos el número de ticket para el caso de, por ejemplo,  las notas de crédito.
Estamos enfocados a las Hasar ya que no disponemos una impresora fiscal de otra marca. Igual con leer el manual de la Epson sería una mejora fácil de implementar.
Igualmente nuestro objetivo es hacer que Fiscalberry sea lo más compatible posible (mientras más compatible, mejor para todos) por lo tanto estaremos agradecidos de recibir colaboraciones.
Si quieres colaborar te explico masomenos la estructura de Fiscalberry, el cual te podría ser fácil de aprender si te encuentras acostumbrado a trabajar con el patrón de software **MVC (Modelo, Vista, Controlador)** y si logras asociar de alguna manera la arquitectura de FB con este patrón de desarrollo.
Si conoces MVC, podrás comparar la arquitectura de FB de la siguiente manera: 

- El **Traductor** sería como un controlador, ya que salvo el _TraductoresHandler_, los demás son propios de cada tipo de impresora. Se encarga de traducir lo que el JSON contiene para que posteriormente se ejecute el/los comando/s de la clase de Comando que corresponda. 

- El **Comando** vendría a hacer el 'modelo' por así decirlo, ya que en este caso no hay base de datos. El comando contiene las acciones que podes usar en la impresora, y cada clase tiene un sendCommand que se encarga de enviar ese comando a la impresora, y la respuesta la capturara posteriormente el driver.

- El **Driver** viene a ser como la 'vista'. En el sentido que viene a ser lo que genera el resultado final. Una vez que recibe respuesta de la impresora, esta pasa a manos del server.py que te la re-envía al cliente de WS.

- También tenemos al _TraductoresHandler_ que viene a ser a su vez como el dispatcher. Ese traductor contiene todos los **comandos genéricos** que no requieren que les pases el printerName en el JSON para su uso. Pero cuando viene el printerName y el comando que le mandas a ejecutar no es genérico, es el encargado de mirar el nombre de impresora y el comando que llega por ese mismo JSON, para definir a que traductor redirigir la petición.

Bueno, esa es la forma que tuve yo de entenderlo fácilmente (quizás pude haberle errado en algún detalle) pero me ayudo muchísimo para entender Fiscalberry el asociar su estructura a la misma forma de trabajo que tengo con CakePHP. La wiki dispone de info acerca de la arquitectura, la cual te recomiendo que también la leas. 
De la wiki nos queda mejorarla bastante en las secciones sobre los comandos y también mover cosas del readme hacía ella. Y en cuanto dispongamos de tiempo nos pondremos con eso ya que tener la documentación al día es una parte vital del proyecto.
Espero haberte guiado, y ante dudas podes mirar la wiki o preguntarnos. Muchas gracias por tu interés en Fiscalberry, esperaremos tu pull request si te animas al desafió. :)
Saludos!

catrielr, 2017-07-19:
Logre que me devuelva el punto de venta, agregando el comando "getInitData"   ese trae toda la info de la impresora fiscal.
En Hasar funciona bien, en Epson tengo el problema que abri en el [https://github.com/paxapos/fiscalberry/issues/18](url)

catrielr, 2017-07-26:
Aca dejo lo que modifique para obtener el Punto de Venta. Estaria bueno agregar este comando para poder guardar el numero completo del ticket o factura emitido.
Saludos!

**EpsonComandos.py:**
Agrego en Linea: 39:
`    CMD_GET_INIT_DATA = (0x2a)`

Agrego en Linea 327:

`    def getPtoVta(self, letter):
        reply = self._sendCommand(self.CMD_GET_INIT_DATA, ["C"], True)
        if len(reply) < 3:
          # La respuesta no es válida. Vuelvo a hacer el pedido y
          # si hay algún error que se reporte como excepción
          reply = self._sendCommand(self.CMD_GET_INIT_DATA, ["C"], False)
        return reply [3]
`
**HasarCommandos.py:**
Agrego en Linea 39:
`  CMD_GET_INIT_DATA = 0x73 `

Agrego en Linea 470:
`    def getPtoVta(self, letter):
        reply = self._sendCommand(self.CMD_GET_INIT_DATA, [], True)
        if len(reply) < 3:
            # La respuesta no es válida. Vuelvo a hacer el pedido y
             # si hay algún error que se reporte como excepción
             reply = self._sendCommand(self.CMD_GET_INIT_DATA, [], False)
        return reply [6]`

**TraductorFiscal.py**

Agrego linea 40:
`	def getPtoVta(self, tipo_cbte):
		"Devuelve el punto de venta de la impresora fiscal"
		letra_cbte = tipo_cbte[-1] \
			if len(tipo_cbte) > 1 else None
		return self.comando.getPtoVta(letra_cbte)`

gonzaabel, 2017-07-26:
Hola Catrielr, gracias por tu aporte. :smiley:
Testeare tu código y lo incorporare lo más pronto que me sea posible.
Saludos!

catrielr, 2017-07-26:
Me falto agregreagar 
**ComandoInterface.py**
Linea 341
`   def getPtoVta(self, letter):
        """"Obtiene el punto de venta """
        raise NotImplementedError`

gonzaabel, 2017-12-27:
Hola Catrielr, lo siento por demorarme demasiado en ponerme a incorporar tu código.
Han pasado varios meses y Fiscalberry ha cambiado bastante. ¿Podrías actualizar tu fork de fiscalberry, incorporar nuevamente tu código y testearlo de nuevo?
Una vez hecho eso, nos generas una pull request para aprovechar las herramientas de esta plataforma y así hacer un merge más fácil.
Si necesitas ayuda para actualizar tu fork, subir tus cambios y generar una pull request, no dudes en contactarme a través de esta issue.
Gracias, saludos!