paxapos/fiscalberry #18: Problemas con Emulador Fiscar usando Epson Driver

catrielr, 2017-07-19:
Hola amigos, driver: Hasar y driver: File funciona todo de maravillas, pero cuando uso el driver: Epson tengo errores.

Esto es lo que aparece en consola:
------------------------------------
inicializando ConectorDriverComando driver de Epson
inicializando ConectorDriverComando driver de ReceiptDirectJet
Exception in thread Thread-6:
Traceback (most recent call last):
  File "C:\Python27\lib\threading.py", line 801, in __bootstrap_inner
    self.run()
  File "C:\Python27\lib\threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "C:/fiscalberry\ConectorDriverComando.py", line 23, in init_driver
    self.driver = driverClass(**kwargs)
TypeError: __init__() got an unexpected keyword argument 'path'

Iniciando Fiscalberry Server
iniciando en modo HTTP
Hay 2 impresoras disponibles
  - IMPRESORA_FISCAL
      marca: Epson, driver: Epson
  - IMPRESORA_RED
      marca: EscP, driver: ReceiptDirectJet
------------------------------------

El error esta saltando en **ConectorDriverComando.py**, cuando entra a la funcion "init_driver" y ahi me pierdo y no se que pasa.

Revisando un poco el codigo (tengo muy poco de python) encontre eso:
## En **ComandoInterface.py** linea 152:

`    def _sendCommand(self, commandNumber, parameters, skipStatusErrors=False):
        print "_sendCommand", commandNumber, parameters
        try:
            logging.getLogger().info("sendCommand: SEND|0x%x|%s|%s" % (commandNumber,
                skipStatusErrors and "T" or "F",
                                                                     str(parameters)))
            return self.conector.sendCommand(commandNumber, parameters, skipStatusErrors)
        except **epsonFiscalDriver**.ComandoException, e:
            logging.getLogger().error("epsonFiscalDriver.ComandoException: %s" % str(e))
            raise ComandoException("Error de la impresora fiscal: " + str(e))`

"Unresolved Reference" me dice mi IDE.

En **EpsonDriver.py**  Linea 3:
 `from Drivers.FiscalPrinterDriver import FiscalPrinterDriver` agregue ",ComunicationError" como esta en el HasarDriver".

No pude probar todavia en una fiscal Epson verdadera, estoy usando el emulador de http://www.impresoras-fiscales.com/emulador.htm

Saludos!
Catriel

gonzaabel, 2017-07-20:
Hola Catriel, mostranos como tenes en el config.ini, la config de la impresora fiscal Epson.
Acabo de probar y no me tira errores, y es raro que te tire "unexpected keyword path" ya que el path es requerido.

catrielr, 2017-07-20:
Huy si perdon, no copie el config.ini, ahi va:

### [SERVIDOR]
puerto = 12000

[IMPRESORA_FISCAL]
marca = Epson
modelo = tm-220-af
path = COM2
driver = Epson

[IMPRESORA_RED]
marca = EscP
host = 127.0.0.1
driver = ReceiptDirectJet

Viste lo del archivo EpsonDriver.py? si lo comparas con el HasarDriver.py en la linea 3 le falta el "ComunicationError"

Igual agregando eso, el error de path lo sigue tirando con todos los modelos de Epson, apenas arrarca el server.py.

Con este config.ini no tira el error:
[IMPRESORA_FISCAL]
marca = Epson
modelo = tm-220-af
path = respuestas.txt
driver = File

[IMPRESORA_RED]
marca = EscP
host = 127.0.0.1
driver = ReceiptDirectJet

gonzaabel, 2017-07-20:
Ya lo estoy viendo, tira error con el driver Epson, lo mirare.

gonzaabel, 2017-07-20:
Era el init del EpsonDriver el tema, fíjate ahora si podes hacer funcionar el emulador y avísame. :)

catrielr, 2017-07-20:
Exelente! solucionado!

closed 2017-07-20 by gonzaabel