paxapos/fiscalberry #19: Hasar 715v2 repite Direccion del cliente

catrielr, 2017-07-20:
Amigos, modifique el codigo del HasarComando.py porque para la 715 me estaba enviando 2 veces la direccion.

copio el codigo modificado a ver que les parece:

`    def _setCustomerData(self, name=" ", address=" ", doc=" ", docType=" ", ivaType="T"):
        # limpio el header y trailer:

        #self.setHeader()
        #self.setTrailer()
        doc = str(doc)
        doc = doc.replace("-", "").replace(".", "")
        if doc and docType != "3" and filter(lambda x: x not in string.digits, doc):
            # Si tiene letras se blanquea el DNI para evitar errores, excepto que sea
            # docType="3" (Pasaporte)
            doc, docType = " ", " "
        if not doc.strip():
            docType = " "

        if ivaType != "C" and (not doc or docType != "C" ):
            raise ValidationError("Error, si el tipo de IVA del cliente NO es consumidor final, "
                "debe ingresar su número de CUIT.")
#        parameters = [self._formatText(name, 'customerName'),
#                       doc or " ",
#                       ivaType,   # Iva Comprador
#                       docType or " ", # Tipo de Doc.
#                       address or " ",
#                       ]
#        if self.model in ["715v1", "715v2", "320"]:
#            parameters.append(self._formatText(address, 'custAddressSize') or " ") # Domicilio

        parameters = [self._formatText(name, 'customerName'),
                      doc or " ",
                      ivaType,   # Iva Comprador
                      docType or " ", # Tipo de Doc.
                      self._formatText(address,'custAddressSize') or " ", #Domicilio
                      ]
        return self._sendCommand(self.CMD_SET_CUSTOMER_DATA, parameters)
`

Saludos
Catriel

gonzaabel, 2017-07-20:
Una pregunta: de casualidad entre las dos repeticiones de la dirección, ¿una es más grande que la otra?
según estoy viendo en el setCustomerData, el tema pasa por la linea https://github.com/paxapos/fiscalberry/blob/master/Comandos/HasarComandos.py#L246 y 247 de Hasarcomandos.
Cuando estuve averiguando tiempo atrás el porque no imprimía una NC en un restaurante, era porque no se le estaba enviando la dirección del cliente. Me di cuenta no solo al ver el código, sino que al tirar prints y ver lo que me decía el manual, era porque aunque facturaras a un cliente por ejemplo: consumidor final y no tuvieses su dirección, le tenias que mandar un string vacio `address = ""` lo cual era lo que faltaba.
No preste importancia sinceramente del if, de hecho no quise sacarlo ya que aparentemente "formatea el texto" y seguro es para que en esos 3 modelos de la hasar aparezca la dirección de una forma distinta.
El tema es si es necesario que haga eso o no (por lo que veo, lo que hace la linea 247 es llamar a una función que le cambia el tamaño al texto de la dirección), habría que ver como imprime tu impresora el ticket o NC mostrando ambas direcciones (estaría bien si nos pudieses mandar una foto del mismo en esta issue). Y así vemos si es necesario que siga ese if o quitarlo, o buscar otra alternativa que de hecho la hay.
Saludos!

alevilar, 2017-07-20:
> Comandos/HasarComandos.py#L246 y 247 de Hasarcomandos.

esa linea lo que asegura es no pasarse de largo con los strings.

el largo esta definido en la linea:
https://github.com/paxapos/fiscalberry/blob/master/Comandos/HasarComandos.py#L79

y es necesario porque sino tiraria error la fiscal y no imprimiria el
ticket.

Fijate que en realidad no manda un string vacio, sino que le agrega un
espacio en blanco
" " (linea 247)

Pero es verdad que es necesario enviar la direccion cuando se mandan datos
de un cliente.

Cuando es factura "B" y el consumo es mayor a los $1000 pesos (la ley quedo
super desactualizada) es necesario enviar los datos del comprador. Aunque
sea consumidor final. En ese caso se mandaria el DNI, el numero y el
domicilio.
La verdad que muchos programadores lo que hacemos es mandarle un domicilio
vacio (espacio en blanco", yo muchas veces pongo un punto "." y le mando
eso como domicilio. Porque no da obligar a un cliente a que te de su
direccion.
Pero lo cierto es que la ley (argentina) lo exige.

@gonzaabel arregló las notas de credito. yo no las probé a fondo aun.. pero
creo que funcionaban para factura "A" y "B"

gonzaabel, 2017-07-20:
> Fijate que en realidad no manda un string vacio, sino que le agrega un
espacio en blanco
" " (linea 247)

Pero es verdad que es necesario enviar la direccion cuando se mandan datos
de un cliente.

Claro, me equivoque en decir string vació jej, lleva un espacio.

> @gonzaabel arregló las notas de credito. yo no las probé a fondo aun.. pero
creo que funcionaban para factura "A" y "B"

Si, las probaste conmigo. Funcionan hasta para factura "C" también.

catrielr, 2017-07-20:
Pero el problema que con el parameters.append como esta.. duplica el campo direccion. 
Pego salida de consola:

{"printTicket":{"encabezado":{"tipo_cbte":"FC","nro_doc":"30709211101","domicilio_cliente":"Av. Mitre 1111","tipo_doc":"CUIT","nombre_cliente":"Loco Computacion SRL","tipo_responsable":null},"items":[{"alic_iva":"21.000","importe":"12.10000000","ds":"B232-B232","qty":"1"},{"alic_iva":"21.000","importe":"12.10000000","ds":"test","qty":"1"}]},"printerName":"IMPRESORA_FISCAL"}
{u'printerName': u'IMPRESORA_FISCAL', u'printTicket': {u'encabezado': {u'tipo_cbte': u'FC', u'nro_doc': u'30709211101', u'domicilio_cliente': u'Av. Mitre 1111', u'tipo_doc': u'CUIT', u'nombre_cliente': u'Loco Computacion SRL', u'tipo_responsable': None}, u'items': [{u'alic_iva': u'21.000', u'qty': u'1', u'ds': u'B232-B232', u'importe': u'12.10000000'}, {u'alic_iva': u'21.000', u'qty': u'1', u'ds': u'test', u'importe': u'12.10000000'}]}}
*** OUTPUT Command: 98, Parameters: ['Loco Computacion SRL', '30709211101', None, 'C', u'Av. Mitre 1111', 'Av. Mitre 1111']

Ven como repite el domicilio, en el emulador da error. 
La funcion  https://github.com/paxapos/fiscalberry/blob/master/Comandos/HasarComandos.py#L187 tiene el parametro "context" ahi se le puede poner el modelo de la impresora y sacar el IF de https://github.com/paxapos/fiscalberry/blob/master/Comandos/HasarComandos.py#L224

gonzaabel, 2017-07-21:
Listo, proba ahora para ver si te imprime la dirección y avísanos si te anda bien :+1: .

catrielr, 2017-07-21:
De 10!

closed 2017-07-21 by gonzaabel

catrielr, 2017-07-25:
Ahora pasa algo mal ahora con el modelo 615, el error esta en la direccion.

Paso salida de consola.
ComandoException: Error de la impresora fiscal: Campo de datos no válido.
Comando enviado: SEND|0x62|F|['Loco Computacion SRL', '30709211101', 'C', 'C', u'Direccion']

Sale con esa "u"