paxapos/fiscalberry #21: Problemas con EpsonComandos

catrielr, 2017-07-21:
Estoy viendo que se pasa en blanco doc y typedoc
En la funcion: https://github.com/paxapos/fiscalberry/blob/master/Comandos/EpsonComandos.py#L122

cuando pasa por https://github.com/paxapos/fiscalberry/blob/master/Comandos/EpsonComandos.py#L124

Se quedan en blanco esas variables cuando va a enviar el comando da error en la impresora.

Saludos
Catriel

alevilar, 2017-08-08:
creo que esta tambien estaria funcionando ahora.. si lo podes verificar seria un golazo!

abrazo

catrielr, 2017-08-09:
Ale, cuando hago  Facturas con CUIT dan error, cuando hago una B o C sin CUIT no hace nada.
Paso los Json y las salidas de consola:

**Json Factura A:** 
 {"printTicket":{"encabezado":{"tipo_cbte":"FA","nro_doc":"30709269301","domicilio_cliente":"Av. Mitre 1111","tipo_doc":"CUIT","nombre_cliente":"Loco Computacion SRL","tipo_responsable":"RESPONSABLE_INSCRIPTO"},"items":[{"alic_iva":"21.000","importe":"1.21","ds":"B232-B232","qty":"3"},{"alic_iva":"21.000","importe":"12.1","ds":"test","qty":"3"}]},"printerName":"IMPRESORA_FISCAL"}

**Salida de Consola:**
----- - -- - - - ---
 {"printTicket":{"encabezado":{"tipo_cbte":"FB","nro_doc":"30709269301","domicilio_cliente":"Av. Mitre 1111","tipo_doc":"CUIT","nombre_cliente":"Loco Computacion SRL","tipo_responsable":"RESPONSABLE_INSCRIPTO"},"items":[{"alic_iva":"21.000","importe":"1.21","ds":"B232-B232","qty":"3"},{"alic_iva":"21.000","importe":"12.1","ds":"test","qty":"3"}]},"printerName":"IMPRESORA_FISCAL"}
{u'printerName': u'IMPRESORA_FISCAL', u'printTicket': {u'encabezado': {u'tipo_cbte': u'FB', u'nro_doc': u'30709269301', u'domicilio_cliente': u'Av. Mitre 1111', u'tipo_doc': u'CUIT', u'nombre_cliente': u'Loco Computacion SRL', u'tipo_responsable': u'RESPONSABLE_INSCRIPTO'}, u'items': [{u'alic_iva': u'21.000', u'qty': u'3', u'ds': u'B232-B232', u'importe': u'1.21'}, {u'alic_iva': u'21.000', u'qty': u'3', u'ds': u'test', u'importe': u'12.1'}]}}
zarasasaposapospoaospaosoaisa
_sendCommand 96 ['F', 'C', u'B', '1', 'P', '17', 'I', 'F', 'Loco Computacion SRL', '', '', '', 'N', 'Av. Mitre 1111', '', '', '', '', 'C']
Traceback (most recent call last):
  File "C:/fiscalberry\FiscalBerryStarter.py", line 42, in on_message
    response = traductor.json_to_comando( jsonMes )
  File "C:/fiscalberry\Traductores\TraductoresHandler.py", line 99, in json_to_comando
    rta["rta"] = traductor.run( jsonTicket )
  File "C:/fiscalberry\Traductores\TraductorInterface.py", line 24, in run
    res = fnAction( **jsonTicket[action] )
  File "C:/fiscalberry\Traductores\TraductorFiscal.py", line 54, in printTicket
    self._abrirComprobante(**encabezado)
  File "C:/fiscalberry\Traductores\TraductorFiscal.py", line 116, in _abrirComprobante
    nro_doc, doc_fiscal, pos_fiscal)
  File "C:/fiscalberry\Comandos\EpsonComandos.py", line 121, in openBillTicket
    return self._openBillCreditTicket(type, name, address, doc, docType, ivaType, isCreditNote=False)
  File "C:/fiscalberry\Comandos\EpsonComandos.py", line 185, in _openBillCreditTicket
    return self._sendCommand(self.CMD_OPEN_BILL_TICKET, parameters)
  File "C:/fiscalberry\Comandos\EpsonComandos.py", line 79, in _sendCommand
ERROR:root:PrinterException: Campo de datos no válido
    raise ComandoException("Error de la impresora fiscal: " + str(e))
ComandoException: Error de la impresora fiscal: Campo de datos no válido
_sendCommand 42 ['N']

**Json Factura B o C:** 
 {"printTicket":{"encabezado":{"tipo_cbte":"FB","nro_doc":" ","domicilio_cliente":false,"tipo_doc":"SIN_CALIFICADOR","nombre_cliente":"ConsumidorFinal","tipo_responsable":"CONSUMIDOR_FINAL"},"items":[{"alic_iva":"21.000","importe":"1.21","ds":"B232-B232","qty":"3"},{"alic_iva":"21.000","importe":"12.1","ds":"test","qty":"3"}]},"printerName":"IMPRESORA_FISCAL"}

**Salida de Consola:**
----- - -- - - - ---
 {"printTicket":{"encabezado":{"tipo_cbte":"FB","nro_doc":" ","domicilio_cliente":false,"tipo_doc":"SIN_CALIFICADOR","nombre_cliente":"ConsumidorFinal","tipo_responsable":"CONSUMIDOR_FINAL"},"items":[{"alic_iva":"21.000","importe":"1.21","ds":"B232-B232","qty":"3"},{"alic_iva":"21.000","importe":"12.1","ds":"test","qty":"3"}]},"printerName":"IMPRESORA_FISCAL"}
{u'printerName': u'IMPRESORA_FISCAL', u'printTicket': {u'encabezado': {u'tipo_cbte': u'FB', u'nro_doc': u' ', u'domicilio_cliente': False, u'tipo_doc': u'SIN_CALIFICADOR', u'nombre_cliente': u'ConsumidorFinal', u'tipo_responsable': u'CONSUMIDOR_FINAL'}, u'items': [{u'alic_iva': u'21.000', u'qty': u'3', u'ds': u'B232-B232', u'importe': u'1.21'}, {u'alic_iva': u'21.000', u'qty': u'3', u'ds': u'test', u'importe': u'12.1'}]}}
zarasasaposapospoaospaosoaisa

Lo que veo en el debug es lo siguente:

Puse un break point en: [https://github.com/paxapos/fiscalberry/blob/master/Traductores/TraductorFiscal.py#L46](url)

Ahi los parametros que se le pasan a la funcion son los siguientes:
**encabezado:** {u'tipo_cbte': u'FA', u'nro_doc': u'30709269301', u'domicilio_cliente': u'Av. Mitre 1111', u'tipo_doc': u'CUIT', u'nombre_cliente': u'Loco Computacion SRL', u'tipo_responsable': u'RESPONSABLE_INSCRIPTO'}

Pero fijate como en: [https://github.com/paxapos/fiscalberry/blob/master/Traductores/TraductorFiscal.py#L104](url)
Se cambia la de "CUIT" a "C" y despues cuando pasa por: [https://github.com/paxapos/fiscalberry/blob/master/Comandos/EpsonComandos.py#L126](url) no encuentra la "C" en "self.docTypeNames"

Si fuerzo a las variables "doc" y "docType" desde el debugger funciona OK.
Lo que no funca es con las  que no vienen con CUIT que no imprime nada.

Saludos
Catriel

catrielr, 2017-08-09:
Hola Alejandro, no puedo pasar a Open el Issue, te postie las pruebas que hice.

Abrazo

Catriel

gonzaabel, 2017-12-27:
¿Esta issue quedo resuelta?

alevilar, 2019-12-13:
si. resuelta

commit 8f76dbe8 (2017-08-08): fix #21
closed 2019-12-13 by alevilar