paxapos/fiscalberry #24: Nuevas impresoras fiscales

jeremotas, 2017-08-03:
Hola gente, antes que nada, gracias por el esfuerzo. Leyendo encontré que sólo operan con ciertos tipos de impresoras fiscales ¿Están considerando incoporar los nuevos controladores fiscales en el proyecto? ¿Cómo podemos ayudar?

Saludos!

catrielr, 2017-08-04:
Hola, aca encontre un emulador para las Hasar nuevas: http://controladoresfiscales.hasar.com/
y aca esta el manual de comandos de la epson nueva: https://files.support.epson.com/pdf/pos_latin/fiscal/mpc_lx300fplii_dionisos.pdf

Cuando pueda me pongo a ver esto

Saludos

alevilar, 2017-08-04:
> lx300fplii_dionisos.pdf

Hola!, si quieren organizamos una call por hangout, skype o como sea para
conocernos y explicarles mejor como podrian agregar nuevos drivers, marcas
y modelos al proyecto.

la estructura en si no es compleja, pero quizas con un poco de explicacion
puedan hacer algo mas rapido.

cual es tu idea geremias? que tipo, marca y modelo de impresoras quisieras
agregar?

gonzaabel, 2017-08-04:
Hola Jeremías, actualmente fiscalberry tiene compatibilidad con las Hasar y las Epson.
Actualmente de Hasar contamos con los modelos 751v1, 751v2, 615 y 320.
Nuestra idea es ir haciendo fiscalberry algo cada vez más compatible, para colaborar en la inclusión de nuevos modelos / marcas de impresora, lo ideal seria que hagas un fork de nuestro repositorio y nos envíes un Pull Request de los commits que subas. :smile: :+1: 
Agradecemos su interés, estamos felices de ver como de a poco se va sumando más gente y generando una comunidad alrededor de nuestro proyecto.
Saludos!

closed 2017-12-27 by gonzaabel