paxapos/fiscalberry #27: Soporte a Epson TM-T900F-A

AlexisCaffa, 2017-10-12:
La nueva impresora fiscal TM-T900F Argentina no tiene ningun tipo de SDK

alevilar, 2017-10-13:
Creo que la mayoria de los comandos te deberian funcionar bien. Pero quizas para este modelo tengas que agregar un par de IF´s (aunque lo dudo), el unico archivo que deberias tocar es el _[Comandos/EpsonComandos.py](https://github.com/paxapos/fiscalberry/blob/master/Comandos/EpsonComandos.py)_  ahi tenes toda la lógica de cada modelo distinto para la marca de impresoras fiscales EPSON.

Te recomiendo comenzar con un comando simple, como por ejemplo enviar un dailyClose "X" (no el Zeta, dado que el X no es fiscal, mientras que el Zeta si lo es)

Yo no tengo esa impresora como para hacer pruebas, pero te puedo ayudar, andá describiendo acá los errores con los que te vas chocando y trataré ayudarte lo mas pronto que pueda.

AlexisCaffa, 2017-10-13:
Ejecuté dailyClose, junto con otros..

Esto se ve en el navegador [screenshot](https://i.snag.gy/IhDe5t.jpg).

Y la consola del navegador:
```sh
START INICIANDO
No hay conexion con el Web Socket promise ERROR
***Connection Closed***
 reconectando en 3 segundos 
CLOSE
START INICIANDO
***Connection Opened***
OPEN CONECTADO
Sending Message: {"getAvaliablePrinters":""}
Vino algo del websocket %o
vino la respuesta %o
getAvaliablePrinters : IMPRESORA_FISCAL
Vino RTA DE  getAvaliablePrinters: %o
Sending Message: {"getStatus":{},"printerName":"IMPRESORA_FISCAL"}
Vino algo del websocket %o
AttributeError("TraductorFiscal instance has no attribute 'getStatus'",)- TraductorFiscal instance has no attribute 'getStatus'
Sending Message: {"getLastNumber":"T","printerName":"IMPRESORA_FISCAL"}
Vino algo del websocket %o
vino la respuesta %o
getLastNumber : 365
Sending Message: {"dailyClose":"X","printerName":"IMPRESORA_FISCAL"}
Vino algo del websocket %o
vino la respuesta %o
dailyClose : 9928,9928,9928,9928,9928,9928,9928,9928
Sending Message: {"dailyClose":"X","printerName":"IMPRESORA_FISCAL"}
Vino algo del websocket %o
vino la respuesta %o
dailyClose : 184,184,184,184,184,184,184,184
```
Y esto arroja el server.py
```sh
Iniciando Fiscalberry Server
La IP es 192.168.0.2
Impresora disponible:
  - IMPRESORA_FISCAL
      marca: Epson, driver: File

*** Websocket Server Started as HTTP at 127.0.1.1 port 12000***
new connection
----- - -- - - - ---
{"getAvaliablePrinters":""}
{u'getAvaliablePrinters': u''}
----- - -- - - - ---
{"getStatus":{},"printerName":"IMPRESORA_FISCAL"}
{u'getStatus': {}, u'printerName': u'IMPRESORA_FISCAL'}
inicializando ConectorDriverComando driver de File
ERROR:root:AttributeError("TraductorFiscal instance has no attribute 'getStatus'",)- TraductorFiscal instance has no attribute 'getStatus'
Traceback (most recent call last):
  File "/home/jj/Proyectos/_Repositories/fiscalberry/FiscalberryApp.py", line 59, in on_message
    response = traductor.json_to_comando(jsonMes)
  File "/home/jj/Proyectos/_Repositories/fiscalberry/Traductores/TraductoresHandler.py", line 64, in json_to_comando
    rta["rta"] = traductor.run(jsonTicket)
  File "/home/jj/Proyectos/_Repositories/fiscalberry/Traductores/TraductorInterface.py", line 13, in run
    fnAction = getattr(self, action)
AttributeError: TraductorFiscal instance has no attribute 'getStatus'
----- - -- - - - ---
{"getLastNumber":"T","printerName":"IMPRESORA_FISCAL"}
{u'getLastNumber': u'T', u'printerName': u'IMPRESORA_FISCAL'}
inicializando ConectorDriverComando driver de File
_sendCommand 42 ['A']
*** OUTPUT Command: 42, Parameters: ['A']

----- - -- - - - ---
{"dailyClose":"X","printerName":"IMPRESORA_FISCAL"}
{u'printerName': u'IMPRESORA_FISCAL', u'dailyClose': u'X'}
inicializando ConectorDriverComando driver de File
_sendCommand 68 ['Cancelar', '0', 'C']
*** OUTPUT Command: 68, Parameters: ['Cancelar', '0', 'C']

_sendCommand 57 [u'X', 'P']
*** OUTPUT Command: 57, Parameters: [u'X', 'P']

----- - -- - - - ---
{"dailyClose":"X","printerName":"IMPRESORA_FISCAL"}
{u'printerName': u'IMPRESORA_FISCAL', u'dailyClose': u'X'}
inicializando ConectorDriverComando driver de File
_sendCommand 68 ['Cancelar', '0', 'C']
*** OUTPUT Command: 68, Parameters: ['Cancelar', '0', 'C']

_sendCommand 57 [u'X', 'P']
*** OUTPUT Command: 57, Parameters: [u'X', 'P']
```

Pareciera que está todo bien. Preguntas:

1. ¿La impresora debería moverse o algo? :laughing: 
2. ¿Cómo hago para imprimir un ticket sin validez fiscal o algo para testear?

AlexisCaffa, 2017-10-13:
Acá el branch con los cambios [epson-tmt900fa](https://github.com/AlexisCaffa/fiscalberry/tree/epson-tmt900fa)...

:smile: Perdón por las mil vueltas que le dí pero hice varios cambios:
* A medida que iba haciendo el set-up me iban surgiendo problemas y los fuí actualizando en el _readme.md_
    * Copié un script que lista los ttyUSB
    * Optimicé un poco la instalación de dependencias y agregué otras necesarias.
    * Creé una alternativa "dev" con virtualenvs y requirements.txt para que sea más práctico.
* También estandaricé un poco el código en general:
    * En el caso de python de acuerdo a las PEP8 (y falta aún :stuck_out_tongue_closed_eyes: )
    * En el caso de javascript y html normalicé las tabulaciones
    * El readme lo acomodé para que se vea un poquito más bonito no más

gonzaabel, 2017-10-13:
1. ¿La impresora debería moverse o algo? :laughing:

Si esta bien configurada, debería imprimir al enviarle un comando si la impresora esta preparada y habilitada para emitir comprobantes. 

2. ¿Cómo hago para imprimir un ticket sin validez fiscal o algo para testear?

Podes tirar un informe X que al contrario del Z, podes tirar todos los que quieras y no va a pasar nada :smile:

gonzaabel, 2017-10-13:
Si queres pasame por acá como tenes el archivo config.ini. Así veo si esta todo ok en la configuración

AlexisCaffa, 2017-10-13:
config.ini :
```
[SERVIDOR]
puerto = 12000
ip_privada = 192.168.0.2

[IMPRESORA_FISCAL]
marca = Epson
modelo = tm-t900fa
path = /dev/ttyUSB0
driver = File
```
donde `tm-t900fa` es lo mismo que `tm-220-af` para el branch [epson-tmt900fa](https://github.com/AlexisCaffa/fiscalberry/tree/epson-tmt900fa) por el momento

gonzaabel, 2017-10-13:
El config esta bien. También añado por si acaso, que para imprimir no podrás usar el driver File, deberás usar el driver Epson.
El driver File es también otra utilidad que hay para los casos de prueba el cual va escribiendo las salidas de los comandos en un txt (donde apunta el path, el cual en estos casos podría ir en /tmp/respuestas.txt).

gonzaabel, 2017-12-27:
Hola Alexis:
¿Has logrado imprimir con tu impresora Epson?

AlexisCaffa, 2017-12-27:
Hola Gonza, no.. :disappointed: Y ya no tenemos más esa impresora para testear..

diegopunk27, 2018-03-27:
Hola, alguien siguió desarrollando algo para esta impresora ( epson-tmt900fa )??

diegopunk27, 2018-03-28:
Hola gonzaabel, desde ya gracias por contestar... Te comento que con el servidor python de paxapos me tira un error cuando lo lanzo [error](https://snag.gy/yoGN3x.jpg)... por lo tanto estuve probando el fork de AlexisCaffa, y si pude lanzar el servidor python... 

Luego con cualquiera de los dos me tira error a la hora de imprimir (Utilizando windows 10), no me reconoce el puerto USB... Cómo configuras USB en el config para windows?? Probé con "USB0", "USB1", con el 0 también, porque se que es el codigo USB para estas impresoras.

alevilar, 2018-03-28:
En Windows me parece que tenes que instalar un driver virtualizador de
puerto usb para que te aparezca como un COM1,COM2, pero con el número que
le corresponda.

En hasar se que te lo podes descargar desde la web. Con epson no se. Quizás
el mismo funcione

diegopunk27, 2018-03-28:
Perdon, así estuve configurando el archivo [config.ini](https://snag.gy/MHavw5.jpg).
Luego este error me genera ("[ERR] SerialException("could not open port 'USB4': WindowsError(2, 'El sistema no puede encontrar el archivo especificado.')")....

@alevilar gracias lo pruebo y te aviso!

diegopunk27, 2018-03-28:
@alevilar estuve utilizando el "EPSON TM Virtual Port Driver Port Assignment Tool" y le asigne un puerto COM virtual, ya no da error... El tema es que no imprime, ni tampoco me genera una respuesta ante los comandos...

Aqui el [log de los mensajes](https://snag.gy/jID1a7.jpg)

gonzaabel, 2018-03-28:
Hola Diego, para usar la versión master de Fiscalberry, por lo que nos muestra el primer error, es que te falta la dependencia jwt la cual se instala con `pip install PyJWT`. (Recién acabo de incluirla en la wiki).
Así que ya con esa dependencia instalada, no tendrías que tener más problemas para usar la versión actual de Fiscalberry y no la de Alexis que ya quedo desactualizada.

gonzaabel, 2018-03-28:
Lo que no genere una respuesta en los comandos puede ser por la versión desactualizada de Alexis o que sean errores en concreto que únicamente se visualicen por consola.

diegopunk27, 2018-03-28:
@gonzaabel , listo quedó funcionando el servidor [así](https://snag.gy/5W1OKd.jpg). Pero sigue sin imprimir nada... Vos probaste que funcionara para este modelo???

joseluisbossio, 2018-03-28:
Le diste permisos al /dev/ttyUSBx de 666 y que sea del grupo dialout y que
tú usuario pertenezca al grupo?

gonzaabel, 2018-03-28:
@diegopunk27  no tengo forma de testearla :cry:  manda a imprimir un cierre X y muéstranos que te tira por consola.
Hay dos cosas que puedan pasar: la 1ra es que no haya una correcta comunicación con la impresora (error con el puerto USBx) o que la impresora devuelva una respuesta con estado de error (que te lo debería mostrar por consola).

@joseluisbossio el no esta usando Fiscalberry en Linux, lo usa con Windows jeje.

diegopunk27, 2018-03-28:
@gonzaabel ésta mañana cuando estuve probando, escribía los comandos pero no imprimía, ni se visualizaba nada en pantalla... luego de varios intentos comenzaba a fallar mostrando este [error](https://snag.gy/SboXpD.jpg)... Ahora estoy lanzando pruebas de nuevo, y directamente me muestra ese error, me imagino que tiene que ver con el virtual port driver.
Lo que es extraño es que sigo utilizando el virtualizador de puerto, y con la interfaz SATF (copia de [pantalla)](https://snag.gy/Qbq7XR.jpg) me imprime correctamente los tikets (Tengo a la impresora en modo de prueba)...

@joseluisbossio si sigo renegando así voy a probar con ubuntu

joseluisbossio, 2018-03-28:
Instalate un Ubuntu desktop 16.04 para probar.

gonzaabel, 2018-03-29:
Estas ejecutando Fiscalberry con permisos de administrador?? [acá](https://files.support.epson.com/pdf/pos/bulk/tmt900fa_pc_manual_revc.pdf) encontre el manual de desarrollo de esta impresora, mañana en cuanto tenga tiempo le echaré un vistazo para ver si puedo sacar el porque no te funciona (aunque puede que te este fallando el virtualizador de puertos)
Acá te dejo para que te descargues el com0com, virtualizador que use para la impresora Hasar de 2daGeneración. Es otra cosa que podes probar para ver si con eso llegase a funcionar jeje...

Descarga la versión según tu SO, si es 32 bits: https://mega.nz/#!bkIAGbLJ!f-xt9qFVSCgfpa_jrS24lbykw3CrdAqwA1o_1PN8LkI

Y si es 64bits: https://mega.nz/#!SgxlQIhJ!IgCnFoqEp3vmHd-tP5ErBNZ7w9LiWBmkRSAnpT1DtQY

Y si aún así no funciona... probaría como dice Jose, te instalas un linux y ahí lo tenes que sacar andando si o si jajaja. Porque el problema en Windows es que se manejan los dispositivos de distinta manera que con Linux, pero yo con ese virtualizador pude hacer andar el emulador de Hasar, ponele un número de puerto COM30 por ejemplo y capaz te sale andando.

Saludos!

joseluisbossio, 2019-02-15:
Buenas, al final ¿que paso con la TM-T900FA? tengo que migrar de TM-U220AF a una de 2da generación en EPSON, pero no quiero ir a un camino sin salida. :)

nicociri, 2019-04-25:
@diegopunk27 me engancho de este thread para preguntarte como conseguiste el satf para la t900? Podrias pasarmelo? Estoy rebuscando toda la web a ver si alguno lo postea pero no tengo suerte.

Tengo mi primer t900 aca y ni mi programa ni fiscalberry logro que conecten. Y me pregunto si no es este cablecito adaptador serial->usb

diegopunk27, 2019-06-13:
@nicociri disculpas por la tardanza, hace mucho no revisaba el mail. Te comento que por este hilo no pude resolver el problema... En su momento encontré una dll llamada EpsonFiscalInterface, ahí están todas las funciones necesarias para manejar la fiscal. A partir de ahí desarrollé una API en python (+Tornado) partiendo de un ejemplo que la misma gente de epson distribuye en "Epson Fiscal Interface". El ejemplo lo podes descargar desde aqui, entre otras cosas [Link](https://epson.com.ar/Soporte/Punto-de-venta/Impresoras-fiscales/Epson-TM-T900FA/s/SPT_C31CB76402?review-filter=Windows+8+32-bit). La verdad que me hubiese gustado implementarlo sin un spooler, pero no tenía más tiempo para seguir investigando. Espero que te sirva la info, sino estamos en contacto....

alevilar, 2019-12-13:
en la ultima version de fiscalberry ya esta funcionando la Epson TM900FA por medio de la libreria oficial de Epson llamada EpsonFiscalInterface que se descarga de la pagina oficial.
Es de la release 0.9 https://github.com/paxapos/fiscalberry/releases/tag/0.9

No es la mejor solucion. pero es algo que nos permite usar esa impresora. En windows aun no esta del todo probado.

y el soporte es parcial, por ejemplo al momento no podemos imprimir descuentos ni retenciones. 

Pero lo mas importante del ticket ya puede ser impreso.

closed 2019-12-13 by alevilar

noeliapinto, 2020-01-07:
Hola! Alguno pudo lograr la impresión de DNFH con este modelo de impresora?

FerPortoFar, 2023-08-18:
Hola tengo un problema quiero imprimir un ticket fiscal en una Epson tm900 y quiero colocar la descripción del producto en dos líneas y no me lo permite cuando coloco el  carry return me da un error indicándome que el carácter de el ítems es invalido, que podría hacer??