paxapos/fiscalberry #32: No logro que la impresora Epson U220AFII imprima

MarianoGomezBidondo, 2017-12-11:
Hola!
Primero felicitarte y agradecerte por el trabajo que hiciste, es muy completo.
Tengo todo instalado en un raspberry pi 3. El servidor funciona y si lo configuro para imprimir  a un archivo veo los comandos.
Pero cuando utilizando por nombre de impresora la que tengo configurada hacia la Epson, me da el error "Inappropriate ioctl for device".
La impresora la probé en windows y funciona OK. 
Desde un equipo mando por websocket el json y en el raspberry veo este error que te pego:

Te agradesco tu ayuda, probe de todo pero no logro encontrarle la vuelta. 
Saludos!
**Configuracion** 
[IMPRESORA_FISCAL]
marca = Epson
modelo = tickeadoras
path = /dev/usb/lp0
driver = Epson

**Error completo:**
"ERROR:root:SerialException("Could not configure port: (25, 'Inappropriate ioctl for device')",)- Could not configure port: (25, 'Inappropriate ioctl for device')
Traceback (most recent call last):
  File "/home/pi/Downloads/fiscalberry-server-rc/FiscalberryApp.py", line 59, in on_message
    response = traductor.json_to_comando(jsonMes)
  File "/home/pi/Downloads/fiscalberry-server-rc/Traductores/TraductoresHandler.py", line 60, in json_to_comando
    traductor = self.__init_printer_traductor(printerName)
  File "/home/pi/Downloads/fiscalberry-server-rc/Traductores/TraductoresHandler.py", line 180, in __init_printer_traductor
    comando = comandoClass(**dictSectionConf)
  File "/home/pi/Downloads/fiscalberry-server-rc/ComandoInterface.py", line 144, in __init__
    self.conector = ConectorDriverComando(self, driver, **kwargs)
  File "/home/pi/Downloads/fiscalberry-server-rc/ConectorDriverComando.py", line 23, in __init__
    self.driver = driverClass(**kwargs)
  File "/home/pi/Downloads/fiscalberry-server-rc/Drivers/FiscalPrinterDriver.py", line 53, in __init__
    self._serialPort = serial.Serial(port=path, timeout=None, baudrate=speed)
  File "/usr/lib/python2.7/dist-packages/serial/serialutil.py", line 236, in __init__
    self.open()
  File "/usr/lib/python2.7/dist-packages/serial/serialposix.py", line 272, in open
    self._reconfigure_port(force_update=True)
  File "/usr/lib/python2.7/dist-packages/serial/serialposix.py", line 315, in _reconfigure_port
    raise SerialException("Could not configure port: {}".format(msg))
SerialException: Could not configure port: (25, 'Inappropriate ioctl for device')"

gonzaabel, 2017-12-18:
Hola, para mi hay un problema en el path.
Estas seguro que es el correcto? va conectada por serial o por usb la impresora? (pregunto por la 'SerialException' que tira). Normalmente al ser usb debería el path ser /dev/ttyUSB0, o /dev/ttyS0 si es serial.
Y otra pregunta: estas ejecutando el servidor con permisos de super usuario?

gonzaabel, 2017-12-18:
Perdón, no había leído tu más reciente issue.
Quizá los puertos usb de la Raspberry no estén funcionando bien. Échale un vistazo a esta guía, puede que te sea de ayuda: https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=53832
Como también decirnos lo que te salga de consola al tirar un `lsusb`. Como te pide Ale Vilar en la issue 
#34 :nerd_face:

MarianoGomezBidondo, 2017-12-21:
Problema solucionado, ver en mi otro post.

closed 2017-12-21 by MarianoGomezBidondo