paxapos/fiscalberry #33: Problemas c/windows

catrielr, 2017-12-17:
Pruebas realizada con Windows 10 x64. Python 2.7.13

Este el es el error de la consola:

C:\Python27\python.exe C:/fiscalberry/server.py
Traceback (most recent call last):
  File "C:/fiscalberry/server.py", line 5, in <module>
    from FiscalberryApp import FiscalberryApp
  File "C:\fiscalberry\FiscalberryApp.py", line 18, in <module>
    from signal import signal, SIGPIPE, SIG_DFL, SIGTERM, SIGINT
ImportError: cannot import name SIGPIPE

Process finished with exit code 1

Saludos
Catriel

gonzaabel, 2017-12-18:
Hola Catriel, según lo poco que googlie, aparentemente SIGPIPE no estaría disponible en Windows.
Proba quitando SIGPIPE del import, si te tira error con las siguientes 3 importaciones (SIG_DFL, SIGTERM, SIGINT) entonces seguro se deba a que te falta la librería de signal por instalar para python.
Acá te dejo el link de una issue que encontre en otro proyecto acerca de ese error de importación: https://github.com/CaliDog/certstream-python/issues/3

Saludos!

catrielr, 2017-12-18:
Solucione el problea asi: 
#from signal import signal, SIGPIPE, SIG_DFL, SIGTERM, SIGINT
#signal(SIGPIPE,SIG_DFL)
from signal import signal, SIG_DFL, SIGTERM, SIGINT

Otra cosa que me paso es el encoding del archivo logging.ini tenia LF de UNIX y lo pase Windows LF 
Saludos
Catriel

gonzaabel, 2017-12-19:
Genial, me alegra que lo hayas podido solucionar!
Entonces según esto, SIGPIPE no esta disponible en Windows definitivamente.
Saludos.

closed 2018-02-19 by gonzaabel