paxapos/fiscalberry #34: En RaspberryPi 3 no veo /dev/ttyUSB, solo /dev/usb/pl0 pero no funciona

MarianoGomezBidondo, 2017-12-17:
Hola! perdona que insista, En RaspberryPi 3 no veo /dev/ttyUSB, solo /dev/usb/pl0 pero no funciona.... estoy conectandola con un cable USB al RPi3 y mini USB a una Epson 220 AFII. 
Funciona por el puerto USB de la impresora o es por medio de la entrada Serial de la impresora si o si?

Gracias y saludos!!

alevilar, 2017-12-18:
esa impresora no la probé nunca.
ejecuta el comando lsusb y copianos el resultado de la consola a ver si
encontramos con que ayudarte

MarianoGomezBidondo, 2017-12-18:
Hola! Primero, mil gracias por la ayuda, linux no es mi fuerte.

Resumo: Es una EMPSON 220 AF II conectada por cable USB (USB nomal al raspberry, mini usb a la impresora. Esa impresora tiene salida micro USB y RJ45 pero formato serial).
Pregunta, las impresoras suelen conectarlas a conexiones seriales?

Al agregarla lo unico que aparece nuevo en "ls /dev/" es lo que ya comente "/dev/usb/lp0". No aparece nada de /dev/ttyUSB0, o /dev/ttyS0.
Nota: desde algun programa de prueba en windows pude imprimir un test del sistema... ahora voy a probar de imprimir un comprobante, pero la impresora en si, (aparentemente) funciona bien.

Alguna idea? Muchas gracias
Les pego la salida de un par de comandos:

**pi@raspberrypi:~/Downloads/FiscalBerry/fiscalberry $ lsusb**
Bus 001 Device 006: ID 04b8:0202 Seiko Epson Corp. Receipt Printer M129C/TM-T70
Bus 001 Device 005: ID 0458:003a KYE Systems Corp. (Mouse Systems) NetScroll+ Mini Traveler / Genius NetScroll 120
Bus 001 Device 004: ID 04d9:1503 Holtek Semiconductor, Inc. Keyboard
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. SMC9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

> (es el device 006)

**pi@raspberrypi:~/Downloads/FiscalBerry/fiscalberry $ dmesg**
[ 2779.180311] usb 1-1.5: new full-speed USB device number 8 using dwc_otg
[ 2779.313363] usb 1-1.5: New USB device found, idVendor=04b8, idProduct=0202
[ 2779.313378] usb 1-1.5: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 2779.313386] usb 1-1.5: Product: FISCAL
[ 2779.313394] usb 1-1.5: Manufacturer: EPSON
[ 2779.317362] usblp 1-1.5:1.0: usblp0: USB Bidirectional printer dev 8 if 0 alt 0 proto 2 vid 0x04B8 pid 0x0202

alevilar, 2017-12-18:
mandanos a ver cmo tenes tu conig.ini

alevilar, 2017-12-18:
perdon, escribi mal. Mandanos una copia de tu *config.ini* a ver como lo
tenes configurado

MarianoGomezBidondo, 2017-12-18:
Configure varias, utilizo la que se llama "IMPRESORA_FISCAL"

pi@raspberrypi:~/Downloads/FiscalBerry/fiscalberry $ cat config.ini
[SERVIDOR]
puerto = 12000
ip_privada = 192.168.0.14

**[IMPRESORA_FISCAL]
marca = Epson
modelo = tickeadoras
path = /dev/usb/lp0
driver = Epson**

[IMPRESORA_USB]
marca = Epson
modelo = tickeadoras
path = /dev/usb/lp0
driver = ReceiptUSB
> esto lo probe despues de mirar tu ultimo update

[IMPRESORA_ARCHIVO]
marca = Epson
modelo = tickeadoras
driver = File
path = /tmp/archivo.txt

[IMPRESORA_RED]
marca = EscP
host = 127.0.0.1
driver = ReceiptDirectJet

MarianoGomezBidondo, 2017-12-20:
Consulta: Con las impresoras que probaste, tenian entrada RS232 o USB?
La epson tiene entrada RS232 pero con una ficha RJ45 (cosa rarisima!), podria probar de un cable USB-RS232 y armarle una ficha a rj45.... a ver si el tema es ese. Te parece?

MarianoGomezBidondo, 2017-12-21:
Problema Solucionado!
La historia es asi. Estaba probando por el puerto USB de la EPSON TM-U220 AF II y aunque el raspberry me reconocia el dispositvo USB, no se puede tratar como si fuera un dispositivo serie.

Le puse un cable conversor USB-RS232 (esos que tienen el CHIP 340) y luego un cable adaptador de conector RS232 a RJ45, que conecte al RJ45 de la EPSON. Ahi me aparecio el dispositivo en /dev/ttyUSB0 y ya todo funciono perfecto.
Estaria bueno que el server tenga soporte del puerto USB de la impresora EPSON.
Mas alla de eso, muy bueno el proyecto. Saludos!