paxapos/fiscalberry #35: Les funciona con Notas de Credito ?

catrielr, 2017-12-18:
Ya en las versiones anteriores tampoco me funciono, y como estaba con otras cosas no le di mucha bola, pero ahora vuelvo a retomar el tema.

Pego la salida de la consola con Impresora Epson LX-300F 
_sendCommand 96 ['M', 'C', u'A', '1', 'P', '17', 'E', 'F', 'Carlos Tevez', '', '-', '-', 'N', 'zaraza', '', '', '-', '', 'C']
2017-12-18 19:44:01,930 - root - INFO - sendCommand: SEND|0x60|F|['M', 'C', u'A', '1', 'P', '17', 'E', 'F', 'Carlos Tevez', '', '-', '-', 'N', 'zaraza', '', '', '-', '', 'C']
2017-12-18 19:44:01,930 - root - ERROR - PrinterException: Campo de datos no válido
2017-12-18 19:44:01,930 - FiscalberryApp - ERROR - ComandoException('Error de la impresora fiscal: Campo de datos no v\xc3\xa1lido',)- Error de la impresora fiscal: Campo de datos no válido
Traceback (most recent call last):
  File "C:/fiscalberry\FiscalberryApp.py", line 69, in on_message
    response = traductor.json_to_comando(jsonMes)
  File "C:/fiscalberry\Traductores\TraductoresHandler.py", line 64, in json_to_comando
    rta["rta"] = traductor.run(jsonTicket)
  File "C:/fiscalberry\Traductores\TraductorInterface.py", line 20, in run
    res = fnAction(**jsonTicket[action])
  File "C:/fiscalberry\Traductores\TraductorFiscal.py", line 47, in printTicket
    self._abrirComprobante(**encabezado)
  File "C:/fiscalberry\Traductores\TraductorFiscal.py", line 110, in _abrirComprobante
    pos_fiscal, referencia)
  File "C:/fiscalberry\Comandos\EpsonComandos.py", line 112, in openBillCreditTicket
    return self._openBillCreditTicket(type, name, address, doc, docType, ivaType, isCreditNote=True)
  File "C:/fiscalberry\Comandos\EpsonComandos.py", line 179, in _openBillCreditTicket
    return self._sendCommand(self.CMD_OPEN_BILL_TICKET, parameters)
  File "C:/fiscalberry\Comandos\EpsonComandos.py", line 76, in _sendCommand
    raise ComandoException("Error de la impresora fiscal: " + str(e))
ComandoException: Error de la impresora fiscal: Campo de datos no válido

----------------------------------------
Pego el config.ini
[SERVIDOR]
puerto = 12000
ip_privada = 10.0.0.13

[IMPRESORA_FISCAL]
marca = Epson
modelo = epsonlx300
path = COM2
driver = Epson

[IMPRESORA_RED]
marca = EscP
host = 127.0.0.1
driver = ReceiptDirectJet

---------------------------------------------
Pego salida de Consola con impresora Hasar 715v2

2017-12-18 19:55:31,009 - root - INFO - sendCommand: SEND|0x62|F|['Carlos Tevez', '20407778884', 'I', 'C', 'zaraza']
2017-12-18 19:55:31,009 - root - INFO - reply: ['0080', '0600']
2017-12-18 19:55:31,009 - root - INFO - sendCommand: SEND|0x93|F|['1', '00066778']
2017-12-18 19:55:31,009 - root - INFO - reply: ['0080', '0600']
2017-12-18 19:55:31,009 - root - INFO - sendCommand: SEND|0x80|F|['R', 'T', '00066778']
2017-12-18 19:55:31,026 - root - ERROR - PrinterException: Campo de datos no válido
2017-12-18 19:55:31,026 - FiscalberryApp - ERROR - NameError("global name 'ComandoException' is not defined",)- global name 'ComandoException' is not defined
Traceback (most recent call last):
  File "C:/fiscalberry\FiscalberryApp.py", line 69, in on_message
    response = traductor.json_to_comando(jsonMes)
  File "C:/fiscalberry\Traductores\TraductoresHandler.py", line 64, in json_to_comando
    rta["rta"] = traductor.run(jsonTicket)
  File "C:/fiscalberry\Traductores\TraductorInterface.py", line 20, in run
    res = fnAction(**jsonTicket[action])
  File "C:/fiscalberry\Traductores\TraductorFiscal.py", line 47, in printTicket
    self._abrirComprobante(**encabezado)
  File "C:/fiscalberry\Traductores\TraductorFiscal.py", line 110, in _abrirComprobante
    pos_fiscal, referencia)
  File "C:/fiscalberry\Comandos\HasarComandos.py", line 272, in openBillCreditTicket
    self._sendCommand(self.CMD_OPEN_CREDIT_NOTE, [type, "T", reference])
  File "C:/fiscalberry\Comandos\HasarComandos.py", line 144, in _sendCommand
    raise ComandoException("Error de la impresora fiscal: %s.\nComando enviado: %s" % \
NameError: global name 'ComandoException' is not defined

gonzaabel, 2017-12-19:
Hola Catriel, en las Hasar te puedo confirmar que funcionan.
Para mi ese error que te esta tirando, es debido a que el número del comprobante es inexistente.
Deberías imprimir un ticket de 1 peso a consumidor final y luego mandar el siguiente JSON:

```JSON
				"printTicket": {
					"encabezado": {
						"tipo_cbte": "NCB",
						"referencia": "00000057" // Número de comprobante, tipo string.
					},
					"items": [{ // El/Los item/s  del ticket a anular tal cual se muestra debajo.
						"alic_iva": 21.0,
						"importe": 0.01,
						"ds": "PEPSI",
						"qty": 1.0
					}, {
						"alic_iva": 21.0,
						"importe": 0.12,
						"ds": "COCA",
						"qty": 2.0
					}]
				}
```
Y con eso te debería funcionar si o si. 

En las Epson no lo hemos podido probar, pero según el error que te tira con ella, habría que leer la docu para desarrolladores y mandar los datos tal como piden que se los mandemos. 😸

gonzaabel, 2017-12-27:
Hola Catrielr, acabo de probar con un ticket A y pude imprimir una NCA.
Así que funcionan. Use una Hasar 441F.
Ahora, en Epson es probable que aun no. No tenemos con que probarlas :disappointed:

alevilar, 2018-01-10:
Catriel, tenes posibilidad de probarlo con otro modelo de impresora? por ejemplo la TM 220? 
creo que es una pavada el error, pero depende del modelo de la impresora

pushnube, 2019-03-22:
Hola gente, hace un tiempo vengo utilizando fiscalberry y ahora surgio la necesidad de solucionar lo de las notas de Crédito en la TM220U. La cosa es que no encontre un fix de esto y me puse a probar ya cuento con la impresora. el tema no esta en codigo fuente sino en el armado del json. Como puedo colaborar para pasarles esto correctamente? Seria mi primer aporte y no se muy bien como funciona github.

alevilar, 2019-03-22:
bien! te ayudaremos. comentame mejor hasta donde llegaste, que error te
tira? mandanos algo de la consola a ver si vemos por donde meter mano

joseluisbossio, 2019-03-22:
Si hay que probar algo yo en la semana tengo EPSON TMU-220 sin inicializar.