paxapos/fiscalberry #36: Consulta Estado

catrielr, 2018-01-02:
No puedo precisar en que version empezo esto, pero lo que veo es que no llegan mas consulta de estado a la impresora. (si esta sin papel, o tapa abierta).
En versiones anterioes veia que cada minuto aproximandamente habia una consulta de estado.

Saludos 
Catriel

gonzaabel, 2018-01-02:
Hola Catrielr.
Esas advertencias siguen estando, pero solo se disparan cuando la impresora lanza esos estados.
Ya no funciona cuando se hace uso de drivers como el File.
Así que, debería seguir andando en cuanto ocurran esos eventos con una impresora de verdad. Caso contrario es un bug a resolver.
Saludos!

alevilar, 2018-02-19:
en un nuevo branch estamos resolviendo este problema. Gracias por reportarlo @catrielr