paxapos/fiscalberry #44: Se cuelga el servicio ante situaciones de error

MarianoGomezBidondo, 2018-01-29:
Cuando por algun motivo hay error (la impresor no responde, se desconecto, el comando es incorrecto, o lo que fuera) tira el error en el servidor y el servicio deja de responder; para que continue hay que cerrarlo y volver a ejecutar el server.py.
Hay posibilidad de que retorne el error y luego continue esperando comandos?

Saludos!

alevilar, 2018-01-29:
En realidad ese es el comportamiento. Retornar los errores pero seguir
funcionando.
Seguramente estés teniendo un error que no hayamos detectado. Podrías
copiarnos todo lo que te aparece en consola? Saludos

alevilar, 2018-02-15:
@MarianoGomezBidondo podrias enviarnos el error asi te ayudamos con esto!, saludos

closed 2019-10-28 by alevilar