paxapos/fiscalberry #46: [feat] evolucion: driver basic para kit fiscal de venezuela

mckaygerhard, 2018-01-31:
Llevar el fiscalberry a un producto de la talla del https://www.odoo.com/page/point-of-sale-hardware#part_3 POSBOX una caja ARM que centraliza el hardware (impresoras) y envia las ordenes.. ya que el piunto de venta de ambos proyectos no es desktop sino web

En este capitulo es implementar un driver  fiscalberry/Drivers/EpsonDriver.py que domine el kit fiscal de Venezuela, esto implica algunas cosas que llevan a la evolucion:

1. los drivers no pueden tener el nombre de la impresora simplemente, esto complica el desarrollo, ya que cada pais tiene su propio kit, en el caso de Venezuela para Epson hay un solo proveedor el cual ya toda comunicacion yo la se, pero no podemos meter todo el codigo en un solo archivo que domine todas las Epson, 
2. ademas el kit de las Epson de Vnzla es particular porque rescata las facturas y es capaz de prosegir y guardar las transacciones, es practicamente un minipos, no siendo asi para las Bixolon/Samsung pero para estas no hay implementacion en linux, lo que acorta su implementacion en el fiscalberry, se tiene pendiente un manual de bajo nivel
2. el lenguaje es python, un tanto dificil de aprender (de aprender avanzado para trabajar con hardware, muy distinto a la curva de avance de BASIC y GAMBAS) se requiere **mas informacion especifica de como hacer esto con hardware en python** para no perder tiempo en el avance, ya que todos nostros tenemos tambien trabjao y vidas que atender..

estoy realmente decidido a implementar esto porque **solos no podemos, juntos si** de eso se trata el software de contribucion..

Ya yo he hablado con los proveedores de Epson en este caso, y ya se comunicarme, para saber el estado por ejemplo es 0x02+0x5F+0x03 (comando "_" de estatus) y devuekve un string simple de impelmentar.

Se uso un programa el stc https://www.teuniz.net/serial-com-tester/index.html  ya se tienen los principales comandos y como se envian, en el programa es ma simple, se enviaba es 025f03 muy simple, sin embargo sin saber nada de python avanzado (veo que se programo orientado objetos) la curva de implementacion sera muy lenta..

mckaygerhard, 2018-01-31:
NOTA: por ahora esto solo lleva a que el fiscalberry puedo implementar el dispositivo en si, una vez superado se construye el receptor terminal wifi, que se comunicara con el futuro FISCALBOX. Pero para eso necesito progresar con los kits que llevo.

alevilar, 2018-02-01:
Respondo de a 1:

   1. los drivers no pueden tener el nombre de la impresora simplemente,
   esto complica el desarrollo, ya que cada pais tiene su propio kit, en el
   caso de Venezuela para Epson hay un solo proveedor el cual ya toda
   comunicacion yo la se, pero no podemos meter todo el codigo en un solo
   archivo que domine todas las Epson,

Existe 1 Driver por cada Marca de impresora, pero tambien sepueden crear
Drivers para la misma marca, pero otro pais, por ejemplo, podriamos crear
un Driver llamado *EpsonVenezuela* o *EpsonRepDominicana *con la logica
propia de cada pais. Eso no seria un problema, esmuy facil de hacer leyendo
el PDF con el conjunto de comandos de esa marca y modelo de impresora,

   1. ademas el kit de las Epson de Vnzla es particular porque rescata las
   facturas y es capaz de prosegir y guardar las transacciones, es
   practicamente un minipos, no siendo asi para las Bixolon/Samsung pero para
   estas no hay implementacion en linux, lo que acorta su implementacion en el
   fiscalberry, se tiene pendiente un manual de bajo nivel

Fiscalberry esta pensado para ser fácilmente extensible y poder agregar
infinidad de marcas y modelos. Quizas podamos hacer una video conferencia
donde podria explicar mejor como funciona. Es complejo entenderlo, pero una
vez que se comprende la estructura de fiscalberry, es muy fácil adaptarlo a
muchas otras impresoras.

   1. el lenguaje es python, un tanto dificil de aprender (de aprender
   avanzado para trabajar con hardware, muy distinto a la curva de avance de
   BASIC y GAMBAS) se requiere *mas informacion especifica de como hacer
   esto con hardware en python* para no perder tiempo en el avance, ya que
   todos nostros tenemos tambien trabjao y vidas que atender..

El lenguaje python es buena alternativa para que sea multiplataforma y
lograr buen funcionamiento en dispositivos con procesadores ARM y poca
memoria RAM. Ademas cuenta como muchas librerías que facilitan la
comunicación via SERIAL, ethernet, y muchas otras cosas mas utilizadas como
dependencias en el proyecto. Nosotros estamos muy contentos con ese
lenguaje de programación. Existen muchos tutoriales para aprender a usarlo.