paxapos/fiscalberry #47: [feat] fiscalberry -> fiscalbox evolucionar a caja de hardware

mckaygerhard, 2018-01-31:
Llevar el fiscalberry a un producto de la talla del https://www.odoo.com/page/point-of-sale-hardware#part_3 POSBOX una caja ARM que centraliza el hardware (impresoras) y envia las ordenes.. ya que el piunto de venta de ambos proyectos no es desktop sino web

Esto PODRA LEVAR EL PAXAPOS A USARSE SIN PC SINO EN LAS TABLETS QUE SON EL FUTURO DE LA CLIENTELA, menos cables menos complicado, mas vendible.

¿Qué es -PosBox- FISCALBOX?

PosBox es un dispositivo pequeño que le permite usar los mismos periféricos USB POS estándar de la industria en todos los dispositivos (PC, Mac, Linux, iOS, Android). 

PErmitira que el cliente y el negocio pueda usar el Paxapos en la tablet, en un dispositivo mas independiente de un pc con un XAMP corriendo, sino que en todo el negocio se corre una solucion empresarial unica con servidor unico.
Esto implica:

1. los drivers no pueden tener el nombre de la impresora simplemente, esto complica el desarrollo, ya que cada pais tiene su propio kit, incluso puede haber dos empresas distintas con la misma impresora en el mismo pais
2. lamentablemtne hay que hablar con los proveedores, en el caso de dominicana es especial, porque el proveedor del kit no es de dominicana, es argrentino y es la propia epson, se adjunta el pdf con las especificaciones a bajo nivel

estoy realmente decidido a implementar esto porque **solos no podemos, juntos si** de eso se trata el software de contribucion..

esto engloba lso issues #46 y #45 que conjuntos y otros mas llevarian a fiscal berry a un acaja de impresion completa ERP

alevilar, 2018-02-01:
En realidad eso es la esencia de fiscalberry. Funciona en una raspberry PI
(procesador ARM) que es, creo entender, justamente lo que estas necesitando.
*ya esta funcionando de esta manera* y soporta impresoras marca HASAR,
EPSON y BEMATECH.

Incluso funcionaria en routers con open wrt y mucho ortos dispositivos mas.

Fiscaberry hace eso. Desde el inicio....

Deberias descargarte el proyecto y probarlo. Fue desarrollado para
funcionar en linux, y tambien (yo nunca lo probe pero hay otros que si lo
hicieron) funciona en windows.

Al estar desarrollado en python, funciona en casi todas las plataformas
disponibles.

mckaygerhard, 2018-02-01:
ya comenzamos hacer el prototipo del dispositivo wifi, en un anio estarias si la situacion no empeora, asi con un solo fiscalberry tendrias muchas printers, 

lei ya los otros correos, en pro de colaboracion y viendo el de Epson , si en realidad los comandos sin practicamente identicos a el kit de dominicana.. .. tengo mucho que leer

ya lo descarge, .. lo de "casi todas las plataformas" contrasta la curva de aprendizaje, es lenta.. tengo mucho que leer..

alevilar, 2018-02-01:
Pasa que Python es más difícil de aprender que gambas, pero es un lenguaje
de programación de verdad. Para hacer trabajos serios. Lo otro es armado
como con piezas de LEGO. Para jugar principiantes.

Fiscalberry funciona como un servidor web que espera recibir json con los
comandos.
Estos comandos se convierten al formato de consenso correspondiente de cada
impresora marca y modelo.

Lo qué hay que redefinir es un nuevo comando preparemos un Hangouts donde
podamos discutir todos. Trataremos de proponer una fecha. Y ver si logramos
sumar programadores de la comunidad fiscalberry para hablar de estos temas

mckaygerhard, 2018-02-01:
>. Para hacer trabajos serios. Lo otro es armado como con piezas de LEGO. Para jugar principiantes
bueno con la alta potencia, el JIT-gb esto es tecnicamente falso, adicional el creador de python reconocio que no fue bien disenado sino hasta la version serial 3, cosa que en fiscal berry no se usa..

tengo una duda, vi el codigo de secuencia, al parecer es definida por el usuario , y si el kit es hecho en el mismo ligar puedo asumir que el numero de secuencia lo define uno, es decir ?los numeros enviados no deben estar ligados con el comando? en el caso de el kit de argentina que programe hace anios, y lo confirmo viendo el codigo de uds este:\

``` python
 def _initSequenceNumber(self):
self._sequenceNumber = random.randint(0x20, 0x7f)
def _incrementSequenceNumber(self):
# Avanzo el número de sequencia, volviendolo a 0x20 si pasó el limite
self._sequenceNumber += 1
if self._sequenceNumber > 0x7f:
self._sequenceNumber = 0x20
```

**si ambos kits fueron disenados en el mismo pais, cual es el comando para pedir estatus en el de uds actualmente? es decir estan enviando: 0x02+0x<algo>+0x00+0x01+0x03 ?**

mckaygerhard, 2018-02-01:
> . Para hacer trabajos serios. Lo otro es armado como con piezas de LEGO. Para jugar principiantes

bueno con la alta potencia, el JIT-gb esto es tecnicamente falso, adicional el creador de python reconocio que no fue bien disenado sino hasta la version serial 3, cosa que en fiscal berry no se usa..

tengo una duda, vi el codigo de secuencia, al parecer es definida por el usuario , y si el kit es hecho en el mismo ligar puedo asumir que el numero de secuencia lo define uno, es decir ?los numeros enviados no deben estar ligados con el comando? en el caso de el kit de argentina que programe hace anios, y lo confirmo viendo el codigo de uds este:\

``` python
 def _initSequenceNumber(self):
self._sequenceNumber = random.randint(0x20, 0x7f)
def _incrementSequenceNumber(self):
# Avanzo el número de sequencia, volviendolo a 0x20 si pasó el limite
self._sequenceNumber += 1
if self._sequenceNumber > 0x7f:
self._sequenceNumber = 0x20
```

**si ambos kits fueron disenados en el mismo pais, cual es el comando para pedir estatus en el de uds actualmente? es decir estan enviando: 0x02+0x+0x00+0x01+0x03 ?**

mckaygerhard, 2018-02-01:
ya lo vi en el `Comandos/EpsonComandos.py` bueno calcular el checksum.. pufff ok pregunta autocontestada..

alevilar, 2018-02-01:
Perfecto! El Driver Epson te va a funcionar.así tal cómo está. Ese no te
recomiendo tocar nada.

Lo que tenes que crear es un nuevo archivo
Comandos/EpsonVenezuelaComandos.py
Y que use el Driver común para Epson (te calcula el número de secuencia
entre otras cosas)

Probalo primero redefiniendo el método daily close (de tu archivo
EpsonVenezuelaComandos.... Es el comando más sencillo para arrancar.

pronto te estará funcionando el modelo de impresora que necesitas.
Yo hasta la semana que viene no podré ayudarte en el desarrollo . Solo
pasarte estos tips rápidos.

mckaygerhard, 2018-02-01:
estoy en eso, el de venezuela es el kit ficscal mas flexible del munod no tiene nada de eso de checksum pero es mas bajo nivel, solo necesito enviar y la respuesta tiene ya todo, no se usa rectificaicones con el checksum (redundante obviamente combinado con el numero de secuencia)

wDirac, 2018-02-01:
Hola Alejandro, Gonzalo y PICCORO como estais!? que tal todo; como va el
proyecto de fiscal,✌solo me pasaba a saludar chicos un fuerte saludo.

mckaygerhard, 2018-02-01:
@wDirac hola, aprobechais y traenos unos helados la proxima..