paxapos/fiscalberry #60: en EPSON TM-220 no funciona la NCB

alevilar, 2018-02-27:
[ERR] ComandoException('Error de la impresora fiscal: Campo de datos no v\xc3\xa1lido',)- Error de la impresora fiscal: Campo de datos no válido

al enviar
´´´json
{
"printTicket": {
"encabezado": {
"tipo_cbte": "NCB",
"referencia": "001000000025"
},
"items": [{
"alic_iva": 21.0,
"importe": 0.01,
"ds": "PEPSI",
"qty": 1.0
}, {
"alic_iva": 21.0,
"importe": 0.12,
"ds": "COCA",
"qty": 2.0
}]
},
"printerName": "IMPRESORA_FISCAL"
}
´´´

joseluisbossio, 2018-03-07:
¿A alguien le funciona ese JSON que está bien formado como dice la WIKI para una Nota de Crédito B?

gonzaabel, 2018-03-07:
Hola Jose, el JSON es correcto y esta testeado con las impresoras de primera generación de Hasar. En este caso, el error que tira la epson es que el campo de datos no es valido (puede estar faltando un campo que quizás sea opcional, pero aún así la impresora requiera que se lo envíes como un string vació). Tampoco indica la issue en que momento falla, si es al imprimir items o al imprimir el header de la NCB.
De todas maneras, necesitamos a alguien con Windows, que se baje el emulador, lea la documentación para ese modelo de impresora y averigüé porque esta fallando. (Si no hay quien lo resuelva, me tocara a mi tener que hacerlo, dentro de poco... jejejeje).

joseluisbossio, 2018-03-07:
Tengo la impresora en modo prueba.

Veo si en esta semana me hago tiempo y reviso con el manual a ver qué
encuentro y les voy informando!

gonzaabel, 2018-03-07:
Genial! :smile:

pushnube, 2019-03-22:
Me gustaria aportar la solucion para la TM220u. Como hago?

alevilar, 2019-03-22:
Buenisimo, trataremos de ayudarte lo mas que podamos.

lo primero que te sugiero es COPIAR, je!, copiar lo que ya hicimos.

Para ello deberiamos definir:

*primero*, como vas a conectar la impresora: USB, serial o ETH?
Segun tu respuesta eso representa el "Driver", asi lo llamamos en
fiscalberry a lo referente a la conectividad entre la PC y la impresora
tenes todas las opciones de Driver disponibles en la carpeta "*Drivers*"

*segundo*. los comandos (cuyas opciones estan en la carpeta "*Comandos*").
Estos son los que varian segun los modelos.
Para impresoras fiscales:

Se puede usar el mismo archivo EpsonComandos (
https://github.com/paxapos/fiscalberry/blob/master/Comandos/EpsonComandos.py
)
aunque nosotros para la Hasar 2Gen tuvimos que crear otro conjunto de
comandos totalmente diferentes porque no eran compatibles con los de las
impresoras de la anterior generacion
Tenemos creado, en el archivo "*Sam4s2GenComandos.py*" los comandos para la
Ellix 40F que es compatible con Epson. por lo que creemos que será parecido
a lo que podrias necesitar vos.

Para impresoras de comandas

para las de comandas es mas facil porque generalmente son todas compatibles
con el protocolo EscP

Probá por ese lado y si estas muy trabado o lo que sea me vas escribiendo y
trataremos de ayudarte!

saludosss

pushnube, 2019-03-23:
Hola Ale. Quizá no me exprese bien. El tema es que yo no podía generar notas de crédito con la TM220U. Busque acá y vi que Catriel y otros tampoco habían llegado a una solución. Entonces lo que yo quería EA compartir la solución a las Notas de Crédito. ( No tengo problemas para generarlas solo quería dejar constancia de a donde esta el problema con Json) Gracias!

alevilar, 2019-03-25:
aaah excelente!!, Si no te animas a meterte con github y subir una PULL
REQUEST (lo ideal y mas recomendado asi quedan los cambios con tu autoria
;) , podrias copianos aca mismo las lineas de codigo que deberiamos
modificar y nosotros lo metemos en la ultima version.

Los pasos para hacer una pull request es la siguiente:

1- Forkeas el proyecto fiscalberry en tu cuenta
2- haces un git clone del repo de tu cuenta y trabajas en el
3- subis los cambios a tu repo clonado con "git push"
4- Haces un pull request (aunque eso tambien lo podriamos hacer nosotros)

[image: image.png]
(te copio esta imagen de un repo que yo tengo forkeado para que veas donde
esta el generar "new pull request"

closed 2019-12-13 by alevilar