paxapos/fiscalberry #62: No funciona con P-441F

hernan826, 2018-03-06:
Estimados

Hemos estado probando con una hasar nueva modelo P-441F y no funciona. Pero tampoco tira error, solo queda "tildado" al intentar imprimir.

¿Alguna recomendación de como podríamos solucionar el inconveniente?.

Gracias.

gonzaabel, 2018-03-06:
Hola Hernan, necesitaríamos que nos pases el JSON que le enviaste a la impresora para poder ver si se están mandando bien los datos.
También necesitaríamos ver tu archivo config.ini, y lo que la consola muestra al momento de mandar a imprimir (de ser posible).
Una pregunta: la impresora imprime el header al menos?? (en el header van los datos como por ej: la razón social y CUIT de la SRL que emite el ticket al controlador fiscal).
Ese modelo de Hasar funciona perfectamente, es raro que por consola no te tire error de la impresora.

Saludos!

MarianoGomezBidondo, 2018-03-06:
Hola! lo estoy ayudando con este tema... como habria que configurarla, con que modelo? 441?
[impresora_fiscal]  
marca = Hasar  
modelo = 441
path = /dev/ttyUSB0
driver = Hasar

Creo que es solo un problema de config.
saludos!

hernan826, 2018-03-06:
Hola! Gracias por responder tan rápido..

Te paso lo solicitado:

En el config.ini puse lo siguiente:

[IMPRESORA_NEGRA]
marca = Hasar
modelo = P441
path = /dev/ttyUSB0
driver = Hasar

Verifiqué que la impresora está conectada a /dev/ttyUSB0

Use la página de prueba para enviar lo siguiente:

{"printTicket":{"encabezado":{"tipo_cbte":"T"},"items":[{"alic_iva":21,"importe":0.01,"ds":"PEPSI","qty":1},{"alic_iva":21,"importe":0.12,"ds":"COCA","qty":2}]},"printerName":"IMPRESORA_NEGRA"}

Esta es la respuesta completa de la consola:

2018-03-06 20:57:32,871 - FiscalberryApp - INFO - Preparando Fiscalberry Server
2018-03-06 20:57:32,872 - FiscalberryApp - INFO - Iniciando Fiscalberry Server
2018-03-06 20:57:32,876 - FiscalberryApp - INFO - La IP es 192.168.123.129
2018-03-06 20:57:32,881 - FiscalberryApp - INFO - Hay 4 impresoras disponibles
2018-03-06 20:57:32,881 - FiscalberryApp - INFO -   - IMPRESORA_ARCHIVO
2018-03-06 20:57:32,882 - FiscalberryApp - INFO -       marca: Epson, driver: File
2018-03-06 20:57:32,882 - FiscalberryApp - INFO -   - IMPRESORA_BLANCA
2018-03-06 20:57:32,882 - FiscalberryApp - INFO -       marca: Epson, driver: Epson
2018-03-06 20:57:32,883 - FiscalberryApp - INFO -   - IMPRESORA_NEGRA
2018-03-06 20:57:32,883 - FiscalberryApp - INFO -       marca: Hasar, driver: Hasar
2018-03-06 20:57:32,883 - FiscalberryApp - INFO -   - IMPRESORA_DUMMY
2018-03-06 20:57:32,884 - FiscalberryApp - INFO -       marca: Epson, driver: Dummy
2018-03-06 20:57:32,887 - FiscalberryApp - INFO - *** Websocket Server Started as HTTP at 127.0.1.1 port 12000***
2018-03-06 20:57:33,725 - FiscalberryApp - INFO - new connection
2018-03-06 20:57:35,815 - FiscalberryApp - INFO - new connection
2018-03-06 20:57:54,715 - FiscalberryApp - INFO - ----- - -- - - - ---
2018-03-06 20:57:54,716 - FiscalberryApp - INFO - {"printTicket":{"encabezado":{"tipo_cbte":"T"},"items":[{"alic_iva":21,"importe":0.01,"ds":"PEPSI","qty":1},{"alic_iva":21,"importe":0.12,"ds":"COCA","qty":2}]},"printerName":"IMPRESORA_NEGRA"}
2018-03-06 20:57:54,717 - root - INFO - Iniciando procesamiento de json...
{u'printerName': u'IMPRESORA_NEGRA', u'printTicket': {u'encabezado': {u'tipo_cbte': u'T'}, u'items': [{u'alic_iva': 21, u'qty': 1, u'ds': u'PEPSI', u'importe': 0.01}, {u'alic_iva': 21, u'qty': 2, u'ds': u'COCA', u'importe': 0.12}]}}
inicializando ConectorDriverComando driver de Hasar
HasarDriver.HasarDriver
()
{'path': '/dev/ttyUSB0'}
2018-03-06 20:57:54,750 - root - INFO - sendCommand: SEND|0x40|F|['T', 'T']
^C2018-03-06 20:58:01,797 - FiscalberryApp - INFO - Caught signal: 2
2018-03-06 20:58:01,799 - FiscalberryApp - ERROR - SerialException("read failed: (4, 'Interrupted system call')",)- read failed: (4, 'Interrupted system call')
Traceback (most recent call last):
  File "/home/pi/fiscalberry/FiscalberryApp.py", line 65, in on_message
    response = traductor.json_to_comando(jsonMes)
  File "/home/pi/fiscalberry/Traductores/TraductoresHandler.py", line 64, in json_to_comando
    rta["rta"] = traductor.run(jsonTicket)
  File "/home/pi/fiscalberry/Traductores/TraductorInterface.py", line 20, in run
    res = fnAction(**jsonTicket[action])
  File "/home/pi/fiscalberry/Traductores/TraductorFiscal.py", line 47, in printTicket
    self._abrirComprobante(**encabezado)
  File "/home/pi/fiscalberry/Traductores/TraductorFiscal.py", line 99, in _abrirComprobante
    ret = printer.openTicket()
  File "/home/pi/fiscalberry/Comandos/HasarComandos.py", line 241, in openTicket
    self._sendCommand(self.CMD_OPEN_FISCAL_RECEIPT, ["T", "T"])
  File "/home/pi/fiscalberry/Comandos/HasarComandos.py", line 139, in _sendCommand
    ret = self.conector.sendCommand(commandNumber, parameters, skipStatusErrors)
  File "/home/pi/fiscalberry/ConectorDriverComando.py", line 29, in sendCommand
    return self.driver.sendCommand(*args)
  File "/home/pi/fiscalberry/Drivers/FiscalPrinterDriver.py", line 73, in sendCommand
    reply = self._sendMessage(message)
  File "/home/pi/fiscalberry/Drivers/HasarDriver.py", line 67, in _sendMessage
    self._sendAndWaitAck(message)
  File "/home/pi/fiscalberry/Drivers/HasarDriver.py", line 56, in _sendAndWaitAck
    c = self._read(1)
  File "/home/pi/fiscalberry/Drivers/FiscalPrinterDriver.py", line 82, in _read
    ret = self._serialPort.read(count)
  File "/usr/lib/python2.7/dist-packages/serial/serialposix.py", line 503, in read
    raise SerialException('read failed: {}'.format(e))
SerialException: read failed: (4, 'Interrupted system call')
2018-03-06 20:58:01,808 - FiscalberryApp - INFO - Stopping http server
2018-03-06 20:58:01,809 - FiscalberryApp - INFO - Will shutdown in 2 seconds ...
2018-03-06 20:58:01,810 - FiscalberryApp - INFO - Shutdown
2018-03-06 20:58:01,810 - FiscalberryApp - INFO - Bye!

hernan826, 2018-03-06:
Estimado

Quería comentarte que encontramos el error, eran dos cosas:

1. El modelo puse p441 y era 441
2. El cable estaba equivocado, lo cambiamos por un USB nuevo.

Gracias por la asistencia.

Saludos!

MarianoGomezBidondo, 2018-03-07:
Quisiera agregar mas detalles para que al proximo no le pase lo mismo.
Era una Hasar p441 que tiene salida USB (conector cuadrado) y Serial (con ficha de red, RJ45), la conectamos inicialmente a un Raspberry con un cable adaptador de ficha RJ45 a RS232 (conector tipo D 9 pines) y eso a un cable Manhattan de RS232 a USB. De esa manera no funcionaba, directamente se quedaba esperando luego del primer comando.
Luego la conectamos con el cable USB y anduvo de una!

gonzaabel, 2018-03-07:
Genial! gracias por el aporte Mariano.

closed 2018-03-07 by gonzaabel