paxapos/fiscalberry #66: itemNegative falla en P-441F

hernan826, 2018-04-13:
Estimados

Quería consultarles si existe algún bug conocido al enviar items negativos en la impresora fiscal Hasar P-441F.

El comportamiento observado, es que comienza la impresión, imprime los items normales y corta la impresión al llegar al item negativo dejando inconcluso el ticket.

Luego de eso, no imprime nada más, hasta mandar un Cierre X, el cual anula la impresión del ticket y vuelve a funcionar la impresora.

Según pudimos ver, es siempre que se le manda un item con la siguiente configuración...

```
{
  "alic_iva" : 21,
  "importe" : 10.00,
  "ds" : "Descuento",
  "qty" : 1,
  "itemNegative" : true
}
```

¿Alguna idea de que podría estar causando la falla?.

Muchas gracias por la asistencia.

Saludos!

gonzaabel, 2018-04-14:
Hola Hernan, gracias por el reporte.
Si puedes, por favor muéstranos la salida final por consola si es que te llega algún mensaje de error de la impresora fiscal cuando intentas enviar el ticket con ese item.
Esto lo veré el lunes ya que cuento con ese mismo modelo de impresora en mi lugar de trabajo, para hacer tests.
Saludos!

hernan826, 2018-04-14:
Hola Gonzalo

Gracias por la respuesta, el error que me llega es el siguiente...

[ERR] NameError("global name 'ComandoException' is not defined",)- global name 'ComandoException' is not defined

Avísame si te puedo ayudar en algo más.

Gracias!

Saludos!

hernan826, 2018-04-14:
Ah, me olvidaba, por si te ayuda en algo, este es el objeto con el que acabo de probar...

`{
  "printTicket": {
    "encabezado": {
      "tipo_cbte": "T"
    },
    "items": [
      {
        "alic_iva": 21,
        "importe": 2,
        "ds": "PEPSI",
        "qty": 1
      },
      {
        "alic_iva": 21,
        "importe": 1,
        "ds": "Descuento",
        "qty": 1,
        "itemNegative": true
      }
    ]
  },
  "printerName": "PRINTER_1"
}`

gonzaabel, 2018-04-14:
Bien, hay un par de problemas dando vuelta en Fiscalberry. El primero, es que trata de mostrar el error que le envía la impresora fiscal pero no puede a falta de importar el ComandoException.

Si puedes, por favor en [la siguiente linea](https://github.com/paxapos/fiscalberry/blob/master/Comandos/HasarComandos.py#L6) agrégale ComandoException para que quede de la siguiente manera:

`from Comandos.ComandoFiscalInterface import ComandoFiscalInterface, ComandoException`

Con eso debería poder mostrarte el error de la impresora para que lo puedas subir a esta issue. En caso contrario, el lunes estará arreglado.
Y lo segundo, es que la fiscal tira un error a causa de un fallo en lo que se le envía pero ese JSON es correcto. Así que el lunes lo arreglare y te mantendré al tanto. 😸

gonzaabel, 2018-04-16:
Ya esta arreglado lo del exception. El error que tira es "Comando invalido para el estado fiscal" (gracias Hasar por detallar bien donde esta el error, como siempre :unamused: ).
Tira el error justamente en el item que mando con 'm' para que sea negativo y no sume, sino que reste.
Y según la documentación. Así como se envía el comando, esta bien. Por lo tanto solo se sabe que falla cuando esta la 'm' indicando que tiene que restar en lugar de sumar, pero no especifica  el porque.
A causa de esto no pude solucionarlo, pero tienes una alternativa, la cual es la siguiente:

meterle el descuento al ultimo item impreso (imprime el item, y después imprime el descuento en una linea aparte).
`{
"alic_iva": 21.0,
"importe": 12,
"ds": "COCA",
"qty": 1,
"discount":12,
"discountDescription":"Descuento de promoción: día de la zaraza"
}`

No importa si el descuento tiene el mismo valor que el item, también sirve para imprimir un recargo si añadís discountNegative en false (por defecto discountNegative va en true para descuento):

`{
"alic_iva": 21.0,
"importe": 1,
"ds": "COCA",
"qty": 1.0,
"discount":10.5,
"discountDescription":"Copa rota",
"discountNegative": false
}`

Espero te sirva, y si puedes encontrar la solución al error que tira la fiscal,  avísanos o envíanos una Pull Request con el fix.
Saludos!

hernan826, 2018-04-16:
Gonzalo

Gracias por la respuesta. Muy informativo el comando de Hasar sobre el problema jaja...

Por el momento, hice que el descuento se prorratee entre todos los items. Porque si tenía un descuento general de $ 200, y ningún item de ese porte, tampoco lo acepta la fiscal. Es decir, si queda en negativo, nuevamente tira error y no funciona.

Así que por el momento lo pase a todos los items, pero el ticket queda muy desprolijo, porque la fiscal incluye una leyenda extra en todos los items.

Voy a intentar contactar con el soporte técnico de Hasar para ver si de casualidad me sugieren alguna opción, estamos trabajando con un contacto de confianza así que es posible me puedan dar una guía.

Si obtengo más información probaré la solución. Ya encontré la línea en cuestión que configurar la "m" en el código, así que puedo probar directamente.

Si por casualidad lo reparamos, lo posteo por acá para compartir la solución.

Gracias!

alevilar, 2018-04-16:
> Alejandro
>
> La idea es imputar un descuento en pesos a todo el ticket.
>
> Ejemplo:
> Coca Cola x 2 --- > $100
> Comida x 1 -------> $ 150
> Descuento general -----> - $30
>
> Total --------------------> $ 220
>
> Saludos

Ok! en tal caso eso ya está implementado. No tenes que agregar un producto
negativo, sino que tenes que agregar un descuento general (tambien sirve
para hacer recargos)

en las fiscales eso se llama addAditional

en el printTicket le tenes que agregar el JSON addAditional y LISTO!

{addAdditional: {
  descripcion: "Descuento",
  amount: 200,
  iva: 21,
  negative: True
}}

te dejo el link la wiki donde tenes mejor info:
https://github.com/paxapos/fiscalberry/wiki/JSON-Comandos-Para-impresora-Fiscal#addadditional-opcional

gonzaabel, 2018-04-16:
@hernan826 lo que comenta @alevilar es claramente la solución a tu problema.
Lo del item debería funcionar. No hay nada mal en el envió del comando como para tirar ese error. (lo probé tal cual muestra de ejemplo la documentación y nada...)
Así que si en tal caso, logras encontrarle la vuelta y arreglarlo, avísanos.
Saludos!

hernan826, 2018-04-16:
Muchas gracias a ambos por la asistencia, lo implementaré de la forma que comenta @alevilar. De todas formas, intentaré averiguar con Hasar el problema del negativo, para ver si logro aportar dicha solución.

Gracias nuevamente.

Saludos!

hernan826, 2018-04-19:
Solo para quién lea el follow up de este bug. Efectivamente para crear el descuento general, usando addAditional funciona bien, pero el objeto de la documentación tiene un pequeño error...

El correcto es:

```
{addAdditional: {
  description: "Descuento",
  amount: 200,
  iva: 21,
  negative: True
}}
```

La diferencia es la propiedad "descripcion" debe ser "description". Con esto funcionará OK.

Gracias a @gonzaabel  y @alevilar  por la asistencia en este asunto, gracias a ello pudimos corregir el código y estar funcionando.

Si encontramos más información sobre el número negativo la postearemos.

Saludos!

closed 2019-03-01 by gonzaabel