paxapos/fiscalberry #69: Device disconnected or multiple access on port?

joseluisbossio, 2018-04-17:
Buenas, nos ocurre lo siguiente aleatoriamente que el port de la impresora queda en estado desconectado o que de esta intentando acceder en forma simultánea y no lo podemos controlar lo que nos genera problemas.

He visto en algunos foros de PySerial, pero hacian referencia a un bug en una versión antigua de Kernel de Linux.

Lo tengo funcionando sobre Linux Ubuntu 16.04 x64 y parece que ante un problema con un mensaje enviado a imprimir, no se recupera y lo deja en modo "ocupado" por sobre una Epson TM-U220 conectada por el cable RJ-45 a DB3 + adaptador Manhatan de DB9 a USB para llegar al Linux y usarlo como /Dev/ttyUSB[x]

return_value = printer.getLastNumber("A")
File "/usr/local/lib/python2.7/dist-packages/pyfiscalprinter/epsonFiscal.py", line 361, in getLastNumber
reply = self._sendCommand(self.CMD_STATUS_REQUEST, ["A"], True)
File "/usr/local/lib/python2.7/dist-packages/pyfiscalprinter/epsonFiscal.py", line 108, in _sendCommand
return self.driver.sendCommand(commandNumber, parameters, skipStatusErrors)
File "/usr/local/lib/python2.7/dist-packages/pyfiscalprinter/epsonFiscalDriver.py", line 112, in sendCommand
reply = self._sendMessage( message )
File "/usr/local/lib/python2.7/dist-packages/pyfiscalprinter/epsonFiscalDriver.py", line 166, in _sendMessage
c = self._read(1)
File "/usr/local/lib/python2.7/dist-packages/pyfiscalprinter/epsonFiscalDriver.py", line 85, in _read
ret = self._serialPort.read( count )
File "/usr/local/lib/python2.7/dist-packages/pyserial-3.4-py2.7.egg/serial/serialposix.py", line 501, in read
'device reports readiness to read but returned no data '
SerialException: device reports readiness to read but returned no data (device disconnected or multiple access on port?)

Gracias (totales)
Jose Luis Bossio

gonzaabel, 2018-04-19:
Parecería ser por lo que comentas, que ante un error al mandar a imprimir, no cierra la conexión y entonces al mandar a imprimir una segunda vez te tira ese error (por tener una conexión aún activa).
¿Has comprobado si esto ocurre en Fiscalberry? puede que sea un bug del proyecto de Mariano Reingart y que aquí este arreglado.

gonzaabel, 2018-04-20:
Mmm... si no es la impresora, puerto o cable, es debido a que por alguna razón no cierra la conexión con la impresora y al intentar conectar de nuevo te tira ese error por acceso múltiple.
Aunque si no le ocurre a Mariano en su proyecto es extraño, por eso preguntaba si con Fiscalberry te ocurría lo mismo jeje.
Saludos!

joseluisbossio, 2018-04-20:
Se le ocuren cosas varias, pero de momento se podrian ver con soporte
comercial o esperar.

closed 2019-12-13 by alevilar