paxapos/fiscalberry #72: Problemas con codepage, no funciona

gonzaabel, 2018-05-08:
2018-05-08 13:44:02,920 - root - INFO - Iniciando procesamiento de json...
{u'printPedido': {u'encabezado': {u'pedido_recepcionado': False, u'telefono': u'-', u'nombre_proveedor': u'BEBIDAS DEL MAR', u'cuit': u'20407214634'}, u'items': [{u'observacion': u'', u'unidad_de_medida': u'Kilo', u'importe': u'', u'ds': u'yogur', u'qty': u'3.00'}, {u'observacion': u'', u'unidad_de_medida': u'Bolsa', u'importe': u'', u'ds': u'mirinda', u'qty': u'2.00'}]}, u'printerName': u'barra'}
inicializando ConectorDriverComando driver de ReceiptDirectJet
<class 'Drivers.ReceiptDirectJetDriver.ReceiptDirectJetDriver'>
()
{'codepage': 'cp858', 'vendor': 'zaraza', 'host': '192.168.1.124'}
2018-05-08 13:44:02,976 - FiscalberryApp - ERROR - UnicodeDecodeError('ascii', 'Nueva \xc3\xb3rden de compra \n', 6, 7, 'ordinal not in range(128)')- 'ascii' codec can't decode byte 0xc3 in position 6: ordinal not in range(128)
Traceback (most recent call last):
  File "/home/gonza/fiscalberry/FiscalberryApp.py", line 69, in on_message
    response = traductor.json_to_comando(jsonMes)
  File "/home/gonza/fiscalberry/Traductores/TraductoresHandler.py", line 65, in json_to_comando
    rta["rta"] = traductor.run(jsonTicket)
  File "/home/gonza/fiscalberry/Traductores/TraductorInterface.py", line 19, in run
    res = fnAction(**jsonTicket[action])
  File "/home/gonza/fiscalberry/Traductores/TraductorReceipt.py", line 11, in printPedido
    return self.comando.printPedido(**kwargs)
  File "/home/gonza/fiscalberry/Comandos/EscPComandos.py", line 64, in printPedido
    printer.text("Nueva órden de compra \n")
  File "/usr/local/lib/python2.7/dist-packages/escpos/escpos.py", line 435, in text
    self._raw(txt.encode(self.codepage))
  File "/usr/lib/python2.7/encodings/cp858.py", line 12, in encode
    return codecs.charmap_encode(input,errors,encoding_map)
UnicodeDecodeError: 'ascii' codec can't decode byte 0xc3 in position 6: ordinal not in range(128)
Exception socket.error: error(9, 'Bad file descriptor') in <bound method ReceiptDirectJetDriver.__del__ of <Drivers.ReceiptDirectJetDriver.ReceiptDirectJetDriver object at 0x7f7f3af54ad0>> ignored

alevilar, 2018-05-08:
hay una funcion por algun lado que es la encargada de procesar los strings
para arreglar los codepages, lo habia hecho el GROSO Ariel, en el año
2016/2017.... buscala ..

por el momento se podria arreglar facil quitando el acento a la "orden de
compra"? igualmente eso ya lo habiamos arreglado. no recuerdo exactamente
cómo....

gonzaabel, 2018-05-10:
El codepage funciona, pero para los strings que vos le envias por JSON a Fiscalberry.
si vos mandas caracteres como acentos o eñes en un  printer.text(); ahí te tira el error.
Igualmente sigue siendo un bug a resolver.

closed 2018-05-10 by gonzaabel