paxapos/fiscalberry #73: Driver fiscal Samsung?

hernan826, 2018-05-09:
Buenas, quería consultar si existe documentación en algún sitio para conocer el listado de drivers completo para configurar en el archivo config.ini.

En particular, estoy buscando si existe el driver para conectar con el modelo de impresora fiscal Samsung SRP-270F. En teoría me dijeron que es igual al driver para Epson TM U220 AF (Que tampoco conozco si esta soportado).

¿Podrían indicarme a dónde buscar esta info?.

Gracias!

gonzaabel, 2018-05-09:
Hola Hernan, en la [carpeta de drivers](https://github.com/paxapos/fiscalberry/tree/master/Drivers) encontraras todos los que estan actualmente disponibles, ya sea para casos de testeos (como el FileDriver) o los de las distintas marcas de impresoras fiscales y de comandas.

> En particular, estoy buscando si existe el driver para conectar con el modelo de impresora fiscal Samsung SRP-270F. En teoría me dijeron que es igual al driver para Epson TM U220 AF (Que tampoco conozco si esta soportado).

La impresora fiscal Epson TM 220 es soportada por Fiscalberry, así que en teoría podrías usar [EpsonComandos](https://github.com/paxapos/fiscalberry/blob/master/Comandos/EpsonComandos.py#L51) con esa impresora de Samsung.

Saludos!

gonzaabel, 2018-05-09:
Los modelos disponibles los podes ver dentro de los archivos de comandos de Hasar y Epson. Si en el config especificas un modelo que no esta dentro del array (o directamente no lo especificas) se usa uno de los modelos por defecto (el 615 para Hasar, si mal no recuerdo). Y es por esto que funciona la Hasar 441-F, con Epson pasa lo mismo.

Saludos!

alevilar, 2018-05-10:
Debería ser algo así:

[impresora_fiscal]
marca = Epson
modelo = tm-220
path = /dev/ttyUSB0

( o path = COM1 si es windows)

Probalo y avísame si funciono

alevilar, 2018-05-24:
@hernan826  pudiste probarlo?

hernan826, 2018-05-30:
@alevilar gracias por el followup y mil disculpas por la demora. No hemos tenido suerte aún, en especial porque creo que tenemos un problema con el cable.

Estimo pronto poder destrabar el tema y probar si podemos hacer funcionar la Samsung. En caso de que logremos validar que es posible usar el driver de Epson para Samsung, lo estaré posteando para cualquier interesado que podría servirle a futura referencia.

hernan826, 2019-01-18:
Lamentablemente nunca logramos hacerlo funcionar correctamente. Terminamos desistiendo.

gonzaabel, 2019-01-18:
Okk. Ahora si somos compatibles con esa fiscal en modo Epson.
Configurada se vería de la siguiente forma:
```ini
[impresora_fiscal]
marca = Epson
modelo = sm-srp-270
path = Linux [/dev/ttyUSBX] or Windows [COMX]
driver = Epson
```
La única peculiaridad con esta impresora conectada con un adaptador Manhattan de Serial a USB, es que nos traba la conexión con fiscalberry de manera aleatoria en cualquier ticket. Sin un error por parte de la impresora, queda todo como si la comunicación se hubiese detenido.

Cierro issue.

closed 2019-01-18 by gonzaabel