paxapos/fiscalberry #74: Soporte de EscP para comanderas con conexción USB en Windows

qmfkdldks, 2018-07-23:
Hola! Buenas.
Quiero saber si el programa soporta la conexcion USB en windows.
Si se soporta como tengo que configurar el config.ini.

[impresora1]  
marca = EscP  
driver = RecieptUSB
path = USB002

¿Por ejemplo, así sería? 
No sé, tengo que ponder en vez de "/dev/ttyUSB0" al parametro path para que funcione en windows con conexción USB.

La impresora es: SAM4S Ellix 30

Saludos

joseluisbossio, 2018-07-23:
"/dev/ttyUSB0" para Windows seguro que no.

alevilar, 2018-07-24:
La verdad no la tengo clara con Windows. Pero si fuese conexión SERIAL
sería usando el virtualizador de puertos COM que muchas veces viene como
Driver en la misma impresora

Y se usa poniendo el nombre del puerto donde iestanla impresora
Y si usas el puerto COM tenes que usar el driver Serial:

Quedaría así la configuración

[impresora1serial]
marca = EscP
driver = ReceiptSerial
devfile = COM3

También podrías probar la siguiente configuración:

[impresora2usb]
marca = EscP
driver = RecieptUSB
usb_vendor= xxxxxxx
usb_product= yyyyyy

Para sacar el id del vendor y product tenes acá una respuesta

https://stackoverflow.com/questions/3279800/how-to-get-usb-vendor-and-product-info-programmatically-on-linux

Avísame si te funciono que te estoy respondiendo de memoria con el celu y
puede que esté fallando en algún parámetro

qmfkdldks, 2018-08-03:
Estuve investigando sobre el tema. Y parece que es dificíl trabajar con comanderas en Windows. Es 
Hay varias alternativas pero entre todos los más probables son:
1. py32win o win32print (librerias de python)
2. PyQt

spetrungaro, 2022-03-12:
Lo que se necesita son 2 cosas, encontrar cuales son los endpoints asignados al dispositivo y despues cambiarle el driver para que use el de libusb.

1- Cambiar el driver con la herramienta https://zadig.akeo.ie/:
Abrirlo, ir a Options -> List all Devices. **Seleccionar la impresora** del desplegable y reemplazarle el driver por 
**"libusb-win32 (v1.2.6.0)"**

[image]

2- Encontrar los endpoints: Hay que descargar el SDK desde https://developer.microsoft.com/en-us/windows/downloads/windows-sdk/. Instalar solo los necesario seleccionando **unicamente** la casilla de **"Debugging Tools for Windows"**. 
El programa se llama usbview.exe y se va a instalar en %\Windows Kits\10\Debuggers\x64\usbview.exe
(O jugar con combinaciones de in_ep (0x81 - 0x84) y out_ep (0x01 - 0x04) que es donde suelen estar siempre).
Seleccionar el dispositivo correspondiente de la barra lateral y buscar este apartado:
[image]

3- El config.ini tiene que quedar como este ejemplo:

[impresora]
marca = EscP
driver = ReceiptUSB
in_ep = 0x82                 <- Conseguido de usbview
out_ep = 0x01               <-  Idem
usb_vendor = 0x14f3    <- VID de la impresora
usb_product = 0x2a3b  <- PID

Los VID y PID se encuentran fácilmente en el adm de dispositivos de Windows, en el Zadig o en el mismo usbview.exe
Funciona siempre.

closed 2022-03-12 by spetrungaro