paxapos/fiscalberry #79: Es posible conectar varias impresoras Receipt (de comandas) y direccionarlas?

jobiols, 2018-11-08:
Queria saber si puedo conectar dos o tres impresoras no fiscales a la raspberry y de alguna manera direccionarlas al mandarles el json para imprimir.
O tengo que poner una raspberry por impresora.?

joseluisbossio, 2018-11-08:
A distintos puertos USB y podés tener más de una impresora.

gonzaabel, 2018-11-09:
Hola @jobiols 
Puedes conectar más de una comandera tanto por USB como por red. Para hacer esto, tienes que editar el config.ini y agregar nuevas secciones:

[IMPRESORA_RED] //nombre de sección usado como el printerName en el JSON, puede ser cualquier nombre entre corchetes []
marca = EscP 
host = 192.168.1.122 //IP de la impresora en la red
driver = ReceiptDirectJet

[otra_impresora]
marca = EscP
path = /dev/ttyUSB0 //Path hacía el puerto del dispositivo (en Windows si no me equivoco puede ser COM1, COM2, etc)
driver = ReceiptDirectJet

[otra_impresora_de_red]
marca = EscP
host = 192.168.1.123
driver = ReceiptDirectJet

Luego para direccionarlas, en los JSON, vas a tener que mandar como valor del printerName, el nombre que le pusiste a la sección para mandar a imprimir por esa comandera. Por ejemplo:
`
{"printRemito":
{"encabezado":
{"tipo_cbte":"T"},
"items":[
{"alic_iva":21,"importe":0.01,"ds":"PEPSI","qty":1},
{"alic_iva":21,"importe":0.12,"ds":"COCA","qty":2}
]
},"printerName":"otra_impresora_de_red"}
`

Saludos!

jobiols, 2018-11-09:
Perfecto muchisimas gracias!!

gonzaabel, 2018-11-09:
De nada! :smile:

closed 2018-11-09 by gonzaabel