paxapos/fiscalberry #80: Error al enviar json

jobiols, 2018-11-22:
Para testear el sofware arme una maquina virtual con raspbian en virtualbox.
y en el host tengo un script de esta forma:

```
get_printers = {"getAvaliablePrinters": ""}
ws = websocket.WebSocket()
ws.connect("ws://192.168.0.160:12000/ws")
ws.send(json.dumps(get_printers))
answ = ws.recv()
print answ
ws.close()
```

En este caso la fiscalberry responde correctamente:
{"rta": {"action": "getAvaliablePrinters", "rta": ["IMPRESORA_FISCAL", "COMANDERA"]}}

sin embargo cuando le mando este json o cualquier otro que apunte a la impesora
last_number = {"getLastNumber": "T", "printerName": "IMPRESORA_FISCAL"}

me contesta:
{"err": "Error parseando el JSON __init__() takes at least 2 arguments (1 given)"}

y en la consola dice:

alevilar, 2018-11-22:
eso es por un problema enn la configuracion. pasanos tu config.ini a ver
como lo tenes para la impresora *IMPRESORA_FISCAL*

joseluisbossio, 2018-11-22:
Hola Jorge, como va? Como dice Ale, pasanos el ini, marca, modelo de la
impresora y si usas adaptador serial a USB (por ejemplo para una TMU-220 de
epson)

jobiols, 2018-11-22:
Este es el config. La impresora no la tengo estoy usando un emulador que corre en otra maquina virtual windows. Y conecto los dos con un pipe que esta verificado y funciona.
Se supone que el pipe emula un adaptador serial

[SERVIDOR]
puerto = 12000
discover_url = 
ssl_crt_path = 
ssl_key_path = 
ip_privada = 192.168.0.160
uuid = 080027f4b44e

[IMPRESORA_FISCAL]
marca = Hasar
modelo = 615
path = /dev/ttyS0
driver = ReceiptDirectJet

[COMANDERA]
marca = EscP
path = /dev/tty0
driver = ReceiptDirectJet

closed 2018-11-22 by jobiols

gonzaabel, 2018-11-22:
El problema esta en el driver de la impresora fiscal, debería ser Hasar, FiscalPrinter o Epson para poder usar tanto ese comando como cualquier otro comando que sea fiscal.
El driver ReceiptDirectJet es para las impresoras de comandas que van por red. En los casos de prueba se usa el driver File que es una salida que escribe en un archivo de texto lo que saldría impreso en el papel (apuntando con el parametro path el lugar donde se creará el archivo).

Saludos!