paxapos/fiscalberry #86: Problema al obtener el ultimo comprobante impreso

renzohcoronel, 2019-03-19:
Hola a todos, estuve haciendo algunas pruebas con fiscalberry y la impresora smh/p 340. Logre conectarla y probar los siguientes comandos: 

1) El comprobante X se imprimió correctamente. Ahora cuando quise realizar un Z
```[json]
{"printerName": "IMPRESORA_FISCAL", "dailyClose": "X"}
```
En la consola obtuve este error:
2019-03-15 17:07:31,532 - root - ERROR - No se pasó un comando válido
2019-03-15 17:07:31,532 - FiscalberryApp - ERROR - Traductor Comandos: No se pasó un comando válido

2) Al Imprimir un comprobante con el siguiente JSON
```[json]
{"printTicket": {"encabezado": {"tipo_cbte": "FA", "nro_doc": "23341261759", "domicilio_cliente": "Rua 76 km 34.5 Alagoas", "tipo_doc": "CUIT", "nombre_cliente": "Joao Da Silva", "tipo_responsable": "RESPONSABLE_INSCRIPTO"}, "items": [{"alic_iva": 21.0, "qty": 1.0, "ds": "PIPI", "importe": 0.01}, {"alic_iva": 21.0, "qty": 2.0, "ds": "MKF010", "importe": 0.22}]}, "printerName": "IMPRESORA_FISCAL"}

```
No hubo forma de poder obtener el numero de el ultimo comprobante impreso, ni siquiera cuando termino de imprimirlo notifico  cual había sido ¿Sera cuestión del modelo de impresora que estuve probando? El Log de la aplicación no pareciera arrojar ningún error. 
Otra cosa y no menor, en la empresa manejamos comprobantes que son en Efectivo y a CC, como la impresora estaba conectada a una pc desde la cual se factura  y el ultimo comprobante fue realizado a CC, no pude encontrar forma de cambiarlo desde fiscalberry.

2019-03-15 17:23:40,742 - FiscalberryApp - INFO - {"getLastNumber":"FA","printerName":"IMPRESORA_FISCAL"}
2019-03-15 17:23:40,743 - root - INFO - Iniciando procesamiento de json:::: {"getLastNumber": "FA", "printerName": "IMPRESORA_FISCAL"}
2019-03-15 17:23:40,755 - root - INFO - mandando comando de impresora
inicializando ConectorDriverComando driver de Hasar
2019-03-15 17:23:40,765 - root - INFO - inicializando ConectorDriverComando driver de Hasar
2019-03-15 17:23:40,765 - root - INFO -   - Driver: HasarDriver.HasarDriver
2019-03-15 17:23:40,767 - root - INFO - sendCommand: SEND|0x2a|T|[]
2019-03-15 17:23:40,859 - root - INFO - reply: ['C080', '0600', '00000011', '0002', '00001040', '0000', '00000003', '00000028', '00017461']

4) Entre los comprobantes disponibles no se encuentran los remito ¿Hay algún issue o alguna intención de en el mediano o corto plazo agregarlos?

Desde ya muchas gracias. Es un excelente proyecto 

Saludos

gonzaabel, 2019-03-22:
Hola Renzo, perdón por la demora en responder:

> Hola a todos, estuve haciendo algunas pruebas con fiscalberry y la impresora smh/p 340. Logre conectarla y probar los siguientes comandos:
> 
> 1. El comprobante X se imprimió correctamente. Ahora cuando quise realizar un Z
> 
> ```
> {"printerName": "IMPRESORA_FISCAL", "dailyClose": "X"}
> ```
> En la consola obtuve este error:
> 2019-03-15 17:07:31,532 - root - ERROR - No se pasó un comando válido
> 2019-03-15 17:07:31,532 - FiscalberryApp - ERROR - Traductor Comandos: No se pasó un comando válido

La única forma de que el traductor devuelva ese error, es porque estas enviando un comando desconocido o que no es para el uso general de Fiscalberry. Dichos comandos no requieren de **printerName**.
El JSON correcto para mandar a imprimir un Z sería: ` {"printerName": "IMPRESORA_FISCAL", "dailyClose": "Z"}`.
En pocas palabras, lo que te sucedió se debe a que el JSON que enviaste para el Z le falto el printerName. Prueba nuevamente y nos comentas si funciono :)

> 1. Al Imprimir un comprobante con el siguiente JSON
> 
> ```
> {"printTicket": {"encabezado": {"tipo_cbte": "FA", "nro_doc": "23341261759", "domicilio_cliente": "Rua 76 km 34.5 Alagoas", "tipo_doc": "CUIT", "nombre_cliente": "Joao Da Silva", "tipo_responsable": "RESPONSABLE_INSCRIPTO"}, "items": [{"alic_iva": 21.0, "qty": 1.0, "ds": "PIPI", "importe": 0.01}, {"alic_iva": 21.0, "qty": 2.0, "ds": "MKF010", "importe": 0.22}]}, "printerName": "IMPRESORA_FISCAL"}
> ```
> No hubo forma de poder obtener el numero de el ultimo comprobante impreso, ni siquiera cuando termino de imprimirlo notifico cual había sido ¿Sera cuestión del modelo de impresora que estuve probando? El Log de la aplicación no pareciera arrojar ningún error.

Normalmente Fiscalberry devuelve un mensaje con el nombre del comando, en este caso printTicket con el número del comprobante emitido. Si el mensaje llega vacio (se puede ver desde la sección de mensajes de la página principal) quiere decir que la fiscal no devuelve ese valor o a lo mejor devuelve algo que no es un string.

> Otra cosa y no menor, en la empresa manejamos comprobantes que son en Efectivo y a CC, como la impresora estaba conectada a una pc desde la cual se factura y el ultimo comprobante fue realizado a [...]

De esto no sabría que decirte, salvo que te refieras a como poder indicar el tipo de pago y el monto del pago de la factura?? esto se haría añadiendo en el JSON un listado de pagos que tendría su descripción "CC" y un monto:

```json
{
"pagos": [
        {
        "ds": "efectivo", 
        "importe": 150
        },
        {
        "ds": "CC", 
        "importe": 200
        }
    ]
}
```

Aunque nose si era eso lo que querías hacer.

> 1. Entre los comprobantes disponibles no se encuentran los remito ¿Hay algún issue o alguna intención de en el mediano o corto plazo agregarlos?

En teoría haría falta hacer algo similar en el TraductorFiscal a lo que hacemos con el printTicket. Ya que esta en HasarComandos el comando openRemito y nose si será funcional (nunca lo hemos probado).
Tampoco tenemos ninguna issue al respecto ni hemos planificado agregar ese comando. Sería una buena colaboración si lograran hacerlo ustedes y abrieran una pull request para incluirlo :wink: 
En nuestro caso imprimimos remito en formato ticket a modo de comprobante de consumición para restaurantes, por medio de una impresora de comandas con el protocolo EscP. 

Saludos :)

renzohcoronel, 2019-04-09:
Hola @gonzaabel, mil disculpa por la demora, no me notifico, ni me llego mail que habías escrito ( y si llego capaz que  lo eh borrado) Todavía no eh podido hacer de nuevo pruebas. No tenemos ninguna impresora para probar largo y tendido. Las únicas que contamos que podemos probrar, son las que hay en un deposito y nos queda muy a tras mano para ir hacer pruebas. 
Con respecto al comando para imprimir el Z, funciono, el que no funcionaba es el X creo que es el cierre. Siempre arrojaba comando invalido.

Y con respecto  la impresión del comprobante; no pudimos obtener el numero de factura que hizo. Ni pidiendo  el ultimo tikect, ni al momento de que se termino de  imprimir el comprobante. Ni bien pueda realizar las pruebas me vuelvo a contactar.

> En teoría haría falta hacer algo similar en el TraductorFiscal a lo que hacemos con el printTicket. Ya que esta en HasarComandos el comando openRemito y nose si será funcional (nunca lo hemos probado).
Tampoco tenemos ninguna issue al respecto ni hemos planificado agregar ese comando. Sería una buena colaboración si lograran hacerlo ustedes y abrieran una pull request para incluirlo 

Si, seria mi idea poder hacerlo, no soy desarrollador python. Pero quisiera aprender en el transcurso. Así que ni bien lea un poco sobre el lenguaje abro el issue para agregar esa posibilidad. Si nadie lo hace antes. Si lo hacen me prendo para ayudar a realizar la tarea.

Desde ya muchas gracias! :D ￼
Saludos

gonzaabel, 2019-04-10:
> Con respecto al comando para imprimir el Z, funciono, el que no funcionaba es el X creo que es el cierre. Siempre arrojaba comando invalido.

Esto es extraño, no debería tirar un mensaje como el de "comando valido para el estado fiscal" (si es este exactamente el mensaje que te retornaba) al intentar imprimir un informe X. Y menos cuando el Z funciona jejeje. Salvo algo particularidad de la impresora :thinking: 

> Y con respecto la impresión del comprobante; no pudimos obtener el numero de factura que hizo. Ni pidiendo el ultimo tikect, ni al momento de que se termino de imprimir el comprobante. Ni bien pueda realizar las pruebas me vuelvo a contactar.

Bien, probablemente estés usando unas de las versiones más recientes de Fiscalberry, desde Febrero se incorporo para los comandos de impresoras la ejecución por medio de multiprocessing, para que cada comando vaya por un hilo distinto. Hemos notado recientemente que no recibíamos respuesta de ningún comando de los que se manejan las impresoras (ni fiscal ni receipt) así que ya llevo como una semana tratando de arreglarlo y cada vez estoy más cerca. Pero es por esto que las respuestas no retornan al cliente desde el servidor :disappointed: 

> Si, seria mi idea poder hacerlo, no soy desarrollador python. Pero quisiera aprender en el transcurso. Así que ni bien lea un poco sobre el lenguaje abro el issue para agregar esa posibilidad. Si [...]

Genial, cualquier problema que tengas durante el proceso la comentas y trataremos de darte una mano en la medida de lo posible :smile: 

Saludos!

gonzaabel, 2019-04-15:
> Bien, probablemente estés usando unas de las versiones más recientes de Fiscalberry, desde Febrero se incorporo para los comandos de impresoras la ejecución por medio de multiprocessing, para que cada comando vaya por un hilo distinto. Hemos notado recientemente que no recibíamos respuesta de ningún comando de los que se manejan las impresoras (ni fiscal ni receipt) así que ya llevo como una semana tratando de arreglarlo y cada vez estoy más cerca. Pero es por esto que las respuestas no llegan al cliente desde el servidor 

@renzohcoronel  te menciono para que te llegue una notificación (capaz que antes por no mencionarte, no te notificaba github al correo jeje). Este problema que comente acá ya esta solucionado en el commit más reciente. Así que hace un pull (o baja un nuevo zip, si no clonaste con git el proyecto) y el problema de que no te retornaba el número del comprobante impreso debería estar solucionado de ahora en más.
Cuando puedas verificarlo, avísame y si funciona damos por cerrada esta issue.

Saludos!

renzohcoronel, 2019-04-16:
@gonzaabel Buenisimo gonzalo. Esta semana es complicado, pero la otra coordino para ir a probar este cambio. Empece a estudiar python para hacer la parte de generacion de remitos. Es muy probable que empiece a consultar cosas como esta estructurado el proyecto y donde colocar las cosas.

Desde ya muchisimas graciass!!!

Saludos.

renzohcoronel, 2019-06-18:
Hola @gonzaabel , disculpa el tiempo que tarde en responder. No contamos con una impresora física y no estoy trabajando dentro de la empresa. Así que cuesta un poco conseguirla para probar. Ahora si me devuelve correctamente el numero del ultimo comprobante impreso. 

Saludos..

gonzaabel, 2019-06-19:
@renzohcoronel  buenísimo! cierro esta issue. Abrí una nueva issue ante cualquier otro problema. Y cualquier consulta que tengas las seguimos por aquí.
Saludos!

closed 2019-06-19 by gonzaabel