paxapos/fiscalberry #90: Documentación SDK

pasmaan, 2019-05-24:
Buenos días,

Quisiera saber si se provee de un SDK para este tipo de impresoras fiscales o si existe el mismo, ya que estoy intentando realizar un desarrollo en Android y me vendría bien alguna ayuda. Estuve buscando en internet pero la verdad es que no hay mucho.

Saludos.

alevilar, 2019-05-26:
para android no te va a servir. Fiscalberry es para ser utilizado como
servidor de websockets con tornado (python). Al menos nosotros aun no
logramos hacerlo funcionar en andriod.

joseluisbossio, 2019-05-26:
Nosotros tampoco (aún)

pasmaan, 2019-05-28:
Uff durisimo, además la mayoria del mercado tiene impresores de primera generación. Voy a investigar más, cualquier cosa aviso. 
Saludos y gracias.

mtotaro, 2019-06-12:
Buenas tardes estimados,

Para a los que les interese, dejo mi humilde aporte:

https://github.com/grupohasar/HasarFiscalDriver

Enjoy 

Saludos.

closed 2019-12-13 by alevilar