paxapos/fiscalberry #91: no puedo instalar las dependencias del proyecto

jobiols, 2019-06-21:
Buenas
Intento instalar el proyecto en una raspberry con raspbian instalado desde cero.
Me bajo la imagen LITE desde https://www.raspberrypi.org/downloads/raspbian/ y la flasheo en la SD
Cuando intento actualizar trata de bajar las cosas del servidor de uruguay que esta caido con lo que cambio las sources list por esto
- deb http://mirrors.ocf.berkeley.edu/raspbian/raspbian/ stable main contrib non-free rpi firmware
- deb http://archive.raspberrypi.org/debian/ stretch main ui

Ahora puedo actualizar el sistema, pero luego de bajar el proyecto al instalar dependencias falla la cryptography y no termina de instalar.

Estimo que estoy bajando los sources mal pero ya probe varios y no logro hacerlo andar.
Alguna sugerencia???

jobiols, 2019-06-22:
Bueno parece que el server de uruguay respondio y pude instalar todas las dependencias. Ahora tengo otro problema. No veo la impresora en /dev pero si aparece cuando hago un lsusb:

Bus 001 Device 005: ID 04b8:0201 Seiko Epson Corp. 

Que habria que poner en el path del config.ini ?

gonzaabel, 2019-06-22:
> Bueno parece que el server de uruguay respondio y pude instalar todas las dependencias. Ahora tengo otro problema. No veo la impresora en /dev pero si aparece cuando hago un lsusb:
> 
> Bus 001 Device 005: ID 04b8:0201 Seiko Epson Corp.
> 
> Que habria que poner en el path del config.ini ? probé varias que no funcionan...
> 
> [IMPRESORA_FISCAL]
> marca = Epson
> modelo = tm-t900fa
> path = USB004
> driver = Epson

Hola @jobiols para configurar una impresora por USB necesitas los siguientes datos: vendor_id (según lo que pasaste con el comando lsusb, es **04b8**), product_id (es **0201**), el input end point y el output end point.
Acá te dejo un [link de nuestra wiki](https://github.com/paxapos/fiscalberry/wiki/Configurando-mediante-config.ini#driver-usb) con un ejemplo y donde también encontraras otro link que llevan a la documentación de python-escpos con los comandos para saber los datos que te faltan.

Cualquier duda consultanos!
Saludos!

closed 2019-06-24 by gonzaabel