paxapos/fiscalberry #93: No puedo enviar comandos a una tm-t900fa via USB

jobiols, 2019-06-23:
Este hilo viene del #91 empece otro porque cambio el asunto.

Tengo conectada una epson tm-t900fa por USB y la veo en el sistema con el comando lsusb:
Bus 001 Device 003: ID 04b8:0201 Seiko Epson Corp.

*Este es mi config.ini*

[IMPRESORA_FISCAL]
marca = Epson
modelo = tm-t900fa
usb_vendor = 04b8
usb_product = 0201
in_ep = 0x82
out_ep = 0x01
driver = Epson

*Este es el error*
2019-06-23 12:27:01,755 - FiscalberryApp - INFO - Preparando Fiscalberry Server
2019-06-23 12:27:01,756 - FiscalberryApp - INFO - La IP privada es 192.168.0.107
2019-06-23 12:27:01,757 - FiscalberryApp - INFO - Iniciando Fiscalberry Server
2019-06-23 12:27:01,759 - FiscalberryApp - INFO - *** Websocket Server Started as HTTP at 127.0.1.1 port 12000***
2019-06-23 12:27:01,759 - FiscalberryApp - INFO - Impresora disponible:
2019-06-23 12:27:01,759 - FiscalberryApp - INFO -   - IMPRESORA_FISCAL
2019-06-23 12:27:01,759 - FiscalberryApp - INFO -       marca: Epson, driver: Epson
2019-06-23 12:27:05,683 - FiscalberryApp - INFO - new connection
2019-06-23 12:27:05,695 - FiscalberryApp - INFO - new connection
2019-06-23 12:27:05,697 - FiscalberryApp - INFO - ----- - -- - - - ---
2019-06-23 12:27:05,697 - FiscalberryApp - INFO - { "getActualConfig":"" }
2019-06-23 12:27:05,697 - root - INFO - Iniciando procesamiento de json:::: {"getActualConfig": ""}
2019-06-23 12:27:19,665 - FiscalberryApp - INFO - ----- - -- - - - ---
2019-06-23 12:27:19,665 - FiscalberryApp - INFO - {"getLastNumber":"T","printerName":"IMPRESORA_FISCAL"}
2019-06-23 12:27:19,665 - root - INFO - Iniciando procesamiento de json:::: {"getLastNumber": "T", "printerName": "IMPRESORA_FISCAL"}
2019-06-23 12:27:19,672 - root - INFO - mandando comando de impresora
Process MultiprocesingTraductor-1:
Traceback (most recent call last):
  File "/usr/lib/python2.7/multiprocessing/process.py", line 258, in _bootstrap
    self.run()
  File "/home/jobiols/git-repos/fiscalberry/Traductores/TraductoresHandler.py", line 51, in run
    traductor = self.traductorhandler.init_printer_traductor(printerName)
  File "/home/jobiols/git-repos/fiscalberry/Traductores/TraductoresHandler.py", line 238, in init_printer_traductor
    comando = comandoClass(**dictSectionConf)
  File "/home/jobiols/git-repos/fiscalberry/ComandoInterface.py", line 144, in __init__
    self.conector = ConectorDriverComando(self, driver, **kwargs)
  File "/home/jobiols/git-repos/fiscalberry/ConectorDriverComando.py", line 29, in __init__
    self.driver = driverClass(**kwargs)
TypeError: __init__() got an unexpected keyword argument 'usb_product'
inicializando ConectorDriverComando driver de Epson
2019-06-23 12:27:19,678 - root - INFO - inicializando ConectorDriverComando driver de Epson
2019-06-23 12:27:19,678 - root - INFO -   - Driver: Drivers.EpsonDriver.EpsonDriver

Intente poner tambien 
driver = ReceiptUSB pero da otros errores.

gonzaabel, 2019-06-24:
Hola @jobiols 
Tira error de unexpected argument porque el driver tiene que ser ReceiptUSB, aunque para el caso de la impresora fiscal puede que no funcione ese driver.
En ese caso, la impresora debería ser configurada de la siguiente forma:
`
[IMPRESORA_FISCAL]
marca = Epson
modelo = tm-t900fa
path = /dev/ttyUSBX //cambiar la X por el número del puerto que este usando la impresora
driver = Epson
`
Para esto deberás entrar en la carpeta /dev (Linux) que se encuentra en la raíz del sistema para ver un listado de tus dispositivos USB con el comando **ls ttyUSB***,  y en caso de aparecerte 2, podrás desconectar la impresora para ejecutar nuevamente el comando y ver cual de los archivos desaparece.

Y ya con eso tendría que funcionar, coméntanos si funciono o postea el log nuevamente si apareció algún error.

Saludos!!

jobiols, 2019-06-25:
Hola @gonzaabel te agradezco tu preocupacion por el tema.
El problema es que esa impresora no aparece en /dev hice un diff de antes y despues de conectarla y es exactamente igual. Eso pasa tanto en mi PC linux como en la raspberry.
Debe ser un tema de como esta hecha la impresora, porque he visto otras que aparecen en /dev

Por otro lado buscando encontre que aparece como
/dev/bus/usb/001/007

pero si la apago y la prendo entonces cambia a
/dev/bus/usb/001/008

si le pongo este path al ini me tira este error:

2019-06-25 15:09:20,739 - FiscalberryApp - INFO - Preparando Fiscalberry Server
2019-06-25 15:09:20,742 - FiscalberryApp - INFO - La IP privada es 192.168.0.107
2019-06-25 15:09:20,751 - FiscalberryApp - INFO - Iniciando Fiscalberry Server
2019-06-25 15:09:20,759 - FiscalberryApp - INFO - *** Websocket Server Started as HTTP at 127.0.1.1 port 12000***
2019-06-25 15:09:20,760 - FiscalberryApp - INFO - Impresora disponible:
2019-06-25 15:09:20,761 - FiscalberryApp - INFO -   - IMPRESORA_FISCAL
2019-06-25 15:09:20,761 - FiscalberryApp - INFO -       marca: Epson, driver: Epson
2019-06-25 15:09:36,539 - FiscalberryApp - INFO - new connection
2019-06-25 15:09:36,561 - FiscalberryApp - INFO - ----- - -- - - - ---
2019-06-25 15:09:36,561 - FiscalberryApp - INFO - { "getActualConfig":"" }
2019-06-25 15:09:36,562 - root - INFO - Iniciando procesamiento de json:::: {"getActualConfig": ""}
2019-06-25 15:09:52,549 - FiscalberryApp - INFO - ----- - -- - - - ---
2019-06-25 15:09:52,550 - FiscalberryApp - INFO - {"getAvaliablePrinters":""}
2019-06-25 15:09:52,550 - root - INFO - Iniciando procesamiento de json:::: {"getAvaliablePrinters": ""}
2019-06-25 15:10:07,142 - FiscalberryApp - INFO - ----- - -- - - - ---
2019-06-25 15:10:07,143 - FiscalberryApp - INFO - {"getLastNumber":"T","printerName":"IMPRESORA_FISCAL"}
2019-06-25 15:10:07,144 - root - INFO - Iniciando procesamiento de json:::: {"getLastNumber": "T", "printerName": "IMPRESORA_FISCAL"}
2019-06-25 15:10:07,261 - root - INFO - mandando comando de impresora
inicializando ConectorDriverComando driver de Epson
2019-06-25 15:10:07,274 - root - INFO - inicializando ConectorDriverComando driver de Epson
2019-06-25 15:10:07,275 - root - INFO -   - Driver: Drivers.EpsonDriver.EpsonDriver
Process MultiprocesingTraductor-1:
Traceback (most recent call last):
  File "/usr/lib/python2.7/multiprocessing/process.py", line 258, in _bootstrap
    self.run()
  File "/home/jobiols/git-repos/fiscalberry/Traductores/TraductoresHandler.py", line 51, in run
    traductor = self.traductorhandler.init_printer_traductor(printerName)
  File "/home/jobiols/git-repos/fiscalberry/Traductores/TraductoresHandler.py", line 238, in init_printer_traductor
    comando = comandoClass(**dictSectionConf)
  File "/home/jobiols/git-repos/fiscalberry/ComandoInterface.py", line 144, in __init__
    self.conector = ConectorDriverComando(self, driver, **kwargs)
  File "/home/jobiols/git-repos/fiscalberry/ConectorDriverComando.py", line 29, in __init__
    self.driver = driverClass(**kwargs)
  File "/home/jobiols/git-repos/fiscalberry/Drivers/FiscalPrinterDriver.py", line 53, in __init__
    self._serialPort = serial.Serial(port=path, timeout=None, baudrate=speed)
  File "/home/jobiols/.virtualenvs/fiscalberry/local/lib/python2.7/site-packages/serial/serialutil.py", line 240, in __init__
    self.open()
  File "/home/jobiols/.virtualenvs/fiscalberry/local/lib/python2.7/site-packages/serial/serialposix.py", line 272, in open
    self._reconfigure_port(force_update=True)
  File "/home/jobiols/.virtualenvs/fiscalberry/local/lib/python2.7/site-packages/serial/serialposix.py", line 326, in _reconfigure_port
    raise SerialException("Could not configure port: {}".format(msg))
SerialException: Could not configure port: (25, 'Inappropriate ioctl for device')

Alguna idea?

gonzaabel, 2019-06-30:
@jobiols siento la respuesta tardía... Nunca nos sucedió algo como lo que comentas, con distintas impresoras (aunque no hemos tenido ese modelo de epson) siempre al conectarlas las teníamos en el ttyUSB0 o ttyUSB1.

Ese error que te tira, `'Inappropriate ioctl for device'` como bien dice [un usuario de StackOverFlow](https://stackoverflow.com/a/1605533) significa que el path especificado contiene un archivo que no es uno 'tty'.
Probaste si con algún otro dispositivo lo logras ver como ttyUSB?? porque de poder hacerlo, sería tema de la impresora.

gonzaabel, 2019-06-30:
> Debe ser un tema de como esta hecha la impresora, porque he visto otras que aparecen en /dev

Bien, es probable que sea por la impresora. Si tienes otra PC con cualquier otro linux o otra raspberry, prueba para ver si te da el mismo resultado.

commit 226008e1 (2019-11-28): fix #93 ahora fiscalberry es compatible con la TM t900FA
closed 2019-12-12 by alevilar