paxapos/fiscalberry #94: Cual es la impresora de 2da generacion recomendada

jobiols, 2019-07-13:
Estuve probando con la EPSON TM-T900FA y  no pude hacerla funcionar, ya hay algunos mensajes mios en este repositorio al respecto.
Les pregunto entonces cual es la que me recomiendan? ya que logre hacer un cambio por otra maquina.
Quisiera que sea una de segunda generación que tengan ya probada.

gonzaabel, 2019-07-15:
Hola @jobiols 
Tenemos compatibles los comandos de segunda generación de Hasar, testeados con el emulador de la Hasar SMH/PT-1000F.
Y también los comandos de la impresora de segunda generación Sam4s ellix-40F que nos la enviaron para hacerla compatible con Fiscalberry.
Saludos!

jobiols, 2019-07-15:
Hola @gonzaabel gracias por tu respuesta, entonces me muevo a HASAR y despues te cuento

Para recapitular, mi problema era que la impresora no aparece en /dev asi que mi ultima opcion desesperada fue cuando encontre el comando udevadm esto permite escribir reglas que disparan al insertar dispositivos usb.

Agregue una regla:
SUBSYSTEM=="usb", ATTRS{manufacturer}=="EPSON", SYMLINK+="epson_tm-t900fa" GROUP="jobiols"

y esto creó un dispositivo pero en forma de link simbolico o sea ahora tengo
lrwxrwxrwx 1 root root 15 jul 15 13:47 /dev/epson_tm-t900fa -> bus/usb/001/005

Sin embargo cuando arrranco el server me da un error
SerialException: Could not configure port: (25, 'Inappropriate ioctl for device')

Evidentemente no le gusta el link simbolico.  :-)
Bien veo entonces si me cambian la impresora...

gonzaabel, 2019-07-15:
Dale, perfecto.

closed 2019-12-13 by alevilar