paxapos/fiscalberry #97: Desde CERO

giuffma, 2019-12-11:
Hola a Todos.
Mi nombre es Miguel, trabajo en una empresa de servicios de estacionamientos, actualmente usamos impresoras fiscales de primera generacion y tenemos una Epson TM900FA termica comprada recientemente que la quiero acoplar a mi sistema desarrollado en PHP y se me esta complicando muchisimo.
Actualmente con las impresoras viejas logre hacer lo que quiero usando la libreria pyfiscalprinter en Python modifique casi todo el codigo y lo adapte a lo que querias, descuentos, notas de creditos, recibos todo tipo de facturas, etc.
La libreria ficalberry veo que funciona perfectamente, pero estoy muy verde con el tema de websocket y tirarles comando, necesitaria una mano o algun codigo base que me ayuda a conectarla desde Linux que es mi servidor y tirarle comandos basicos para asi empezar armar todo lo que necesito para hacerla funcionar.
Seria de gran ayuda su conocimiento.
Desde ya agradezco muchisimo el tiempo que le puedan dedicar a leerme.
Saludos!

alevilar, 2019-12-12:
Hola Miguel!; 

en la wiki tenemos algo como para comenzar. Es verdad que no esta super cmpleta, pero esta bueno que vayas intentando y nos haces consultas por aca mismo hasta que logres tener fiscalberry funcionando en tu proyecto.

lo primero que te recomiendo es

1- INSTALAR FISCALBERRY y asegurarte que corre y funciona. 
En este punto con windows mucho no te pudo ayudar, con linux si.
Basicamente lo que instalas es un servidor de websockets. No tenes que tener conocimientos en websockets, simplemente instalarlo y listo. 
Tendras que lidiar con las dependencias de python y el SO que uses. pero tenemos algunos installers ya preparado.

2- INCORPORAR FISCALBERRY A TU PROYECTO
La recomendacion es usarlo mediante JS. Tenemos una libreria que podes meter en tu codigo
en [esta seccion de la wiki ](https://github.com/paxapos/fiscalberry/wiki/Comenzando-con-fiscalberry) explicamos como enviar un comando dailyClose desde JS.

3- GENERAR COMANDOS Y MANDARLOS POR EL WS
[En este link a la wiki ](https://github.com/paxapos/fiscalberry/wiki/JSON-Comandos-Para-impresora-Fiscal)tenes todos los comandos soportados por una impresora fiscal
Lamentablemente la 2gen de la Epson TM900FA no es 100% soportada porque la gente de Epson no nos ayudó mucho a que hagamos compatible su impresora. 
Cosa que con Hasar nos encontramos con una mejor implementación de la 2gen gracias a que nos facilitaron mejor ayuda.

giuffma, 2019-12-12:
Gracias Alejandro.
En estos momentos estoy jugando un poco con eso. Yo uso Linux todos los serves que uso son linux Centos 7.
Siempre al ejecutar python server.py se queda aca no se si es normal

2019-12-12 13:50:49,026 - FiscalberryApp - INFO - Preparando Fiscalberry Server
2019-12-12 13:50:49,029 - FiscalberryApp - INFO - La IP privada es 10.30.83.126
2019-12-12 13:50:49,032 - FiscalberryApp - INFO - Iniciando Fiscalberry Server
2019-12-12 13:50:49,089 - FiscalberryApp - INFO - *** Websocket Server Started as HTTP at 127.0.0.1 port 12000***
2019-12-12 13:50:49,090 - FiscalberryApp - INFO - Impresora disponible:
2019-12-12 13:50:49,090 - FiscalberryApp - INFO -   - IMPRESORA_FISCAL
2019-12-12 13:50:49,091 - FiscalberryApp - INFO -       marca: Epson, driver: Epson

gonzaabel, 2019-12-12:
Hola Miguel, con la consola así abierta debería estar todo bien. Prueba a entrar desde localhost:12000 a ver si te deja conectar (en la consola aparecería 'new connection'). Si tu servidor es solo en modo CLI sin interfaz de usuario, podrías conectarte en la misma LAN desde otro dispositivo, incluso desde tu celular a la ip del server con protocolo http:  http://10.30.83.126:12000

gonzaabel, 2019-12-12:
Una vez conectado estarás en una página de prueba en la cual podrás enviar comandos JSON al servidor usando los botones y modificando el mismo desde el cuadro de texto.

giuffma, 2019-12-12:
Gonzaabel gracias por la pronta respuesta te cuenta lo que veo
Abri la pagina html que esta en el ejemplo y veo esto
Fiscalberry WS Hello World
OFFLINE
host: 
10.30.83.126

port: 
12000

uri: 
/ws

Mensajes
[MSG] reconectando en 3 segundos
[MSG] ***Connection Closed***
[MSG] reconectando en 3 segundos
[MSG] ***Connection Closed***
[MSG] reconectando en 3 segundos
[MSG] ***Connection Closed***
[MSG] reconectando en 3 segundos
[MSG] ***Connection Closed***
[MSG] reconectando en 3 segundos
[MSG] ***Connection Closed***
[MSG] reconectando en 3 segundos
[MSG] ***Connection Closed***
[MSG] reconectando en 3 segundos
[MSG] ***Connection Closed***
[MSG] reconectando en 3 segundos
[MSG] ***Connection Closed***
[MSG] reconectando en 3 segundos
[MSG] ***Connection Closed***
[MSG] reconectando en 3 segundos
[MSG] ***Connection Closed***
[MSG] reconectando en 3 segundos
[MSG] ***Connection Closed***
[MSG] reconectando en 3 segundos
[MSG] ***Connection Closed***
[MSG] reconectando en 3 segundos
[MSG] ***Connection Closed***

giuffma, 2019-12-12:
Intento ver la direcion http://10.30.83.126:12000 y me dice que no puedo acceder lo que se me ocuerre que el puerto 12000 este cerrado
el config.ini lo tengo configurador asi:

[SERVIDOR]
puerto = 12000
discover_url =
ssl_crt_path =
ssl_key_path =
ip_privada = 10.30.83.126
uuid = 00241d3335f9

[IMPRESORA_FISCAL]
marca = Epson
modelo = ["tickeadoras", "epsonlx300+", "tm-2000-af", "tm-220-af", "tm-t900fa", "sm-srp-270"]
path = /dev/ttyUSB0
driver = Epson

gonzaabel, 2019-12-12:
El config.ini esta bien configurado. 
Si te muestra en el navegador un mensaje de connection refused, esto sucede cuando el server no esta en ejecución o cuando hay un bloqueo de puerto (debido a un firewall)

alevilar, 2019-12-12:
1ro
asegurate que tu centos tenga abierto el puerto 12000

2do
la cofiguracion para la epson TM900FA es simplemente esto

[IMPRESORA_FISCAL]
marca = Epson2Gen
modelo = tm-t900fa

3ero
copia la libreria de epson
https://github.com/paxapos/fiscalberry/tree/master/epsonlib/lin
si tu sistema es 32 bits va en /lib/libEpsonFiscalInterface.so
<https://github.com/paxapos/fiscalberry/blob/master/epsonlib/lin/64/libEpsonFiscalInterface.so>
si tu sistema es 64 va en /lib64/libEpsonFiscalInterface.so
<https://github.com/paxapos/fiscalberry/blob/master/epsonlib/lin/64/libEpsonFiscalInterface.so>

La impresora T900FA no usa el ttyUSB0 porque el USB no es del tipo serial
aparece conectada como */dev/usb/lp0* (verificalo)

si tenes otra impresora configurada te puede aparecer como *lp1*

en tal caso  tu config seria
[IMPRESORA_FISCAL]
marca = Epson2Gen
modelo = tm-t900fa
path = /dev/usb/lp1

joseluisbossio, 2019-12-12:
Deberían probar que primero responda un TELNET IP PORT

si eso no responde... revisar firewall...

giuffma, 2019-12-12:
> marca = Epson2Gen modelo = tm-t900fa path = /dev/usb/lp1

Gracias Ale.
Ya configure todo.
Estoy tratando de abrir el puerto 12000 y no se que pasa que no o abre.
Tambien no tengo permissos en /dev/usb/lp1 ahora voy a ponerlo

giuffma, 2019-12-12:
> […](#)

Si. Gracias. Estoy viendo como puedo abrirlo

giuffma, 2019-12-12:
> 1ro asegurate que tu centos tenga abierto el puerto 12000 2do la cofiguracion para la epson TM900FA es simplemente esto [IMPRESORA_FISCAL] marca = Epson2Gen modelo = tm-t900fa 3ero copia la libreria [...]

Es raro esta abierto
[root@trinidad fiscalberry-master]# nmap localhost -p 12000 | grep -i tcp
12000/tcp open  cce4x

giuffma, 2019-12-12:
Bueno ahora cambio un poco
Pude habilitar el 12000 me ingresa a una pagina de prueba pero me sigue tirando offline

Fiscalberry WS Hello World
OFFLINE
host: 
localhost

port: 
12000

uri: 
/ws

Mensajes
[MSG] reconectando en 3 segundos
[MSG] ***Connection Closed***
[MSG] reconectando en 3 segundos
[MSG] ***Connection Closed***
[MSG] reconectando en 3 segundos
[MSG] ***Connection Closed***
[MSG] reconectando en 3 segundos
[MSG] ***Connection Closed***
[MSG] reconectando en 3 segundos
[MSG] ***Connection Closed***
[MSG] reconectando en 3 segundos
[MSG] ***Connection Closed***
[MSG] reconectando en 3 segundos
[MSG] ***Connection Closed***
[MSG] reconectando en 3 segundos
[MSG] ***Connection Closed***

gonzaabel, 2019-12-12:
Estas ingresando desde el mismo PC servidor o desde otro en la misma intranet? porque veo que en servidor tienes localhost en lugar de la ip.

giuffma, 2019-12-12:
Gonza. Estoy entrando desde otra PC de la intranet.
No desde el mismo servidor

gonzaabel, 2019-12-12:
Bien, en el copy-paste que pegaste antes, por lo visto te falto cambiar localhost por el ip del server. Y con eso ya debería andar.

giuffma, 2019-12-12:
Ya funco!!!!
La tengo online! el tema que le tiro comandos y no imprime nada!

giuffma, 2019-12-12:
La pantalla se ve asi:

[image]

gonzaabel, 2019-12-12:
Escribenos el JSON que estas enviándole y también el log de la consola si muestra algún error para que podamos ver que podría ser :nerd_face:

gonzaabel, 2019-12-12:
Ok, lo primero que tienes que hacer es dar click en el primer botón de la izquierda llamado "Listar Impresoras" y eso te va a escribir un JSON debajo con el comando getAvailablePrinters. Le das al botón azul de Enviar y te lista todas las impresoras configuradas

gonzaabel, 2019-12-12:
Te había escrito lo de listar impresoras pero ahora que me fije ya lo habías hecho :sweat_smile:

giuffma, 2019-12-12:
> Ok, lo primero que tienes que hacer es dar click en el primer botón de la izquierda llamado "Listar Impresoras" y eso te va a escribir un JSON debajo con el comando getAvailablePrinters. Le das al botón azul de Enviar y te lista todas las impresoras configuradas

Hasta ahi perfe

giuffma, 2019-12-12:
> Te había escrito lo de listar impresoras pero ahora que me fije ya lo habías hecho 😅

Si si! eso lo habia entendido. pero por ejemplo le tiro un X y nada.

gonzaabel, 2019-12-12:
al apretar el boton de Cierre X te escribe el JSON debajo para enviar? lo pregunto porque veo el textarea del JSON vacio.
Si al apretar cualquiera de los botones grises no te escribe nada en el textarea, tienes que refrescar con F5 y probar de nuevo.

giuffma, 2019-12-12:
> al apretar el boton de Cierre X te escribe el JSON debajo para enviar? lo pregunto porque veo el textarea del JSON vacio.
> Si al apretar cualquiera de los botones grises no te escribe nada en el textarea, tienes que refrescar con F5 y probar de nuevo.

Si eso hice me queda asi
[image]

gonzaabel, 2019-12-12:
habría que ver el servidor si tira por consola algún error

giuffma, 2019-12-12:
> habría que ver el servidor si tira por consola algún error

en donde lo tendria que ver?
en /var/log/httpd/error_log? yo ahi miro todos los errores del apache.

gonzaabel, 2019-12-12:
me refería en la consola del servidor de python de fiscalberry. Es probable que ahí este lanzando un error.

giuffma, 2019-12-12:
> me refería en la consola del servidor de python de fiscalberry. Es probable que ahí este lanzando un error.
Ok
Perdon no me habia fijado en esto

2019-12-12 15:27:30,745 - FiscalberryApp - INFO - connection closed
2019-12-12 15:27:30,747 - FiscalberryApp - INFO - connection closed
2019-12-12 15:27:42,311 - FiscalberryApp - INFO - new connection
2019-12-12 15:27:42,315 - FiscalberryApp - INFO - ----- - -- - - - ---
2019-12-12 15:27:42,315 - FiscalberryApp - INFO - { "getActualConfig":"" }
2019-12-12 15:27:42,316 - root - INFO - Iniciando procesamiento de json:::: {"getActualConfig": ""}
2019-12-12 15:27:49,923 - FiscalberryApp - INFO - ----- - -- - - - ---
2019-12-12 15:27:49,923 - FiscalberryApp - INFO - {"getAvaliablePrinters":""}
2019-12-12 15:27:49,924 - root - INFO - Iniciando procesamiento de json:::: {"getAvaliablePrinters": ""}
2019-12-12 15:28:03,436 - FiscalberryApp - INFO - new connection
2019-12-12 15:28:03,442 - FiscalberryApp - INFO - ----- - -- - - - ---
2019-12-12 15:28:03,442 - FiscalberryApp - INFO - { "getActualConfig":"" }
2019-12-12 15:28:03,443 - root - INFO - Iniciando procesamiento de json:::: {"getActualConfig": ""}
2019-12-12 15:28:27,212 - FiscalberryApp - INFO - ----- - -- - - - ---
2019-12-12 15:28:27,212 - FiscalberryApp - INFO - {"dailyClose":"X","printerName":"IMPRESORA_FISCAL"}
2019-12-12 15:28:27,213 - root - INFO - Iniciando procesamiento de json:::: {"printerName": "IMPRESORA_FISCAL", "dailyClose": "X"}
2019-12-12 15:28:27,219 - root - INFO - mandando comando de impresora
inicializando ConectorDriverComando driver de Epson
2019-12-12 15:28:27,235 - root - INFO - inicializando ConectorDriverComando driver de Epson
2019-12-12 15:28:27,236 - root - INFO -   - Driver: Drivers.EpsonDriver.EpsonDriver
Process Process-4:
Traceback (most recent call last):
  File "/usr/lib64/python2.7/multiprocessing/process.py", line 258, in _bootstrap
    self.run()
  File "/usr/lib64/python2.7/multiprocessing/process.py", line 114, in run
    self._target(*self._args, **self._kwargs)
  File "/root/fiscalberry-master/Traductores/TraductoresHandler.py", line 65, in runTraductor
    traductor = init_printer_traductor(printerName)
  File "/root/fiscalberry-master/Traductores/TraductoresHandler.py", line 57, in init_printer_traductor
    comando = comandoClass(**dictSectionConf)
  File "/root/fiscalberry-master/ComandoInterface.py", line 144, in __init__
    self.conector = ConectorDriverComando(self, driver, **kwargs)
  File "/root/fiscalberry-master/ConectorDriverComando.py", line 29, in __init__
    self.driver = driverClass(**kwargs)
  File "/root/fiscalberry-master/Drivers/FiscalPrinterDriver.py", line 53, in __init__
    self._serialPort = serial.Serial(port=path, timeout=None, baudrate=speed)
  File "/usr/lib/python2.7/site-packages/serial/serialutil.py", line 261, in __init__
    self.open()
  File "/usr/lib/python2.7/site-packages/serial/serialposix.py", line 282, in open
    self._reconfigurePort()
  File "/usr/lib/python2.7/site-packages/serial/serialposix.py", line 311, in _reconfigurePort
    raise SerialException("Could not configure port: %s" % msg)
SerialException: Could not configure port: (25, 'Inappropriate ioctl for device')

giuffma, 2019-12-12:
Me debe estas faltando alguna configuracion?
Creo haber realizado todo bien.

alevilar, 2019-12-12:
manda la configuracion nuevamente a ver como esta.

y trata de descargarte la ultima version del master, subi unas pequeñas actualizaciones hoy. pero que puden resultarte utiles

giuffma, 2019-12-12:
Otra cosa. lo que es la carpeta fiscalberry-master donde estan drivers y demas debe tener un lugar especial en Linux? no sea cosa que no la puse en un lugar especial

giuffma, 2019-12-12:
> manda la configuracion nuevamente a ver como esta.
> 
> y trata de descargarte la ultima version del master, subi unas pequeñas actualizaciones hoy. pero que puden resultarte utiles

Alejandro.
Segui tus tres pasos.
y el archivo de configuracion es este:

[SERVIDOR]
puerto = 12000
discover_url =
ssl_crt_path =
ssl_key_path =
ip_privada = 10.30.83.126
uuid = 00241d3335f9

[IMPRESORA_FISCAL]
marca = Epson
modelo = tm-t900fa
path = /dev/usb/lp1
driver = Epson

giuffma, 2019-12-12:
Descargue nuevamente y copie al servidor los fuentes. F5 a la web e intente hacer un X
Esto tiro

2019-12-12 16:18:13,665 @ FiscalberryApp [INFO]: Request
 -> {"dailyClose":"X","printerName":"IMPRESORA_FISCAL"}
2019-12-12 16:18:13,666 @ root [INFO]: Iniciando procesamiento de json:::: {"printerName": "IMPRESORA_FISCAL", "dailyClose": "X"}
2019-12-12 16:18:13,676 @ root [INFO]: mandando comando de impresora
inicializando ConectorDriverComando driver de Epson
2019-12-12 16:18:13,714 @ root [INFO]: inicializando ConectorDriverComando driver de Epson
2019-12-12 16:18:13,717 @ root [INFO]:   - Driver: Drivers.EpsonDriver.EpsonDriver
Process Process-1:
Traceback (most recent call last):
  File "/usr/lib64/python2.7/multiprocessing/process.py", line 258, in _bootstrap
    self.run()
  File "/usr/lib64/python2.7/multiprocessing/process.py", line 114, in run
    self._target(*self._args, **self._kwargs)
  File "/root/fiscalberry-master/Traductores/TraductoresHandler.py", line 67, in runTraductor
    traductor = init_printer_traductor(printerName)
  File "/root/fiscalberry-master/Traductores/TraductoresHandler.py", line 59, in init_printer_traductor
    comando = comandoClass(**dictSectionConf)
  File "/root/fiscalberry-master/ComandoInterface.py", line 144, in __init__
    self.conector = ConectorDriverComando(self, driver, **kwargs)
  File "/root/fiscalberry-master/ConectorDriverComando.py", line 29, in __init__
    self.driver = driverClass(**kwargs)
  File "/root/fiscalberry-master/Drivers/FiscalPrinterDriver.py", line 59, in __init__
    self._serialPort = serial.Serial(port=path, timeout=None, baudrate=speed)
  File "/usr/lib/python2.7/site-packages/serial/serialutil.py", line 261, in __init__
    self.open()
  File "/usr/lib/python2.7/site-packages/serial/serialposix.py", line 282, in open
    self._reconfigurePort()
  File "/usr/lib/python2.7/site-packages/serial/serialposix.py", line 311, in _reconfigurePort
    raise SerialException("Could not configure port: %s" % msg)
SerialException: Could not configure port: (25, 'Inappropriate ioctl for device')
2019-12-12 16:18:13,725 @ FiscalberryApp [INFO]: Response
 <- {'rta': ''}

alevilar, 2019-12-12:
estas seguro que esta en lp1?

fijate de hacer 
```bash
cd /dev/usb
ls -la
```

y manda foto a ver que hay en esa carpeta

giuffma, 2019-12-12:
> estas seguro que esta en lp1?
> 
> fijate de hacer
> 
> ```shell
> cd /dev/usb
> ls -la
> ```
> 
> y manda foto a ver que hay en esa carpeta

Si tengo los dos el lp0 es una termica epson no fiscal

0 drwxr-xr-x  2 root root     80 dic 12 16:38 .
0 drwxr-xr-x 21 root root   3280 dic 12 16:38 ..
0 crw-rw----  1 root lp   180, 0 dic 12 16:38 lp0
0 crw-rw----  1 root lp   180, 1 dic 12 16:38 lp1

giuffma, 2019-12-12:
Ahora empece a revisar y veo que no se instalaron las siguientes dependencias
python-escpos
python-nmap
pyutf8
qrcode
requests
singledispatch
six
tornado
typing
urllib3
PyJWT
cryptography

giuffma, 2019-12-12:
Centos 7 no las encuentra esas dependencia.

giuffma, 2019-12-12:
Luche y me quedaron estas dependencias sin bajar
python-escpos-
pyutf8-
qrcode-
typing-

giuffma, 2019-12-12:
Despues de instalar esas dependencias volvi a ejecutar todo igualmente me salen estos mensajes

2019-12-12 17:09:39,385 @ FiscalberryApp [INFO]: Request
 -> {"dailyClose":"X","printerName":"IMPRESORA_FISCAL"}
2019-12-12 17:09:39,385 @ root [INFO]: Iniciando procesamiento de json:::: {"printerName": "IMPRESORA_FISCAL", "dailyClose": "X"}
2019-12-12 17:09:39,399 @ root [INFO]: mandando comando de impresora
inicializando ConectorDriverComando driver de Epson
2019-12-12 17:09:39,413 @ root [INFO]: inicializando ConectorDriverComando driver de Epson
2019-12-12 17:09:39,415 @ root [INFO]:   - Driver: Drivers.EpsonDriver.EpsonDriver
Process Process-1:
Traceback (most recent call last):
  File "/usr/lib64/python2.7/multiprocessing/process.py", line 258, in _bootstrap
    self.run()
  File "/usr/lib64/python2.7/multiprocessing/process.py", line 114, in run
    self._target(*self._args, **self._kwargs)
  File "/root/fiscalberry-master/Traductores/TraductoresHandler.py", line 67, in runTraductor
    traductor = init_printer_traductor(printerName)
  File "/root/fiscalberry-master/Traductores/TraductoresHandler.py", line 59, in init_printer_traductor
    comando = comandoClass(**dictSectionConf)
  File "/root/fiscalberry-master/ComandoInterface.py", line 144, in __init__
    self.conector = ConectorDriverComando(self, driver, **kwargs)
  File "/root/fiscalberry-master/ConectorDriverComando.py", line 29, in __init__
    self.driver = driverClass(**kwargs)
  File "/root/fiscalberry-master/Drivers/FiscalPrinterDriver.py", line 59, in __init__
    self._serialPort = serial.Serial(port=path, timeout=None, baudrate=speed)
  File "/usr/lib/python2.7/site-packages/serial/serialutil.py", line 261, in __init__
    self.open()
  File "/usr/lib/python2.7/site-packages/serial/serialposix.py", line 282, in open
    self._reconfigurePort()
  File "/usr/lib/python2.7/site-packages/serial/serialposix.py", line 311, in _reconfigurePort
    raise SerialException("Could not configure port: %s" % msg)
SerialException: Could not configure port: (25, 'Inappropriate ioctl for device')
2019-12-12 17:09:39,424 @ FiscalberryApp [INFO]: Response
 <- {'rta': ''}

alevilar, 2019-12-12:
por lo que veo te estas conectando usando EpsonDriver, cuando deberia ser
Epson2GenDriver

eso lo seteas desde el config.ini.

donde pones

marca = Epson2Gen
modelo = tm-t900fa
driver = Epson2Gen

alevilar, 2019-12-12:
Proba poniendo estco en tu config.ini de tu fiscal

```ini
marca = Epson2Gen
modelo = tm-t900fa
driver = Epson2Gen
path = serial: /dev/usb/lp0
```

giuffma, 2019-12-12:
> […](#)

Alejandro. Ahi lo cambie.
Pero me sigue tirando error ahora es distinto.

2019-12-12 20:01:28,325 @ FiscalberryApp [INFO]: Request
 -> {"dailyClose":"X","printerName":"IMPRESORA_FISCAL"}
2019-12-12 20:01:28,326 @ root [INFO]: Iniciando procesamiento de json:::: {"printerName": "IMPRESORA_FISCAL", "dailyClose": "X"}
2019-12-12 20:01:28,336 @ root [INFO]: mandando comando de impresora
inicializando ConectorDriverComando driver de Epson2Gen
2019-12-12 20:01:28,358 @ root [INFO]: inicializando ConectorDriverComando driver de Epson2Gen
Process Process-1:
Traceback (most recent call last):
  File "/usr/lib64/python2.7/multiprocessing/process.py", line 258, in _bootstrap
    self.run()
  File "/usr/lib64/python2.7/multiprocessing/process.py", line 114, in run
    self._target(*self._args, **self._kwargs)
  File "/root/fiscalberry-master/Traductores/TraductoresHandler.py", line 67, in runTraductor
    traductor = init_printer_traductor(printerName)
  File "/root/fiscalberry-master/Traductores/TraductoresHandler.py", line 59, in init_printer_traductor
    comando = comandoClass(**dictSectionConf)
  File "/root/fiscalberry-master/ComandoInterface.py", line 144, in __init__
    self.conector = ConectorDriverComando(self, driver, **kwargs)
  File "/root/fiscalberry-master/ConectorDriverComando.py", line 25, in __init__
    driverModule = importlib.import_module(libraryName)
  File "/usr/lib64/python2.7/importlib/__init__.py", line 37, in import_module
    __import__(name)
  File "/root/fiscalberry-master/Drivers/Epson2GenDriver.py", line 21, in <module>
    EpsonLibInterface = ctypes.cdll.LoadLibrary(fullpath)
  File "/usr/lib64/python2.7/ctypes/__init__.py", line 438, in LoadLibrary
    return self._dlltype(name)
  File "/usr/lib64/python2.7/ctypes/__init__.py", line 360, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: /lib64/libEpsonFiscalInterface.so: wrong ELF class: ELFCLASS32
2019-12-12 20:01:28,383 @ FiscalberryApp [INFO]: Response
 <- {'rta': ''}

alevilar, 2019-12-12:
ya vamos mucho mejor !! eso es porque esta llamando a la libreria de 32 en lugar de la de 64. o al reves...

wrong ELF class: ELFCLASS32

gonzaabel, 2019-12-12:
de hecho ese error es porque se esta usando la librería inadecuada para la arquitectura del sistema operativo

giuffma, 2019-12-12:
Ahh Genial, si si vi el OSERROR y me parecio que apuntaba ahi.

giuffma, 2019-12-12:
Calculo que es 64
 Operating System: CentOS Linux 7 (Core)
       CPE OS Name: cpe:/o:centos:centos:7
            Kernel: Linux 3.10.0-1062.4.1.el7.x86_64
      Architecture: x86-64
Voy a volver a descargar el archivo .so de 64B y lo vuelvo a copiar
ahi les digo

gonzaabel, 2019-12-12:
sí, es de 64 bits.

giuffma, 2019-12-12:
Bueno nos vamos acercando. Debe ser ya cuestion de driver miren

2019-12-12 20:15:11,558 @ FiscalberryApp [INFO]: Request
 -> {"dailyClose":"X","printerName":"IMPRESORA_FISCAL"}
2019-12-12 20:15:11,559 @ root [INFO]: Iniciando procesamiento de json:::: {"printerName": "IMPRESORA_FISCAL", "dailyClose": "X"}
2019-12-12 20:15:11,565 @ root [INFO]: mandando comando de impresora
inicializando ConectorDriverComando driver de Epson2Gen
2019-12-12 20:15:11,580 @ root [INFO]: inicializando ConectorDriverComando driver de Epson2Gen
Process Process-3:
Traceback (most recent call last):
  File "/usr/lib64/python2.7/multiprocessing/process.py", line 258, in _bootstrap
    self.run()
  File "/usr/lib64/python2.7/multiprocessing/process.py", line 114, in run
    self._target(*self._args, **self._kwargs)
  File "/root/fiscalberry-master/Traductores/TraductoresHandler.py", line 67, in runTraductor
    traductor = init_printer_traductor(printerName)
  File "/root/fiscalberry-master/Traductores/TraductoresHandler.py", line 59, in init_printer_traductor
    comando = comandoClass(**dictSectionConf)
  File "/root/fiscalberry-master/ComandoInterface.py", line 144, in __init__
    self.conector = ConectorDriverComando(self, driver, **kwargs)
  File "/root/fiscalberry-master/ConectorDriverComando.py", line 25, in __init__
    driverModule = importlib.import_module(libraryName)
  File "/usr/lib64/python2.7/importlib/__init__.py", line 37, in import_module
    __import__(name)
  File "/root/fiscalberry-master/Drivers/Epson2GenDriver.py", line 21, in <module>
    EpsonLibInterface = ctypes.cdll.LoadLibrary(fullpath)
  File "/usr/lib64/python2.7/ctypes/__init__.py", line 438, in LoadLibrary
    return self._dlltype(name)
  File "/usr/lib64/python2.7/ctypes/__init__.py", line 360, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: /lib64/libstdc++.so.6: version `GLIBCXX_3.4.20' not found (required by /lib64/libEpsonFiscalInterface.so)
2019-12-12 20:15:11,601 @ FiscalberryApp [INFO]: Response
 <- {'rta': ''}

alevilar, 2019-12-12:
entonces fijate que tenes que copiar la libreria que esta dentro de
/epsonlib/lin/64/libEpsonFiscalInterface.so
<https://github.com/paxapos/fiscalberry/blob/master/epsonlib/lin/64/libEpsonFiscalInterface.so>

te esta tomando mal esa libreria externa

alevilar, 2019-12-12:
ahi ya me mataste. parece que te falta alguna libreria de c++

pero vas avanzado

giuffma, 2019-12-13:
> […](#)

Buen dia.
Al parecer es la version de Centos asi que voy a ponerle una version mas y volvere a probar todo.
A ver si imprime.

joseluisbossio, 2019-12-13:
Ubuntu y sale funcionando en 10 min, luego cuando aprendas cómo
funcionar... Vas a otros OS

gonzaabel, 2019-12-13:
OSError: /lib64/libstdc++.so.6: version `GLIBCXX_3.4.20' not found (required by /lib64/libEpsonFiscalInterface.so)

Ese directorio lib64 lo tienes dentro del proyecto fiscalberry o en la raíz de tu sistema? porque me parece que ese error se debe porque la librería no esta en el directorio donde van las demás librerias del sistema. (si no me equivoco)...

giuffma, 2019-12-13:
Hola. No la lib64 es del sistema linux esta en el raiz
El proyecto lo tengo puesto en root. Igualmente el GLIBCXX_3.4.20 no esta en Centos 7
Estoy instalando Ubuntu para implementarlo

giuffma, 2019-12-13:
Estimados.
Instale Ubuntu y me encontre con nuevos errores.root@prueba-System-Product-Name:~/fiscalberry-master# sudo python server.py
Traceback (most recent call last):
  File "server.py", line 7, in <module>
    from FiscalberryApp import FiscalberryApp
  File "/root/fiscalberry-master/FiscalberryApp.py", line 6, in <module>
    from Traductores.TraductoresHandler import TraductoresHandler, TraductorException
  File "/root/fiscalberry-master/Traductores/TraductoresHandler.py", line 11, in <module>
    import git
  File "/usr/local/lib/python2.7/dist-packages/git/__init__.py", line 85, in <module>
    raise ImportError('Failed to initialize: {0}'.format(exc))
ImportError: Failed to initialize: Bad git executable.
The git executable must be specified in one of the following ways:
    - be included in your $PATH
    - be set via $GIT_PYTHON_GIT_EXECUTABLE
    - explicitly set via git.refresh()

All git commands will error until this is rectified.

This initial warning can be silenced or aggravated in the future by setting the
$GIT_PYTHON_REFRESH environment variable. Use one of the following values:
    - quiet|q|silence|s|none|n|0: for no warning or exception
    - warn|w|warning|1: for a printed warning
    - error|e|raise|r|2: for a raised exception

Example:
    export GIT_PYTHON_REFRESH=quiet

ejecute el server.py y aparecio esto

gonzaabel, 2019-12-13:
parece ser como que no esta instalado git: sudo apt-get install git o no esta instalado Git-Python (que se instala con las demás dependencias al ejecutar sudo pip install -r requirements.txt)

giuffma, 2019-12-13:
Gonza. Era eso. Gracias Problema menos.
Sigo a ver logro hacer un impresion

giuffma, 2019-12-13:
Bueno ahora me gusta un poco mas creo que estoy ahi de imprimir

2019-12-13 15:17:49,560 @ FiscalberryApp [INFO]: Connection Established
2019-12-13 15:17:49,567 @ FiscalberryApp [INFO]: Request
 -> { "getActualConfig":"" }
2019-12-13 15:17:49,568 @ root [INFO]: Iniciando procesamiento de json:::: {"getActualConfig": ""}
2019-12-13 15:17:49,568 @ FiscalberryApp [INFO]: Response
 <- {'rta': {'action': 'getActualConfig', 'rta': {'SERVIDOR': {'ssl_crt_path': '', 'uuid': 'f46d04cead83', 'ssl_key_path': '', 'discover_url': '', 'puerto': '12000', 'ip_privada': '10.30.83.126'}, 'IMPRESORA_FISCAL': {'driver': 'Epson2Gen', 'modelo': 'tm-t900fa', 'marca': 'Epson2Gen', 'path': 'serial: /dev/lp0'}}}}
2019-12-13 15:17:49,740 @ FiscalberryApp [INFO]: Connection Established
2019-12-13 15:17:49,744 @ FiscalberryApp [INFO]: Request
 -> { "getActualConfig":"" }
2019-12-13 15:17:49,745 @ root [INFO]: Iniciando procesamiento de json:::: {"getActualConfig": ""}
2019-12-13 15:17:49,745 @ FiscalberryApp [INFO]: Response
 <- {'rta': {'action': 'getActualConfig', 'rta': {'SERVIDOR': {'ssl_crt_path': '', 'uuid': 'f46d04cead83', 'ssl_key_path': '', 'discover_url': '', 'puerto': '12000', 'ip_privada': '10.30.83.126'}, 'IMPRESORA_FISCAL': {'driver': 'Epson2Gen', 'modelo': 'tm-t900fa', 'marca': 'Epson2Gen', 'path': 'serial: /dev/lp0'}}}}
2019-12-13 15:17:55,442 @ FiscalberryApp [INFO]: Request
 -> {"getAvaliablePrinters":""}
2019-12-13 15:17:55,442 @ root [INFO]: Iniciando procesamiento de json:::: {"getAvaliablePrinters": ""}
2019-12-13 15:17:55,443 @ FiscalberryApp [INFO]: Response
 <- {'rta': {'action': 'getAvaliablePrinters', 'rta': ['IMPRESORA_FISCAL']}}
2019-12-13 15:18:20,604 @ FiscalberryApp [INFO]: Request
 -> {"dailyClose":"X","printerName":"IMPRESORA_FISCAL"}
2019-12-13 15:18:20,605 @ root [INFO]: Iniciando procesamiento de json:::: {"printerName": "IMPRESORA_FISCAL", "dailyClose": "X"}
2019-12-13 15:18:20,642 @ root [INFO]: mandando comando de impresora
inicializando ConectorDriverComando driver de Epson2Gen
2019-12-13 15:18:20,741 @ root [INFO]: inicializando ConectorDriverComando driver de Epson2Gen
2019-12-13 15:18:21,011 @ root [INFO]:   - Driver: Drivers.Epson2GenDriver.Epson2GenDriver
-------------------------
*************************
   EPSON FISCAL
*************************
-------------------------

Se queda aca y no imprime. Lo unico raro que encuentro es que los puertos lp0 y lp1 estan en /dev
No en /dev/usb

giuffma, 2019-12-13:
Lo probe con el lp0 y lp1 y nada

[4 more replies]

closed 2020-06-05 by alevilar