paxapos/fiscalberry #99: Integrar en sistema PHP

juanmarinelli, 2019-12-12:
Hola buenas tardes, estoy desarrollando un sistema en PHP para un cafe/bar (Argentina), donde actualmente están trabajando con otro sistema y con un controlador fiscal HASAR, quería integrar fiscalberry al sistema para imprimir los tickets fiscales, pero no se si el sistema al estar desarrollado en PHP puede integrarse correctamente con esta libreria. El sistema va a correr en un servidor web pago, y se utilizarán en varios puntos de venta, cada uno con su controlador fiscal localmente instalado en cada maquina. 
He leido bastante toda la info que brindan pero todavía no logro entender como puedo utilizarla con php o js. Espero que puedan orientarme un poco y si esta libreria me serviria para lo que necesito.
Muchas gracias de antemano! 
Saludoss

juanmarinelli, 2019-12-13:
Solucionado, cambie args=('/tmp/fiscal_printer_driver.log',) -> args=('fiscal_printer_driver.log', 'a') de [handler_fileHandlerFiscalPrinterDriver]

closed 2019-12-13 by juanmarinelli

joseluisbossio, 2019-12-13:
5$