paxapos/fiscalberry #101: Problema con self.comando.start() al imprimir ticket

juanmarinelli, 2019-12-15:
Hola buenas tardes estoy probando el aplicativo, y al hacer un cierre X me funciona perfecto, pero al querer enviar a imprimir un ticket me arroja el siguiente problema 

**2019-12-15 16:46:04,186 @ FiscalberryApp [INFO]: Request 
 -> {"printTicket":{"encabezado":{"tipo_cbte":"T"},"items":[{"alic_iva":21,"importe":0.01,"ds":"PEPSI","qty":1},{"alic_iva":21,"importe":0.12,"ds":"COCA","qty":2}]},"printerName":"IMPRESORA_FISCAL"}
2019-12-15 16:46:04,187 @ root [INFO]: Iniciando procesamiento de json:::: {"printerName": "IMPRESORA_FISCAL", "printTicket": {"encabezado": {"tipo_cbte": "T"}, "items": [{"alic_iva": 21, "qty": 1, "ds": "PEPSI", "importe": 0.01}, {"alic_iva": 21, "qty": 2, "ds": "COCA", "importe": 0.12}]}}
2019-12-15 16:46:04,196 @ root [INFO]: mandando comando de impresora
inicializando ConectorDriverComando driver de File
2019-12-15 16:46:04,216 @ root [INFO]: inicializando ConectorDriverComando driver de File
2019-12-15 16:46:04,218 @ root [INFO]:   - Driver: Drivers.FileDriver.FileDriver
Process Process-27:
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/multiprocessing/process.py", line 267, in _bootstrap
    self.run()
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/multiprocessing/process.py", line 114, in run
    self._target(*self._args, **self._kwargs)
  File "/Applications/MAMP/htdocs/fiscalberry/Traductores/TraductoresHandler.py", line 71, in runTraductor
    queue.put(traductor.run(jsonTicket))
  File "/Applications/MAMP/htdocs/fiscalberry/Traductores/TraductorInterface.py", line 19, in run
    res = fnAction(**jsonTicket[action])
  File "/Applications/MAMP/htdocs/fiscalberry/Traductores/TraductorFiscal.py", line 45, in printTicket
    self.comando.start()
AttributeError: HasarComandos instance has no attribute 'start'
2019-12-15 16:46:04,225 @ FiscalberryApp [INFO]: Response 
 <- {'rta': ''}**

no puedo encontrar como solucionarlo ya que no se mucho de python y excede mis conocimientos, si pueden darme una mano lo agradeceria

juanmarinelli, 2019-12-16:
alguien que pueda darme una mano? tengo instalado python 2.7.17, me pasa lo mismo en windows y en una macbook, el cierre X lo tira bien, la configuracion de las impresoras es la que esta por defecto y he probado con otros modelos de impreosa y con las epson y tira lo mismo, algo que pueda probar ? Gracias

gonzaabel, 2019-12-17:
Hola @juanmarinelli disculpa la demora, tu error era un bug. Acaba de ser corregido así que descargando el commit más reciente te debería funcionar sin problemas. Caso contrario, puedes reabrir la issue. Gracias!

Saludos!

closed 2019-12-17 by gonzaabel

juanmarinelli, 2019-12-17:
Gracias ya esta funcionando perfecto!

commit 6a93b6f0 (2019-12-17): Fix #101