paxapos/fiscalberry #105: Se puede imprimir una comanda o un documento NO FISCAL en una IMPRESORA FISCAL?

MarianoGomezBidondo, 2020-07-01:
¿Se puede imprimir una comanda o un documento **NO FISCAL** en una **IMPRESORA FISCAL**?

Seria para el caso de un restaurant que tuviera una impresora fiscal unicamente y quiere imprimir comandas por ahi. Se que la impresora lo permite, pero no encontre como hacer que fiscalberry lo haga.

gracias!

joseluisbossio, 2020-07-02:
No, si está inicializada, no.

joseluisbossio, 2020-07-02:
Tenes el manual de la impresora, marca y modelo? Buscamos el manual que
tiene las secuencias de escape.

MarianoGomezBidondo, 2020-07-02:
JoseLuisBossio, gracias por tu respuesta!

Tengo de las 4 principales. Hasar 441, Hasar 250, Epson Tm-T900FA y , Epson TMu220AF.

Raro que a nadie le surgiera la misma necesidad antes.

alevilar, 2020-07-02:
Hola Mariano. No, no se puede. Si quisieras imoprimir un comprobante NO
FISCAL, te saldria una leyenda cada 5 renglones diciendo "NO FISCAL.-NO
FISCAL- NO FISCAL" el cual es re molesto y queda horrible.

la solucion?

Imprimir factura electronica en una impresora de comandas.
Es un poco mas complejo a nivel software (Pero ahi es donde ganarias mas
plata vos 😉)
Pero tu cliente estaria chocho.

tene en cuenta que un controlador fiscal vale alrededor de $60.000,
mientras que una comandera se encuentra por $15/$20.000 una buena buena.

Nosotros en Paxapos estamos ofreciendo esa solucion hace tiempo y cada vez
ganamos mas adeptos a emitir factura electronica en ticket.

También dependerá del rubro de tu cliente si puede o no usar factura
electronica

saludos

joseluisbossio, 2020-07-02:
con la Epson TMu220AF, no, es como dice Ale, 5 renglones y muchos diciendo
"**************" no sirve, quizas para un saludo o algun mensaje.

Lo ideal es comandera.

drakorg, 2020-09-09:
Perdón que me meta, pero el caso de uso existe y es válido. Entiendo que hoy la solución no lo haga, o no haya interés en implementarlo, pero el caso de uso existe y es válido. Muchos sistemas imprimen cierres de turno que van más allá de los x y z, con información operativa no necesariamente fiscal, y sí, tienen esa limitación, pero tampoco es para descartarlo sólo porque aparezca una leyenda cada 5 renglones. Tampoco se pueden usar ciertas palabras como total para evitar armar tickets truchos, pero el caso de uso es válido y estaría bueno que estuviera implementado. Sino obligas a tener una segunda impresora sólo para eso, o como decis vos pasar directamente a factura electrónica, que es totalmente válido, pero bueno, la gente que tenga fiscales podrían imprimir comprobantes no fiscales sin necesidad de una segunda impresora, y todavía son bastantes. Bueno, un comentario nomás. 

Saludos.

alevilar, 2020-09-09:
Es verdad que el caso de uso existe y es valido. Quizas en algunos rubros
sirva y en otros no tanto.

La issue la dejamos abierta, en algun momento intentamos generar un comando
para imprimir texto libre, de hecho creo que hay algo pero no esta muy
testeado. Si tienen ganas de ayudar pueden probar y yo les ayudo
sin problema!!

drakorg, 2020-09-09:
Si, revisando el codigo (despues de hacer el comentario) vi que en
principio pareceria estar implementado, pero todavia no pude probar nada
porque no tengo ninguna printer a mano. En ese sentido, hay algun driver
dummy que pueda usar de manera de ir probando la parte del ida y vuelta de
json sin necesidad de tener una printer conectada? Entiendo que el driver
FILE iria por ese lado pero estoy mas bien adivinando, recien empiezo a
meterme un poco en tema.
Gracias, saludos.
Eduardo

alevilar, 2020-09-09:
exacto, tenes el driver File y el driver TXT, ambos son muy parecidos, creo
que uno imprimia mas a bajo nivel que el otro. Si mal no recuerdo File
genera el comando en formato ascii .txt y el driver Txt genera una salida
mas debuggueada.

cuando configures en tu .config tenes que poner algo asi:

[fiscal]
marca = Hasar
modelo = 715v2
driver = File
path = /tmp/respuestas.txt

marca y modelo te seleccionan el conjunto de comandos que queres probar. en
este caso seria una Hasar 715v2
y el driver es File y el path es el archivo donde sera generada la salida