paxapos/fiscalberry #108: tm-t900fa usb windows

jpgdixon, 2020-12-01:
Hola una consulta, para conectar una impresora tm-t900fa al USB sobre WINDOWS, cuales serian los parámetros?

alevilar, 2020-12-01:
Tendras que buscar el ID del vendor y del PRODUCT. la verdad en windows no
se bien donde los encontras. pero seguro en la administracion de
dispostivos.
Tambien tendras que instalar el driver para que te funcione
luego la configuracion seria algo asi

[NOMBRE_IMPRESORA_USB]

marca = EscPdriver = ReceiptUSBusb_vendor = *0x1c8a (CAMBIAR
ACA)*usb_product = *0x3a0e (CAMBIAR ACA)*in_ep = 0x81out_ep = 0x02

jpgdixon, 2020-12-12:
Necesito consultarte, realice pruebas con la Epson 220 y se conecto correctamente a la COM1 e imprimió el cierre x sin problemas. Ahora en la misma pc cambiamos esa impresora por la  T900 (SIEMPRE CON EL COM1) y no nos funciono. Por que puede suceder?

aditsoluciones, 2021-01-25:
la t900 es nueva generacion, y no esta terminada esa parte correctamente. estoy armando el traductor pero no tengo permisos sobre este repositorio. ademas agregue el parche para que la nueva generacion ande tambien en windows. usando las dll correspondientes.
probe y me anda en usb y com. pero estoy terminando el traductor, ya que algunas funciones cambiaron como ultimocomprobante.

aditsoluciones, 2021-01-25:
te paso maso el parche, para q te ande:
1- archivo: Epson2GenDriver.py reempaza todo antes de "class Epson2GenDriver(DriverInterface):"

# -*- coding: utf-8 -*-

from datetime import datetime
from DriverInterface import DriverInterface
import logging
from FiscalPrinterDriver import PrinterException
import ctypes
from ctypes import byref, c_int, c_char, c_char_p, c_long, c_short, create_string_buffer

import requests
import platform
import os
from ctypes import windll
archbits = platform.architecture()[0]
sowin = platform._syscmd_ver()[1]
newpath  = os.path.dirname(os.path.realpath(__file__))

if sowin == "Windows":
	if archbits[0:2] == "64":
		fullpath = "./epsonlib/win/64/EpsonFiscalInterface.dll"	
		EpsonLibInterface= windll.LoadLibrary(fullpath)
	else:
		fullpath = "./epsonlib/win/32/EpsonFiscalInterface.dll"
		EpsonLibInterface= windll.LoadLibrary(fullpath)
else:
	if archbits[0:2] == "64":
		fullpath = "./epsonlib/lin/64/libEpsonFiscalInterface.so"
		EpsonLibInterface = ctypes.cdll.LoadLibrary(fullpath)
	else:
		fullpath = "./epsonlib/lin/32/libEpsonFiscalInterface.so"
		EpsonLibInterface = ctypes.cdll.LoadLibrary(fullpath)

aditsoluciones, 2021-01-25:
obviamente tenes q bajar la dll oficial de epson. la podes bajar de https://ftp.epson.com/latin/drivers/pos/EpsonFiscalDriver_03.03.00.zip
la dll esta dentro del example/python
la prepare para tener tanto 32 como 64. pero usan la misma dll al final. asiq es la misma en ambas carpetas.

espero q te sirva. al menos asi ya arranca, falta armar el resto de tradccion pero recien hoy empece a verlo. y bueno esto avance.
te paso lo q tenes q poner en config.ini (si usas la web q trae agrega <option value="Epson2Gen">Epson2Gen</option> ) en marca y driver para que te lo tome. sino igual te paso el config

[epsonusb]
driver = Epson2Gen
path = 0
modelo = TM-T900
marca = Epson2Gen
baudrate = 115200

[epsoncom2]
driver = Epson2Gen
path = 2
modelo = TM-T900
marca = Epson2Gen
baudrate = 115200

espero ser de ayuda. saludos!

aditsoluciones, 2021-01-25:
creo q es obvio donde colocar la dll, pero por las dudas te lo aclaro.

epsonlib/win/32/
epsonlib/win/64/

recorda instalar el driver. mismo link q te pase para las dll

jpgdixon, 2021-01-27:
Hola, Gracias!

alevilar, 2021-01-27:
Excelente! gracias. si podes enviarnos un PULL REQUEST mejor, asi queda el aporte que realizaste a tu nombre y mas ordenado... luego lo testeamos y mergeamos en el proyecto principal

te paso el primer articulo que encontre en google sobre como realizar el PR https://desarrolloweb.com/articulos/pull-request-git.html

aditsoluciones, 2021-01-28:
Que tal ale, dame unos dias y lo subo, porque agregue varias cosas extras, y te la quiero documentar.
Para que sepas maso que le sume:
Try Exception en cada comando. para evitar que se cuelgue o perder comunicacion, con esto hice que en la respuesta de la api si hubo una exception te responde un json, con tu comando, parametros, detalleerror. 
agregue un metodo a la api que se llama listargument, donde le mandas un array de cosas a realizar, y te responde un solo json con sus respuestas en mismo orden y tambien informando que comando mandaste con q parametros, y si tuvo algun errores.
la idea de esto es desde una api poder hacer un solo post para traer informacion y no depender tanto del codigo a mano realizado en el proyecto. 
tambien agregue un nuevo comando llamado runcommand, el cual te permite mandar hasta 10 argumentos y ejecutarlo aunque no este desarrollado. 
y bueno ahora estoy en la etapa aburrida de cargar todos las combinaciones que tiene el modelo. estoy usando la guia oficial de epson para poner todos los comandos. ya que este modelo en argentina es el definitivo por unos años. estoy intentando que ya quede lo mas posible configurado.

aprovecho para preguntarte, tenes alguna informacion de como subir la auditoria de los nuevos modelos? ya que la reglamentacion dice que tenes una vez a la semana subirlo. si tienen maso algo veo de agregarlo en alguna funcion. y ya ayudamos mas a la gente, porq la verdad es muy poca la info que veo sobre el tema.

gracias!

aditsoluciones, 2021-01-28:
te muestro un ejemplo del nuevo listado de solicitudes

{"listcommand":[
    {"dailyClose":"x"},
    {"ConsultarVersionDll":"x"},
    {"getState":"x"},
    {"getLastError":"x"},
    {"getFiscalStatus":"x"},
    {"getPrinterStatus":"x"},
    {"getReturnCode":"x"},
    {"getPrinterStatus":"x"},
    {"getComPort":"x"},
    {"getBaudRate":"x"},
    {"getProtocolType":"x"},    
    {"SetSSLInsecureMode":"x"},
    {"GetTimeOut":"x"},
    {"GetResponseHeadersCount":"x"},
    {"getExtraFieldCount":"x"}
    ],"printerName":"real0"}

y su respuesta:

{
    "rta": [
        {
            "action": "dailyClose",
            "rta": 83886109,
            "params": "x"
        },
        {
            "action": "ConsultarVersionDll",
            "rta": {
                "Machinne Version": 0,
                "Major Machinne Version": 2,
                "Minor Machine Version": 9,
                "String Machinne Version": "Epson Fiscal Interface dynamic library (32-bit): 2.9.8  -  Epson Fiscal Driver dynamic library: 3.3.0"
            },
            "params": "x"
        },
        {
            "action": "getState",
            "rta": 0,
            "params": "x"
        },
        {
            "action": "getLastError",
            "rta": 0,
            "params": "x"
        },
        {
            "action": "getFiscalStatus",
            "rta": 0,
            "params": "x"
        },
        {
            "action": "getPrinterStatus",
            "rta": 0,
            "params": "x"
        },
        {
            "action": "getReturnCode",
            "rta": 0,
            "params": "x"
        },
        {
            "action": "getPrinterStatus",
            "rta": 0,
            "params": "x"
        },
        {
            "action": "getComPort",
            "rta": 0,
            "params": "x"
        },
        {
            "action": "getBaudRate",
            "rta": 115200,
            "params": "x"
        },
        {
            "action": "getProtocolType",
            "rta": 1,
            "params": "x"
        },
        {
            "action": "SetSSLInsecureMode",
            "params": "x",
            "err": "function 'SetSSLInsecureMode' not found"
        },
        {
            "action": "GetTimeOut",
            "params": "x",
            "err": "function 'GetTimeOut' not found"
        },
        {
            "action": "GetResponseHeadersCount",
            "params": "x",
            "err": "function 'GetResponseHeadersCount' not found"
        },
        {
            "action": "getExtraFieldCount",
            "params": "x",
            "err": "function 'getRegetExtraFieldCountturnCode' not found"
        }
    ]
}

alevilar, 2021-01-28:
me encanto el runCommand

respecto de las impresoras de 2da gen la verdad que no le estabamos dando
mucho soporte ya que nuestros clientes, en su mayoria, pueden usar factura
electronica, entonces nosotros estabamos ofreciendo imprimir la factura
electronica con una comandera comun y corrriente, para evitar el gasto de
un controlador fiscal.

Por otro lado, la epson de 2da gen la teniamos funcionando con los comandos
basicos:
imprimir ticket, informe x, z, notas de creditos...
lo que nunca llegamos a implementar era la auditoria que se sube
semanalment. Pero eso hacia la misma impresora. Es decir nosotros la 2gen
la conectabamos siempre por ethernet. no por USB. De esa manera podes
acceder a la IP del controlador y realizar determinadas tareas en la misma
impresora. Una de esas tareas era descargar los informes. Incluso creo que
la misma impresora los presentaba apretando un boton (ahora no estoy seguro
si eso era solo en las Hasar de 2da gen)

aditsoluciones, 2021-01-28:
Claro entiendo lo de la factura electronica. Pasa que en venta masiva conviene una impresora de ticket, yo lo tengo q implementar para que ande sin internet o aunque no ande la pagina de la afip.
para venta mas minorista esta genial la electronica y reducis mucho el gasto.
Cuanto tenga terminado el proceso de la epson lo comparto.
epson te dan un programa q tenes que instalarlo y agregarle los certificados para subir las auditorias.
osea se podria hacer que los mismos empleados lo suban. pero bueno si automatizamos las cosas evitamos el problema de que en alguna suc se olviden de subirlo. ademas eso de dejar los certificados desparramados por cada sucursal no me gusta mucho.
pienso ver si hago que al realizar un cierre z, verifique la fecha actual, si esta en el rango de presentacion que la descargue y la envie y dispare a un webhook con los archivos. 
con esto en un solo lugar te llegarian los de todas las sucursales, y se subirian de ahi. lo cual te permite ver cuantos controladores no llegaron(puede q esten sin internet varios dias, etc.) para proceder a la descarga manual de esos y no traer multas por falta de presentaciones.

asique en unos dias voy a estar subiendo, primero a testear todo bien, ya que python no es mi fuerte, trabajo a pleno con node, js y angular

saludos!

saludos! estamos hablando

NicoMachado, 2021-07-23:
Hola Soy nuevo en este mundo de las impresoras fiscales,
me dieron la tarea de implementar esta libreria en una TM-900, una Epson de 2da Generacion.
Lo que me paso es que no toma los Tickets letra A,B.C, etc .. siempre sale ticket C, (083),
Y lo que vi es que hay que enviar datos del cliente antes de abrir comprobante, y no se donde esta esa parte en el codigo.
Soy muy nuevo para python ...

De algun lugar se puede baja una version acorde a las Epson nuevas?

Saludos

alevilar, 2021-07-23:
Si te sale siempre factura C no es un problema del driver. Sino que asi
tenes configurada la impresora.

Todas las impresoras fiscales deben ser inicializadas, en ese proceso de
inicializacion es cuando la vinculan con la razon social de la empresa y le
settean el tipo de comprobante a imprimir. deberias hablar con la empresa
que les vendio la impresora sobre este problema.

Las Epson de 2da gen son compatibles con fiscalberry! saludos

NicoMachado, 2021-07-23:
Buenas! Gracias por la respuesta

Otra consulta, veo que hay muchos branches, yo tome la version del master,
esa esta actualizada?

Y la ultima, no nos esta devolviendo nada en Rta, luego de enviar un
ticket, deberia devolver el numero ?

Saludos

alevilar, 2021-07-23:
si. pero es asíncrono. o sea. la respuesta a un comando enviado es
simplemente una respuesta ajax.

luego deberas escuchar a la respuesta de la impresora con un evento JS

var comando = {
  "dailyClose": "X",
  "printerName":"epson-fiscal"
}

// lo envío usando la libreria fiscalberry mediante websockets
fbrry.send( comando );

fbrry.on( "fb:rta:dailyClose", function( obj, event){
    console.log("vino la data de respuesta: %o", event.data);} );

ahora si lo que mandas es un comando "printComanda" entonces el evento
a escuchar es

var comando = {
  "printComanda": {. . . . . . .},
  "printerName":"comandera"

}

// lo envío usando la libreria fiscalberry mediante websockets
fbrry.send( comando );

fbrry.on( "fb:rta:printComanda", function( obj, event){

    console.log("vino la data de respuesta al printComanda: %o", event.data);

} );

*O para imprimir ticket:*

var comando = {
  "printTicket": {. . . . . . .},
  "printerName":"impresora-fiscal"

}

fbrry.on( "fb:rta:printTicket", function( obj, event){.....
}

mas info sobre las respuestas aca:

https://github.com/paxapos/fiscalberry/wiki/JSON-Comandos-Para-impresora-Fiscal

NicoMachado, 2021-07-30:
Buenos Dias!,
otra consulta, mientras investigo lo anterior ...

Me comentan que el texto que dice "Vuelto" siempre esta saliendo en 0.
Estuve leyendo un poco la informacion de la interfaz dll de la impresora, y no encuentro nada, (como comente soy nuevo en esto de la imp fiscales) , me podrian dar una punta de como se podria resolver
seria un "texto extra" ?

Saludos

alevilar, 2021-07-30:
el vuelto seria un tipo de pago mas que le tenes que pasar en el JSON
fijate aca cuando detalla como enviarle los "pagos"

https://github.com/paxapos/fiscalberry/wiki/JSON-Comandos-Para-impresora-Fiscal#pagos-opcional

deberias ademas agregarle el "VUELTO"