paxapos/fiscalberry #119: Problema de conexión con Epson TM-T900FA via USB

fchiappano, 2022-03-22:
Buenas tardes,

Estoy intentando conectar fiscal berry con controlador fiscal epson de segunda generación TM-T900FA, pero no logro que fiscal berry se comunique con el CF. Lo extraño, es que en ningun momento se genera excepción alguna por fallo de comunicación o algo por el estilo. Simplemente, a recibir el json con el comando, cierra la comunicación.

El unico comando que probe y responde, es el getStatus, ya que no depende de la conexión con el CF.

Aca les dejo, lo devuelto por fiscal berry:

[root@adempiereqa fiscalberry  git:(master)]# python server.py
/urs/local/python27/lib/python2.7/site-packages/jwt/utils.py:8: CryptographyDeprecationWarning: Python 2 is no longer supported by the Python core team. Support for it is now deprecated in cryptography, and will be removed in the next release.
  from cryptography.hazmat.primitives.asymmetric.utils import (
2022-03-22 17:12:46,028 @ FiscalberryApp [INFO]: Preparando Fiscalberry Server
2022-03-22 17:12:46,030 @ FiscalberryApp [INFO]: La IP privada es 192.168.1.13
2022-03-22 17:12:46,030 @ Fb => Discover [INFO]: No hay Discover configurado
2022-03-22 17:12:46,031 @ FiscalberryApp [INFO]: Iniciando Fiscalberry Server
2022-03-22 17:12:46,097 @ FiscalberryApp [INFO]: *** Websocket Server Started as HTTP at 192.168.1.13 port 12000***
2022-03-22 17:12:46,098 @ FiscalberryApp [INFO]: Impresora disponible:
2022-03-22 17:12:46,098 @ FiscalberryApp [INFO]:   - IMPRESORA_FISCAL_E2G
2022-03-22 17:12:46,098 @ FiscalberryApp [INFO]:       marca: Epson2Gen, driver: Epson2Gen
2022-03-22 17:13:59,780 @ FiscalberryApp [INFO]: Connection Established
2022-03-22 17:13:59,783 @ FiscalberryApp [INFO]: Request 
 -> {"dailyClose":"X"}
2022-03-22 17:13:59,784 @ root [INFO]: Iniciando procesamiento de json:::: {"dailyClose": "X"}
2022-03-22 17:13:59,784 @ FiscalberryApp [INFO]: Connection Closed

=======================================================================================
Por aca, les pego mi config.ini:

[SERVIDOR]
puerto = 12000
discover_url =
ssl_crt_path =
ssl_key_path =
uuid = 548ca0439df4
ip_privada = 192.168.1.13

[IMPRESORA_FISCAL_E2G]
marca = Epson2Gen
modelo = tm-t900fa
driver = Epson2Gen
path = /dev/bus/usb/002/004

En este caso estoy trabajando con SO Centos.
Otra cosa queria consultarles, es como tendria que configurar fiscal berry, para poder operar con el CF via Ethernet, si es que no hay solución para la conexión via USB.

Quedo atento a sus comentarios.

Saludos,

spetrungaro, 2022-03-22:
Estimado, en el config.ini.sample esta el ejemplo de la impresora que desea configurar:

[IMPRESORA_FISCAL_E2G]
marca = Epson2Gen
modelo = tm-t900fa
driver = Epson2Gen
path = serial: /dev/usb/lp0

La parte de "serial: /dev..." en path es lo que te esta faltando. Ese driver en particular utiliza una libreria de Epson. Si lo estas corriendo como servicio, los prints que hace en consola el driver no los vas a ver, pero deberia loggear mas datos de los que estas viendo ahora

fchiappano, 2022-03-22:
Buenas tardes espetrungaro,

Gracias por tu pronta respuesta. Probe exactamente esa misma configuración, pero obtuve exactamente el mismo resultado expuesto anteriormente, donde no devuelve errores pero cierra la conexión ni bien recibe el json con los comandos.
Creo entonces, que lo que me esta faltando es la libreria de epson.

En la pagina oficial de epson, ofrecen esta libreria a modo de "Driver" : libEpsonFiscalDriver.so y libEpsonFiscalDriver.a.

A estas librerias haces referencia? donde espera fiscal berry que esten estan librerias?

Quedo mas que atento a tu respuesta.

Saludos,

spetrungaro, 2022-03-22:
La libreria esta incluída en fiscalberry, la encontras en [fiscalberry](https://github.com/paxapos/fiscalberry)/[epsonlib](https://github.com/paxapos/fiscalberry/tree/master/epsonlib)/lin/

Casualmente probé ese mismo modelo hace poco mas de dos semanas, solo hice un cierre X para asegurarme de que estuviera funcionando. En cuanto recuerde el setup que utilicé te lo puedo pasar

spetrungaro, 2022-03-22:
En el json te esta faltando la key "printername"

2022-03-22 17:13:59,784 @ root [INFO]: Iniciando procesamiento de json:::: {"dailyClose": "X"}

Tendria que ser {"dailyClose": "X", "printerName" : "IMPRESORA_FISCAL_E2G"}

Cuando abris localhost:1200, el primer paso es enviar el comando de Listar Impresoras, luego seleccionar la correspondiente del desplegable y recien ahi enviarle comandos a una impresora, sin printername no hay impresora, TraductoresHandler no lo pasa a ninguna funcion y la ejecucion termina ahi

fchiappano, 2022-03-22:
Entiendo lo que esta faltando. Lo voy a probar y te comento como me fue.

Saludos!

spetrungaro, 2022-03-22:
Me olvide de comentarlo, las librerias que encontras en epsonlib/lin/64/libEpsonFiscalinterface.so o epsonlib/lin/32/libEpsonFiscalinterface.so hay que copiarlas (la de 32 o 64 segun sistema a /lib64/libEpsonFiscalInterface.so o /lib/libEpsonFiscalInterface.so en linux. Despues de esos pasos te deberia funcionar

spetrungaro, 2022-03-22:
Ahi encontre el config que usé:

[fiscal]
marca = Epson2Gen
modelo = tm-t900fa
driver = Epson2Gen
path = serial:  /dev/usb/lp0

Con las librerias copiadas a /lib y /lib64 respectivamente y server.py corriendo con sudo.

Particularmente nunca usé fiscalberry fuera de debian, por lo que no puedo darte precisiones en centOS.

Comprobá que al conectar la impresora por usb se cree el archivo /dev/usb/lp0 (o lp1 etc). Pero es ahi donde se crea

spetrungaro, 2022-03-31:
Ahí detecte la razón del bug, en un rato estaré subiendo la corrección!

closed 2022-03-31 by spetrungaro

fchiappano, 2022-03-31:
Muchas gracias spetrungaro,

Por nuestra parte, queremos aportar en lo que podamos al proyecto (si es posible). No estamos tan habituados al desarrollo en python, pero algo hemos realizado hasta ahora.

Queria consultarte finalmente, si FiscalBerry contempla algo respuesto a impuestos internos, ya que tenemos un cliente que lo esta requiriendo.

Desde ya muchas gracias!!!

Saludos,

spetrungaro, 2022-04-01:
Se podría implementar para esa impresora, de hecho en el codigo estan comentados los parametros que habria que agregar en el metodo. Solo que el rubro donde nos desenvolvemos no maneja esos tributos:

https://github.com/paxapos/fiscalberry/blob/ad9000982b2810bbd18a18b92210c7dad3d14c41/Comandos/Epson2GenComandos.py#L156 

Podría agregarle esa funcionalidad y para **cada** item tendrían que agregar en el Json los siguentes parametros según manual, respetando el tipo requerido de cada uno (int y str): 

[image]

spetrungaro, 2022-04-01:
Abrí una nueva issue acerca de los Imp Internos para que pueda continuar esa temática ahí

commit ad900098 (2022-03-31): ticket type E2G fix #119