paxapos/fiscalberry #120: Controlador Fiscal Epson TM-T900FA, imprime solo comprobantes Ticket (cod 83)

fchiappano, 2022-04-08:
Buenos dias Estimados,

Hemos avanzado con las pruebas en el controlador fiscal epson TM-T900FA y hemos logrado imprimir comprobantes en el mismo.

En primera instancia les comento, que para las nuevas versiones de este controlador (tenemos disponible la version Neptuno 1.03), es necesario descargar la libreria actualizada EpsonLibInterface.so, desde el sitio oficial de epson, ya que de otra manera, el controlador devuelve el error 516.

Por otra parte, realizamos las modificaciones necesarias para el informe de impuestos internos, los cuales se estan pasando al controlador sin problemas.

El problema que estamos teniendo, es que independientemente del tipo de comprobante que se le envie al controlador, el mismo siempre emite un ticket "comun" (como si se le pasara el tipo de comprobante T en el json, 1 al controlador) en lugar de emitir un ticket-factura (tipo comprobante F + letra en json, 2 al controlador).
El tipo de comprobante, se esta pasando correctamente, ya que lo imprimi en consola antes de enviarlo.
Probamos con informar los datos del comprador (cliente) antes de abrir el ticket (siguiendo la indicaciones el manual), ya que actualmente esta operacion no se esta realizando (esta el metodo de informe de datos del comprador dentro de la clase Epson2GenComandos.py, pero no se esta ejecutando), pero obtuvimos el mismo resultado.

Tambien tuvimos que retocar el metodo de informe de datos del comprador, ya que el envio de parametros en "None", generaba que el controlador cancelara la operación.

Esto ejecutaba el metodo original:  self.conector.driver.EpsonLibInterface.CargarDatosCliente(
                                                           name, None, address, None, None, self.docTypes.get(docType), doc, self.ivaTypes.get(ivaType))

Asi lo deje para las pruebas: self.conector.driver.EpsonLibInterface.CargarDatosCliente(
                                               name, " ", address, " ", " ", 3, doc, 1)

Tanto el self.docTypes.get(docType), como el self.docTypes.get(docType), retornaban None, aun cuando se les enviaba en el json el string correctamente especificado ("CUIT" y "RESPONSABLE_INSCRIPTO" respectivamente), por lo cual force el envio de los codigos directamente, para avanzar con las pruebas.

El envío de datos del comprador, probe realizarlo antes y despues de abrir el ticket, en ambos casos obteniendo el mismo resultado (la emisión de un ticket cod 83, no factura).

No tengo claro, que me pueda estar faltando para la emisión de un ticket-factura.

Desde ya, les agradezco su tiempo y su ayuda.

Saludos cordiales,

spetrungaro, 2022-04-08:
Como estas? Mirá, me di cuenta que no estaba terminado de implementar el driver, por lo que en la rama "epson2g" subí la versión con todo implementado. También noté que emitia solo "tique" comun, y eso era hasta le empecé a enviarle los datos completos del consumidor. Te pediría que pruebes esa nueva rama que yo la usé con todas las funciones de Tique, Factura B, Factura A, NCA, NCB, percepciones, impuestos internos y pagos. 

Los datos del cliente se envian antes de abrir el comprobante. Pero igualmente eso lo hace todo el controlador que preparé en el orden que corresponde.

Si probás con el mismo JSON que me enviaste la otra vez, y la rama epson2g, te va a elaborar una factura A:
{"printerName":"IMPRESORA_FISCAL_E2G","printTicket":{"encabezado":{"tipo_cbte":"FA","tipo_doc":"CUIT","domicilio_cliente":"600 HECTAREAS , IGUAZU, PUERTO IGUAZU 3370, Misiones ","nro_doc":"30710458401","tipo_responsable":"RESPONSABLE_INSCRIPTO","nombre_cliente":"AWVB S.A.- Awasi "},"items":[{"alic_iva":21.00,"qty":1.00,"importe":21.80,"ds":"GALL.MAURI VAINILLAS x80g"}]}}

fchiappano, 2022-04-08:
Buenas tardes Santiago!

Volvi a probar con lo desarrollado en la nueva rama y la factura salio perfectamente!!

Lo unico que esta faltando, es la impresión del impuesto interno. En este caso le pase un impuesto interno fijo (codigo 1) y se levanto correctamente en fiscalberry, ya que lo veo impreso en la consola, pero por algun motivo el controlador no lo imprimio.
Voy a serguir revisando este punto y tratar derminar si hay que realizar alguna modificación.

No te puedo decir exactamente cuanto del rollo de papel llevo gastado, ya que estoy trabajando de manera remota (lo que me complica aun mas las pruebas), pero debo haber gastado bastante tambien jaja

Te voy comentando cualquier avance o novedad que pueda tener al respecto.

Saludos!

spetrungaro, 2022-04-08:
Tené en cuenta que aparentemente, segun las pruebas que realicé, el ImpImterno del item tiene limitaciones, por ejemplo que no puede ser cierto porcentaje del total del mismo o cosas asi, probando con precios mas altos e impuesto que significa un porcentaje menor, si me lo incluia. Pero eso ya son cosas más tecnicas de áreas que no manejamos ni comprendemos. Todo eso lo decide el controlador fiscal. Mientras la info haya salido, la impresora lo recibe.

fchiappano, 2022-04-20:
Buenas tardes Santiago,

Te comento que hemos avanzado con las pruebas en el controlador fiscal y detectamos un inconveniente al enviar un impuesto interno fijo al controlador (a nivel de item), tal y como te habia mencionado anteriormente.

El problema lo detectamos al enviar dentro del Json el impuesto interno como string. En la consola se veia impreso tal como estaba en el json, pero al enviarlo al controlador el mismo cambiaba (por ejemplo, enviaba $46,23 y el controlador imprimia solo $4).

La unica forma que encontre de resolver esto, fue enviar el impuesto interno como numero decimal, tal como se envian los importes y convertirlo a string antes de enviarlo al controlador (con un if de por medio, para enviar un string vacio si el impuesto interno es cero). Para ello, quite el parametro ii_importe del **kargs y lo deje como un parametro fijo con el valor por defecto 0.

De esta forma, me quedaron todos los campos que referencian importes dentro del json, con el mismo formato, lo cual a su vez me parecio piola y practico.

Quedo atento a tus comentarios y desde ya, si podemos colaborar con este minimo aporte, conta ello.

Saludos,

spetrungaro, 2022-04-20:
Me podras enviar el json correspondiente al error? Me serviría mucho.
Y si, nos interesaria todo aporte al código. Forkea el proyecto y añadile tus cambios asi los podemos unificar con una PR

fchiappano, 2022-04-21:
Santiago,

Te dejo a continuación el json con el que realice gran parte de las pruebas:

============= PRUEBA ANTERIOR DONDE IMPRIMIO $4 DE II ==================================
{
	"printerName": "IMPRESORA_FISCAL_E2G",
	"printTicket": {
		"encabezado": {
			"tipo_cbte": "FA",
			"tipo_doc": "CUIT",
			"domicilio_cliente": "RUTA 12 KM5, IGUAZU,  PUERTO IGUAZU 3370, Misiones ",
			"nro_doc": "30715161997",
			"tipo_responsable": "RESPONSABLE_INSCRIPTO",
			"nombre_cliente": "COMIT SRL"
		},
		"items": [{
			"id_ii": 1,
			"alic_iva": 21.00,
			"qty": 2.00,
			"ii_valor": "23.52",
			"importe": 172.85,
			"ds": "GANCIA x950 + FERNET 1882 x750ml"
		}]
	}
}

=======================================================================================0

En este caso, se envio un impuesto interno fijo (como string) de $23,52 y una cantidad de 2 unidades, con lo cual el CF deberia haber impreso un impuesto interno de $47,04 y solo imprimio $4,00.

Voy a hacer el fork del proyecto y aplicar mis cambios asi lo podes ver tambien.

Saludos,

yoMatias1, 2023-08-15:
Hola, tengo un problema con una epson TM-T900, imprime tique b pero con los A da error, he estado investigando sobre lo que han escrito y me gustaria probar estos comandos pero no se como podria cargarlos. Gracias

yoMatias1, 2023-08-16:
Adjunto una captura de video de la pantalla y del error que da de campo no opcional
[rec-screen.webm](https://github.com/paxapos/fiscalberry/assets/142352300/adfb7571-95a8-4560-81fe-7f5aefaa8889)
[image]
dice comando 0B01 "Campo no opcional"

alevilar, 2023-08-16:
tenes manera de ver la consola de python, en fiscalberry? quizas ayuda a
ver exactamente cual es el comando que aparece invalido cuyo campo no es
opcional

> Message ID: ***@***.***>
>

alevilar, 2023-08-16:
> No sé cómo usarlo, el controlador es de un cliente y con su programa de
> facturación no permite hacer facturas A pero si B y con el Epson mánager
> hace lo mismo

pero estan usando fiscalberry? o me estas hablando de otro software
jejej? 😐

yoMatias1, 2023-08-16:
No, es otro software, llegué acá por el buscador y ya he agotado todas las opciones, por eso me decidí a consultar

joseluisbossio, 2023-08-17:
PLOP!

closed 2023-08-17 by alevilar